bdring/FluidNC #917: Problem: GHOST LINE with laser

manukieli, 2023-06-09:
### Controller Board

MKS DLC32

### Machine Description

Belt MXL (Pith: 2,03mm)
Pulley 20 tooth
CNC Laser

### Configuration file

```yaml
stepping:
  engine: I2S_STATIC
  #engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 78.82
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 810.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 3000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0: ...

  y:
    steps_per_mm: 78.82
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 790.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 3000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0: ...
```

### Startup Messages

```text
No error message.
```

### What happened?

I've configure steps per millimeter to 78.82 to X and Y beacause I'm using BELT MXL with pitch is 2,03mm.
And when I'm engraving  they appear some "ghost lines" into it and the size of the cutting piece is correct.
But if I'm using steps per millimeter to 80, the "ghost lines" disepear, but the sizes of the pieces is not correct, because the step is not the correct.
I'm using the link to calculate de steps: https://zalophusdokdo.github.io/StepperMotorsCalculator/en/index.html

Into the image, you can the see the problem:

[image]

MitchBradley, 2023-06-09:
Why did you leave out part of the config file above?  Why did you omit the startup messages.

Please do not assume that you know what we need and don't need to see.

MitchBradley, 2023-06-09:
It would also be helpful to see the gcode file.

manukieli, 2023-06-09:
> Why did you leave out part of the config file above? Why did you omit the startup messages.
> 
> Please do not assume that you know what we need and don't need to see.

Sorry, I didn't think it would be useful.
I have already edited the post putting the configution.
Regarding the message, I'll have to look at it tomorrow.
Thank you very much for your suggestion.

manukieli, 2023-06-09:
> It would also be helpful to see the gcode file.

Ok, I put the Gcode here:
`
; LightBurn 1.4.00
; GRBL device profile, current position
; Bounds: X-8 Y-60 to X38 Y0

;USER START SCRIPT
M7;
M8;
;USER START SCRIPT

G00 G17 G40 G21 G54
G91
M4
; Scan @ 500 mm/sec, 10% power
M9
G0 X-8Y-57.95
; Layer C02
G1 X10F30000S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
[... 2184 lines cut ...]
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
; Cut @ 7 mm/sec, 20% power
M7
G0 X8Y2.05
; Layer C00
G1 X30S4000F420
G1 Y-60
G1 X-30
G1 Y60
M9
G1 S0
M5
; return to starting pos
G0 X0Y-0

;USER END SCRIPT
;ENDBLOWER
;ENDBLOWER
;USER END SCRIPT
M2
`

MitchBradley, 2023-06-09:
It would be interesting to see if the ghosting happens in m3 mode instead of m4.

manukieli, 2023-06-10:
[image]

I've add the result with M3 mode, I've the same problem:

[image]

The GCODE with M3 mode is here:
`
; LightBurn 1.4.00
; GRBL-M3 (1.1e or earlier) device profile, current position
; Bounds: X-8 Y-60 to X38 Y0

;USER START SCRIPT
M7;
M8;
;USER START SCRIPT

G00 G17 G40 G21 G54
G91
; Scan @ 500 mm/sec, 10% power
M9
M5
G0 X-8Y-57.95
; Layer C02
M3
G1 X10F30000S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
G1 X-26S2000
G1 X-10S0
G1 Y0.1S0
G1 X10S0
G1 X26S2000
G1 X10S0
G1 Y0.1S0
G1 X-10S0
[... 2191 lines cut ...]
G1 X-10S0
G1 X-26S2000
G1 X-10S0
; Cut @ 7 mm/sec, 20% power
M7
M5
G0 X8Y2.05
; Layer C00
M3
G1 X30S4000F420
G1 Y-60
G1 X-30
G1 Y60
M5
M9
G1 S0
; return to starting pos
G0 X0Y-0

;USER END SCRIPT
;ENDBLOWER
;ENDBLOWER
;USER END SCRIPT
M2
`

manukieli, 2023-06-10:
And the startup message is this:
`
$G
$Report/Interval=50
[MSG:INFO: Channel auto report interval set to 50 ms]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:P|WCO:0.000,0.000,0.000>
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
`

manukieli, 2023-06-10:
It is possible that the problem is caused by some rounding calculation at some point in the firmware due to the pitch of the belt (2.03mm).
I found this post that has nothing to do with my problem itself, but with the issue of the passage of the belt:
https://github.com/bdring/FluidNC/issues/498

bdring, 2023-06-10:
Could it be backlash?

MitchBradley, 2023-06-11:
Try switching from auto-reporting to polling in webui to see if that makes a difference

manukieli, 2023-06-11:
> Try switching from auto-reporting to polling in webui to see if that makes a difference

I've try polling to 5 second and the exatly the same problem.

MitchBradley, 2023-06-11:
Okay that test shows that auto-reporting is not related to the problem.  Could you try some other steps_per_mm values like 80.5, and also some slower speeds.

MitchBradley, 2023-06-12:
If steps_per_mm is 78.74, in order to move exactly 1 mm, it would be necessary to issue 78.74 steps.  But you cannot sent a fractional step pulse to a stepper driver.  You could send 78 pulses to make it move 78/78.74 = .991 mm or 79 pulses for 79/78.74 = 1.003 mm (rounded to 3 decimal places).  That is the "roundoff problem" that is a factor in any CNC system - there are distances that cannot be reached exactly.  Even if the target position is an exact number of steps, mechanical inaccuracies will make the actual position differ somewhat.  If the readout says 1 mm, any real machine will be off by a small amount.  If you go from 0 to 1, the actual physical position will usually be slightly different than if you go from 10 to 1.  The real world is not mathematically exact.

Most systems are designed so that the stepping precision - which in this case is approximately 1/80 mm - is better than the mechanical repeatability.  So even though the reported position is off by a fraction, it doesn't matter because the real machine is going to be off by more than that due to mechanical imperfections.

This does not explain your ghosting.  It could be that it is indirectly related due to other computational factors, but it is not a problem in and of itself.

MitchBradley, 2023-06-12:
The reason why 50 seems to be exact is because 50 * 78.74 is an integer (=3937).  Any multiple of 50 will be exact, but no other distance will.

manukieli, 2023-06-12:
> Did you try slower speeds?

Mitch, thank you so much for your time and all your explanations.
I will try with a slower speed and I tell you.

manukieli, 2023-06-14:
> Did you try slower speeds?

I did a lot of tests and always got the same ghosting result with 78.72 steps/mm.
After all the tests I did a complete cleaning (lenses, mirrors) and complete adjustments (mirror adjustments, belt tension, squaring of the X gantry).
And I did a new test with a slower speed, this is the result:
[image]

The problem persist with 78.72 steps/mm.

MitchBradley, 2023-06-18:
Is it a CO2 laser or a diode laser?  Have you tried running at a lower speed?

manukieli, 2023-06-18:
> Is it a CO2 laser or a diode laser? Have you tried running at a lower speed?

It's a CO2 laser, in the image you see a lower speed of 250mm/s, but I can try to do it at a lower speed still.

manukieli, 2023-06-18:
This is the resulta with lower speed:
[image]

MitchBradley, 2023-06-18:
Does the machine sound different when the steps/mm is different?  Perhaps smoother sound with 80?

manukieli, 2023-06-18:
> Does the machine sound different when the steps/mm is different? Perhaps smoother sound with 80?

That is, with 80 steps per mm, the engraving is correct, it does not have any errors. However, with 80 steps per mm, the dimensions of the cut part are not correct, the part is slightly larger than expected.
With 78.74 steps per mm, the dimensions are as expected, but the ghosting error appears within the engraving.

MitchBradley, 2023-06-18:
The presence of dark and light bands between the ghost lines makes me think that the laser power supply or the gas discharge is not stable.  I do not know why the steps/mm change would cause that.  I suppose it could be related to some sort of mechanical resonance. CO2 lasers are complicated.

Since your machine is fundamentally based on inches (0.08" belt pitch), you might try making the image dimensions a multiple of that.  Instead of 26 mm, you could use an image width of 26.416 mm which works out to an integer number of teeth on your belt.  The overscan should be adjusted too, to say 10.15 mm.  The Y advance would go from .1mm to .1015 mm.  Doing that would eliminate any need to skips a step every so often.

The other thing you could try would be to change the PWM frequency.  According to what I have read, laser power supplies can be sensitive to that.

MitchBradley, 2023-06-18:
I asked about the sound because I was wondering if the non-integer stepping could be causing vibrations.  If so you might be able to hear them.

manukieli, 2023-06-18:
> Does the machine sound different when the steps/mm is different? Perhaps smoother sound with 80?

Excuse me for the mistake with my answer, I haven't read correcty your post, instead of "sound "he had understood "seems" 😅.
They are no problem with vibrations or other sound with some and other steps per mm.

MitchBradley, 2023-06-18:
I modified FluidNC to display the details of step generation and power control.  I did not find anything wrong at either 80 or 78.74 . Maybe there is something that I am missing, but since I do not have access to your hardware there is not much more that I can do without spending a ridiculous amount of time.

When I was working for computer manufacturers I would sometimes spend weeks or months tracking down difficult problems, in order to enable millions of dollars worth of product shipments.  When the "product" is free, and is used on a huge variety of third-party hardware that often costs less than $20, it is hard to justify spending a lot of time on obscure problems that you cannot reproduce.

Another workaround that you could try:  Set steps/mm to 80 and then scale the image externally - shrink everything by a factor of 78.74/80 .

manukieli, 2023-06-20:
Thank you so much @MitchBradley for you help and your time, I understand what you are saying.
I will change all the belts and pulleys to GT2 and solved.

closed 2023-06-21 by manukieli