bdring/FluidNC #923: Problem: Fluidterm.sh fails to load

maxcel2, 2023-06-14:
### Controller Board

N/a

### Machine Description

N/a

### Input Circuits

```yaml
N/a
```

### Configuration file

```yaml
N/a
```

### Startup Messages

```text
N/a
```

### User Interface Software

FluidTerm

### What happened?

Host: Release Linux Mint 20.3 Una 64-bit
Kernel Linux 5.4.0-150-generic x86_64
Fresh load of version 3.7.2, from terminal, ran erase.sh, then installed install-fs.sh & install-wifi.sh.
Then ran ./fluidterm.sh and got this error

> Traceback (most recent call last):
  File "common/fluidterm.py", line 328, in <module>
    class NoControls(NoTerminal):
  File "common/fluidterm.py", line 331, in NoControls
    REPLACEMENT_MAP = {x: 0x2400 + x for x in range(32)} | {
TypeError: unsupported operand type(s) for |: 'dict' and 'dict'.

maxcel2, 2023-06-14:
Version is fluidnc-3.7.1-posix

Alex-CodeLab, 2023-06-15:
@maxcel2 , could you check your Python version ( `python -V` )
I think you are using something older than python3.9 . 
So updating the python version would probably fix it.
 
However, I know that is not always easy or possible, so I will try to create a fix.

       The union operator was implemented for dictionaries in Python 3.9 (see [PEP 584](https://peps.python.org/pep-0584/))

maxcel2, 2023-06-15:
@Alex-CodeLab :
The error appears to lay in V3.7.1's fluidterm.py:
`class NoControls(NoTerminal):
    """Remove all control codes, incl. CR+LF"""

    REPLACEMENT_MAP = {x: 0x2400 + x for x in range(32)} | {
        0x20: 0x2423,  # visual space
        0x7F: 0x2421,  # DEL
        0x9B: 0x2425,  # CSI
    }`
Note the curly braces after '=', Whereas V3.6.4, below, has the same structure as all other calls to 'REPLACEMENT_MAP'.
`class NoControls(NoTerminal):
    """Remove all control codes, incl. CR+LF"""

    REPLACEMENT_MAP = dict((x, 0x2400 + x) for x in range(32))
    REPLACEMENT_MAP.update(
        {
            0x20: 0x2423,  # visual space
            0x7F: 0x2421,  # DEL
            0x9B: 0x2425,  # CSI
        })`

Alex-CodeLab, 2023-06-15:
Correct, that is a newer way of writing the same (since python3.9) . It does the same.

(I will change it back)

maxcel2, 2023-06-15:
Copying and pasting the class NoControls(NoTerminal) from fluidnc.py v3.6.4 to v3.7.1, solves this issue.

closed 2024-12-11 by bdring