bdring/FluidNC #925: Feature: CW/CCW Driver control

Mvene2001, 2023-06-14:
### Machine Context

There are several tipes of stepper driver, some industrial ones with very high torque and advance control features uses CW/CCW insted of step/direction inputs.

Can you add a way to configure the control type?
Or can you tell which file controls the step and direction output?

Or is a preffered way to do this using external hardware?

Thank you.

### Feature Description

In the config.yaml file place a new setting that switches between step/dir and cw/ccw for each axis.

### Other Approaches

Using a external hardware to translate from step/dir to cw/ccw

### How I Can Help

I can program it if given direction of which file generates the step and direction outputs.

MitchBradley, 2023-06-14:
Stepping is controlled by various files including Stepper.cpp, Stepping.cpp, I2SStepping.cpp, and Motors/StandardStepper.cpp .  It is complicated because FluidNC supports multiple hardware interfaces to generate step pulses and multiple motor types.  CW/CCW motors are very rarely used in the class of machines that the vast majority of our users have, so we are very unlikely to put CW/CCW support into our mainline code base.  It would be impossible for the main developers to maintain since we have no relevant hardware and little incentive.

closed 2023-06-14 by MitchBradley

Mvene2001, 2023-06-14:
I understand, thank you. I will look at those files.

MitchBradley, 2023-06-15:
External hardware might be easier to maintain.  A SN74LVC1G19 for each motor might do the trick.  Or you could program a PAL - not that anyone has PAL programmers lying around anymore.