bdring/FluidNC #931: Problem: RS485 Problem between xPro-V5 and VFD HUANYANG 2,2kW (FluidNC 3.7.1)

slvv26, 2023-06-20:
### Controller Board

xPro-V5

### Machine Description

UltimateBee xPro-V5 with VFD Huanyang 2,2 kW and water cooled spindle 2,2kW.
RS485 connected by twisted wire 300mm.

### Configuration file

```yaml
# For more information please see: http://wiki.fluidnc.com/en/config/overview

board: XPro V5
name: xPro V5 xPro V5 UltimateBee 1500x1500 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 159.600
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 722.00
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14:low
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 160.100
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1252.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.5000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2:low
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 10000000

probe:
  pin: gpio.22:low
  check_mode_start: true

control:
  safety_door_pin: gpio.16:high
  macro0_pin: gpio.13:low
#  macro1_pin: gpio.0:low

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  mist_pin: gpio.21
  delay_ms: 0

# For spindle configuration remove the "#" preceding your spindle type BELOW... 
 
Huanyang:
 uart:
   txd_pin: gpio.25
   rxd_pin: gpio.4
   rts_pin: NO_PIN
   baud: 9600
   mode: 8N1
 modbus_id: 1
 tool_num: 0
 speed_map: 0=0% 0=25% 6000=25% 24000=100%
 off_on_alarm: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.7.1]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine xPro V5 xPro V5 UltimateBee 1500x1500 ]
[MSG:INFO: Board XPro V5]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (150.000,872.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14:low CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:INFO: Axis Y (150.000,1402.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Neg Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2:low CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z Neg Limit gpio.39:low]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.16]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: UART1 Tx:gpio.25 Rx:gpio.4 RTS:NO_PIN Baud:9600]
[MSG:INFO: Huanyang Spindle  Tx:gpio.25 Rx:gpio.4 RTS:NO_PIN Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI, CNCjs, FluidTerm

### What happened?

**[MSG:INFO: VFD RS485 Unresponsive]**

No communication with VFD via RS485. **The LED blinks only green every second, but very, very weakly.**

I tried the old version of the firmware from Spark-Concepts - CNC_xPRO_V5_XYYZ_NC.bin. The same settings, but with the $Spindle/Type=HuanYang command in the terminal (FluidTerm, WebUI or CNCjs). Old firmware seems doesn't support config.yaml files so I did it via console.
**On the old firmware,** the connection is established well and the **LED flashes red / green very brightly**. The connection is only broken when the spindle is started.

On FluidNC 3.7.1 there is no hint that the port is being activated.

I burn FluidNC according to HOWTO-INSTALL.txt

erase.bat
install-wifi.bat
install-fs.bat
Upload config.yaml (https://github.com/Spark-Concepts/xPro-V5/blob/main/FluidNC/config.yaml) by FluidTerm or WebUI.
Reboot

Any ideas?

slvv26, 2023-06-20:
Solved...

Spent two days looking for a solution...
Just noticed that on the old firmware settings are:

[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485 **Tx:GPIO(4)** Rx:GPIO(25) RTS:GPIO(0)]

In the default config.yaml (https://github.com/Spark-Concepts/xPro-V5/blob/main/FluidNC/config.yaml) the gpio in the settings are mixed up

Huanyang:
  uart:
    **txd_pin:gpio.25**
    rxd_pin:gpio.4
    rts_pin: NO_PIN
    baud: 9600
    mode: 8N1

Swapped them and everything worked.

Actualy I didn't expect that this is possible with the recommended config-file in terms of interfaces, because they are the same for every xPro-V5.

slvv26, 2023-06-20:
Solved...

Spent two days looking for a solution...
Just noticed that on the old firmware settings are:

[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485 Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]

In the default config.yaml (https://github.com/Spark-Concepts/xPro-V5/blob/main/FluidNC/config.yaml) the gpio in the settings are mixed up

Huanyang:
uart:
txd_pin:gpio.25
rxd_pin:gpio.4
rts_pin: NO_PIN
baud: 9600
mode: 8N1

Swapped them and everything worked.

Actualy I didn't expect that this is possible with the recommended config-file in terms of interfaces, because they are the same for every xPro-V5.

bdring, 2023-06-20:
I can add an xPro 5 wiki page with your working config if you attach it to a reply.

slvv26, 2023-06-20:
Sure, but it is default config.yaml with the only change - swapped gpio Tx and Rx. Maybe it's better to just add a commented out option to the file, something like "if it doesn't work, try using this". Anyway below is my config.

```yaml

# For more information please see: http://wiki.fluidnc.com/en/config/overview

board: XPro V5
name: xPro V5 xPro V5 UltimateBee 1500x1500 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 159.600
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 722.00
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14:low
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 160.100
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1252.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.5000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2:low
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 10000000

probe:
  pin: gpio.22:low
  check_mode_start: true

control:
  safety_door_pin: gpio.16:high
  macro0_pin: gpio.13:low
#  macro1_pin: gpio.0:low

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  mist_pin: gpio.21
  delay_ms: 0

# For spindle configuration remove the "#" preceding your spindle type BELOW... 

Huanyang:
 uart:
   txd_pin: gpio.4
   rxd_pin: gpio.25
   rts_pin: NO_PIN
   baud: 9600
   mode: 8N1
 modbus_id: 1
 tool_num: 0
 speed_map: 0=0% 0=25% 6000=25% 24000=100%
 off_on_alarm: false

```

bdring, 2023-06-20:
OK, I'll start the page with only the huanyang section. People can add more to it later.

``` yaml
Huanyang:
 uart:
   txd_pin: gpio.4
   rxd_pin: gpio.25
   rts_pin: NO_PIN
   baud: 9600
   mode: 8N1
 modbus_id: 1
 tool_num: 0
 speed_map: 0=0% 0=25% 6000=25% 24000=100%
 off_on_alarm: false
 ```

slvv26, 2023-06-20:
Ok, thanks.

slvv26, 2023-06-20:
Nop

MitchBradley, 2023-06-20:
It seems like that is where you got the wrong information, so maybe they should be notified instead of us.

bdring, 2023-06-20:
There already was a wiki page

http://wiki.fluidnc.com/en/hardware/3rd-party/xPro_V5

adambaranyai, 2023-07-02:
I ran into the same exact issue. Meaning that I used the example config (this: https://github.com/Spark-Concepts/xPro-V5/blob/main/FluidNC/config.yaml) for a Huanyang spindle and it just didn't want to work via the RS485 port. I opened https://github.com/Spark-Concepts/xPro-V5/issues/220 to have the example config corrected as I'm sure other people will run into the same problem.