bdring/FluidNC #933: Problem: Image raster is not "smooth"

Avataar120, 2023-06-21:
### Controller Board

Custom board for a K40 laser engraver

### Machine Description

K40 laser machine

### Input Circuits

```yaml
N/A
```

### Configuration file

```yaml
name: K40_AvaShield
board: 4_Ava_Shield_8.x_XY_OLED

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 310.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 210.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.7
        step_pin: I2SO.6
        direction_pin: I2SO.5:low
        disable_pin: I2SO.4

  z:
    steps_per_mm: 1000.000
    max_rate_mm_per_min: 200.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 20.000
    soft_limits: false
    homing:
      cycle: 
      positive_direction: true
      mpos_mm: 9.500
      feed_mm_per_min: 100.000
      seek_mm_per_min: 100.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 9.500
      stepstick:
        ms3_pin: I2SO.11
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

i2c0:
   sda_pin: gpio.25
   scl_pin: gpio.33

oled:
   i2c_num: 0
   i2c_address: 60
   width: 128
   height: 64
   radio_delay_ms: 1000

control:
  safety_door_pin: gpio.27:low:pu
  reset_pin: gpio.12:low:pu
  feed_hold_pin: gpio.14:low:pu
  cycle_start_pin: gpio.13:low:pu

coolant:
  mist_pin: gpio.4
  delay_ms: 50.000

probe:
  pin: gpio.36:low
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 0
  speed_map: 0=0.0% 1000=100.0%
  output_pin: gpio.15
  enable_pin: gpio.16:low
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 15000

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
N/A
```

### User Interface Software

Lightburn

### What happened?

When engraving a image through Lightburn, movement is "jerky"
When starting a new line, I'm expecting : 
- laser to go outside of the image by 2.5% (my overscanning setting)
- start to accelerate on X, reach target speed (350mm/s)
- Turn on and off laser according to GCODE but w/o decelerating / accelerating at each new GCODE command

What I observe : 
- laser is accelerating / deceleration at each G0 GCODE command (even if those commands are moving only on X axis at constant speed)

Do I miss something ?

See extract of my gcode below (F is constant, S is variating), however, laser is accelerating / decelerating wiith each & every command.

; LightBurn 1.4.00
; GRBL device profile, absolute coords
; Bounds: X11.96 Y26.7 to X216.56 Y178.5
G00 G17 G40 G21 G54
G90
M4
; Image @ 350 mm/sec, 20% power
M7
G0 X13.564Y178.499
M3
G0 Z0
; Layer C00
G91
G1 X8.75F21000S0
G1 X1.2S200
G1 X8.75S0
G1 X0.5Y-0.1S0
G1 X-8.75S0
G1 X-2.1S200
G1 X-8.75S0
G1 X-0.2Y-0.1S0
G1 X8.75S0
G1 X2.5S200
G1 X8.75S0
G1 X0.4Y-0.1S0
...

Thanks !

Avataar120, 2023-06-27:
Hello, 

Any hint if this issue please ?

Thanks,

breiler, 2023-06-27:
Does the machine work smoothly if you dial down the `max_rate_mm_per_min` and `acceleration_mm_per_sec2` settings?

How are you connecting, TCP or Serial? If serial are you using the default baud rate 115200?

Avataar120, 2023-06-27:
I'm connected through USB - 115200bauds
If I draw a filled rectangle (meaning that there is no ON / OFF of laser needed to engrave it), the operation is 100% smooth even at very high speed (350mm/s)
The problem appears each when engraving an image as laser has to go ON & OFF according to the image and it seems to me that FluidNC is decelerating / accelerating at each GCODE command where it is only needed to turn ON / OFF the laser.

breiler, 2023-06-27:
My best guess is that the serial buffer is starving. 

Let us assume that you have a ~10 bytes of gcode every 0.2mm, turning the laser off/on. That would make a theoretical bandwith requirement of 16kB/s if running in 350mm/s. The maximum bandwidth possible with 115200 bauds is around 11-14kB/s.

The serial buffer in the controller needs to be filled with enough gcode for it to properly plan future motion. If it doesn't have enough commands it will have to start decelerate. And with your current acceleration settings it should be able to stop from a full speed in 0.175 seconds.

If you compare that to your rectangle test, that is four points, about ~40 bytes of gcode to be transmitted.

If you decrease the max speed to 10000 mm/min and the acceleration to 1000 mm/sec2 does the improve the movement?

Avataar120, 2023-06-27:
If you look at the GCODE example above, you'll see that issue is appearing even with some simple drawings.
Indeed for the first line, only 3 gcodes commands for approx 18mm
G1 X8.75F21000S0
G1 X1.2S200
G1 X8.75S0
And issue occurs.

I can see it as well if I draw a non filled rectangle at 20mm/s and I put some tab on the perimeter of the rectangle to keep the rectanble attached to the sheet of wood.

I don't think this is linked to buffer starving issue but with deceleration / acceleration at each GCODE commands which is not needed for a laser (probably needed for a CNC - I don't know :))

MitchBradley, 2023-06-27:
Just as a test, try setting laser: enable_pin: NO_PIN.  If that changes things it could provide a clue.

Avataar120, 2023-06-27:
Hi Mitch,

I just tested, unfortunately, same behavior

Avataar120, 2023-07-01:
Hi,

I made some new tests this morning.
I modified the GCODE to keep S constant.
So now, F (feed) and S (spindle) are constant in all the GCODE.
I still experienced exactly the same behavior with deceleration & acceleration during X only moves.
It seems to me that each GCODE command is processed as if it needs decel / accel (even if not needed in this case)

Hope it could help in the rootcause analysis.

MitchBradley, 2023-07-01:
Try sending the gcode file with a different sender.  Your symptoms could be explained if the sender is pausing between lines, instead of sending the next line as soon as the previous one is acked with "ok".  I don't know how Lightburn works (it is closed source), and I don't know how it is configured, and you didn't attach any screenshots of Lightburn console output, so I cannot dig deeper into Lightburn sending behavior.  Try UGS or some other sender.  If that works, then the problem is at the serial interface level, not the motion level.

You could even use FluidTerm.  It has a rudimentary sender built-in.  The FluidTerm sender is not complete, lacking many important features, but it can be used for some quick tests.  Make a short gcode file that triggers the problem, then send it from FluidTerm.  The command is CTRL-G.

When I use CTRL-G with a 14-line gcode file that does a lot of moves that take a long time to finish, FluidNC sends all of the lines instantly, and the motion then completes later.

Avataar120, 2023-07-01:
Hi Mitch,

Thanks to take care of this issue,

Same behavior with UGS 

[image]

MitchBradley, 2023-07-02:
Your UGS example run does not tell us much because it uses short segments that exceed the line bandwidth at the specified feed rate - see Joacim's bandwidth analysis above.  You said that it also happens on some other simpler GCode files that do not exceed the line bandwidth.  Please try UGS with them.

Any test that exceeds the line bandwidth is not interesting.

MitchBradley, 2023-07-02:
Here's another way of analyzing it.

For an on/off raster image with S0 and S200 on alternate lines, e.g. 
```
G1 X0.1S0
G1 X0.4S200
```
the number of characters in the two lines is 22, assuming there is one newline at the end of each line.

At 115200 baud, each bit is 1/115200 sec or 8.68 microseconds.  A character is 1 start bit + 8 data bits + 1 stop bit for a total of 10 bits, so 86.8 microseconds per character.  22 characters (off line plus on line) thus takes 22 * 86.8 us = 1.736 milliseconds.  350 mm/sec is .350 mm/msec, so if the laser is moving at speed, it move 1.736 msec * .350 mm/msec = .6 mm during the time it takes to send a pair of on/off lines.

If the pair of lines has a combined motion length of less than 0.6 mm, then the motion will complete before the lines have been transmitted.  If that happens enough times, then the planner will stop, waiting for the next line, and cause a deceleration.

There are alternative ways to encode raster data that use less bandwidth, but FluidNC has not yet implemented them.  It is on our long term wish list.

In the interim, the bandwidth can be reduced slightly by
* Ensuring that the file line ending is only linefeed, not carriage-return+linefeed which is what Windows likes.  That prevents the baseline character count from being 24 instead of 22
* Eliminating the "G1 " at the beginning of each line.  Once G1 has been said once, it is "sticky" and applies to subsequent commands.  Doing that reduces the length of a pair of lines by 6 bytes, so the pair can be transmitted in 16 * 86.8 us = 1.39 milliseconds.
* Changing S200 to S1 and adjusting the speed map so S1 gives you the desired beam intensity.  The eliminates another two bytes, bringing the count down to 14, so 1.22 milliseconds.

If we assume that the raster data has some alternating 0.1 mm on, 0.1 mm off sections, then the laser needs to travel less than 0.2 mm in 1.22 ms.  That works out to <164 mm/sec or 9875 mm/min.

Conclusion: you are trying to go faster than is possible with the current setup.

MitchBradley, 2023-07-02:
With a 22-character line pair, the fastest rate is 105 mm/sec = 628 mm/min, without risking stalls on alternating-pixel data.

If the file has carriage returns, that drops to 96 mm/sec or 576 mm/min

breiler, 2023-07-02:
Sounds like we came to roughly the same conclusion Mitch. I still would like to know what happens if you decrease the max speed to 10000 mm/min and the acceleration to 1000 mm/sec2.

If the problem really is the limited bandwidth, is it possible to recompile the software to use a higher baud or would that break other things in FluidNC? 
Do you know the bandwidth limit on the web serial or websocket channel? This of course depends on the signal strength and noise. But at least a rough estimate.

The gcode optimization that you suggested with filtering modal commands is a great idea which I might steal for UGS. =)

Avataar120, 2023-07-02:
Thanks for your feedbacks.
I'll try to perform more tests this afternoon.
I already recompiled FluidNC with higher baud rates but was not able to go above 220k probably due to USB cable length.

I'll keep you posted, thanks

Avataar120, 2023-07-02:
Hi,

After several tests, it appears that I'm wrong :)
When I see slow down with simple shapes (rectangles with some tabs for instance), it is due to GCODE generator that switch from G0 when performing the main shape to G1 when reaching the tab.
In this case, I think this is normal that planer has to recalculate speed & accel

For image engraving, my issue is therefore probbaly linked to USB limitations...
Thanks a lot for the analysis, and sorry for this issue.

I close it

closed 2023-07-02 by Avataar120