bdring/FluidNC #944: Problem: MSG:WARN: Low memory with latest 8/16MB build

SparkyDan555, 2023-06-24:
### Controller Board

FluidNC Pen Laser Controller (SPI)

### Machine Description

Feungsake 4040 Laser Engraver with 4W PWM Laser Module

### Configuration file

```yaml
name: "TMC2130 XY Laser"
board: "FluidNC Laser (Daniel White)"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high

  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 150
    max_travel_mm: 500
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.00
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 0
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36 
      limit_all_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 2.000
      tmc_2130:
        direction_pin: gpio.12
        step_pin: gpio.14
        cs_pin: gpio.16
        r_sense_ohms: 0.110
        run_amps: 0.5
        hold_amps: 0.25
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
    motor1:
      null_motor:

  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 150
    max_travel_mm: 410
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.00
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 0
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_all_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 2.000
      tmc_2130:
        direction_pin: gpio.26
        step_pin: gpio.25
        cs_pin: gpio.17
        r_sense_ohms: 0.110
        run_amps: 0.75
        hold_amps: 0.5
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

probe:
  pin: NO_PIN

Laser:
  pwm_hz: 5000
  output_pin: gpio.22
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 255=100.000%
  off_on_alarm: true

control:
  safety_door_pin: gpio.32:pu
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

# Board Inputs:
# Input 1: gpio.32
# Input 2: gpio.33
# Input 3: gpio.4
# Input 4: gpio.15

user_outputs:
  digital1_pin: NO_PIN
  digital1_pin: gpio.27
  digital2_pin: gpio.2
  digital3_pin: gpio.21

# Board Outputs:
# Output 1: gpio.27
# Output 2: gpio.2 (LED)
# Output 3: gpio.21

parking:
  enable: true
  axis: Y
  pullout_distance_mm: 0
  pullout_rate_mm_per_min: 250.000
  target_mpos_mm: 410
  rate_mm_per_min: 1000.00
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.8 (main-33576ba2-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine TMC2130 XY Laser]
[MSG:INFO: Board FluidNC Laser (Daniel White)]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: User Digital Output:1 on Pin:gpio.27]
[MSG:INFO: User Digital Output:2 on Pin:gpio.2]
[MSG:INFO: User Digital Output:3 on Pin:gpio.21]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:  Y Neg Limit gpio.39]
[MSG:INFO:  Y All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.32:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.22 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:########]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.2.1.89]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.6 [FluidNC v3.6.8 (main-33576ba2-dirty) (wifi+bt) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:WARN: Low memory: 14780 bytes]
[MSG:WARN: Low memory: 14660 bytes]
[MSG:WARN: Low memory: 14456 bytes]
[MSG:WARN: Low memory: 11964 bytes]
[MSG:WARN: Low memory: 9488 bytes]
[MSG:WARN: Low memory: 9288 bytes]
[MSG:ERR: sdmmc_card_init failed code 0x0x107]
[MSG:WARN: Low memory: 8124 bytes]
[MSG:WARN: Low memory: 7716 bytes]
[MSG:WARN: Low memory: 7284 bytes]
[MSG:INFO: Channel auto report interval set to 50 ms]
[MSG:WARN: Low memory: 5904 bytes]
[MSG:WARN: Low memory: 5712 bytes]
[MSG:WARN: Low memory: 5296 bytes]
```

### User Interface Software

FluidTerm

### What happened?

I have built a 16MB version of FluidNC for a 16MB ESP32-DevKit with the following lines in platform.ini:

```
board_build.partitions = FluidNC/ld/esp32/app3M_spiffs9M_16MB.csv ; For 16Meg ESP32
board_upload.maximum_size = 16777216 ; For 16Meg ESP32
board_upload.flash_size = 16MB ; For 16Meg ESP32
```
This does work but when connecting to the web interface, the console is filled with `MSG:WARN: Low memory`. I have a previous build up to commit [468fee9](https://github.com/bdring/FluidNC/commit/468fee94c191b7729e8d87817dd83dfddfc1ec0c) with the same changes in platformio.ini which does not have this issue.

MitchBradley, 2023-06-24:
The problem is likely to be related to auto-reporting.  We have some work in progress to resolve this.  In the meantime, if you get the WebUI from https://github.com/MitchBradley/ESP3D-WEBUI/blob/revamp/index.html.gz and use the preferences dialog to set the auto-report interval to 0, it might help.

SparkyDan555, 2023-06-24:
Thanks for the reply. I have tried your suggestion but sadly it doesn't seem to help. I have set the auto-report interval to 0 which makes "Auto" unselectable. I set the reporting to "None" but this setting appears to be reset on boot. 

Logs:
```
[MSG:INFO: FluidNC v3.7.1 (main-e99a8abb-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine TMC2130 XY Laser]
[MSG:INFO: Board FluidNC Laser (Daniel White)]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: User Digital Output:1 on Pin:gpio.27]
[MSG:INFO: User Digital Output:2 on Pin:gpio.2]
[MSG:INFO: User Digital Output:3 on Pin:gpio.21]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:  Y Neg Limit gpio.39]
[MSG:INFO:  Y All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.32:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.22 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:########]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.2.1.91]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.1 (main-e99a8abb-dirty) (wifi+bt) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:WARN: Low memory: 12960 bytes]
[MSG:WARN: Low memory: 12768 bytes]
[MSG:WARN: Low memory: 12640 bytes]
[MSG:WARN: Low memory: 12240 bytes]
[MSG:WARN: Low memory: 11792 bytes]
[MSG:WARN: Low memory: 11660 bytes]
[MSG:WARN: Low memory: 11532 bytes]
[MSG:WARN: Low memory: 11404 bytes]
[MSG:WARN: Low memory: 11276 bytes]
[MSG:WARN: Low memory: 11148 bytes]
[MSG:WARN: Low memory: 11020 bytes]
[MSG:WARN: Low memory: 9620 bytes]
[MSG:WARN: Low memory: 9500 bytes]
[MSG:WARN: Low memory: 9372 bytes]
[MSG:WARN: Low memory: 9252 bytes]
[MSG:WARN: Low memory: 9132 bytes]
[MSG:WARN: Low memory: 9124 bytes]
[MSG:WARN: Low memory: 9092 bytes]
[MSG:WARN: Low memory: 7580 bytes]
[MSG:WARN: Low memory: 7420 bytes]
[MSG:WARN: Low memory: 7292 bytes]
[MSG:WARN: Low memory: 7164 bytes]
[MSG:WARN: Low memory: 7036 bytes]
[MSG:WARN: Low memory: 5344 bytes]
[MSG:WARN: Low memory: 5216 bytes]
[MSG:WARN: Low memory: 5088 bytes]
[MSG:WARN: Low memory: 5004 bytes]
[MSG:WARN: Low memory: 4780 bytes]
[MSG:WARN: Low memory: 4648 bytes]
[MSG:WARN: Low memory: 4520 bytes]
[MSG:WARN: Low memory: 4392 bytes]
[MSG:WARN: Low memory: 4264 bytes]
[MSG:WARN: Low memory: 4136 bytes]
[MSG:WARN: Low memory: 3984 bytes]
[MSG:WARN: Low memory: 3860 bytes]
[MSG:WARN: Low memory: 3732 bytes]
[MSG:WARN: Low memory: 3336 bytes]
[MSG:WARN: Low memory: 3156 bytes]
[MSG:WARN: Low memory: 2896 bytes]
[MSG:WARN: Low memory: 2776 bytes]
[MSG:WARN: Low memory: 2648 bytes]
[MSG:WARN: Low memory: 2592 bytes]
[MSG:WARN: Low memory: 2584 bytes]
[MSG:WARN: Low memory: 2556 bytes]
[MSG:WARN: Low memory: 2168 bytes]
```

MitchBradley, 2023-06-24:
If you set both the status report interval and the auto-report interval to 0, None will be used.

MitchBradley, 2023-06-25:
I made a prerelease with my latest attempt to fix this problem.  Please try it out. https://github.com/bdring/FluidNC/releases/tag/v3.7.2-pre1

SparkyDan555, 2023-06-25:
Unfortunately this hasn't helped.

Logs:
```
[MSG:INFO: FluidNC v3.0.x (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine TMC2130 XY Laser]
[MSG:INFO: Board FluidNC Laser (Daniel White)]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: User Digital Output:1 on Pin:gpio.27]
[MSG:INFO: User Digital Output:2 on Pin:gpio.2]
[MSG:INFO: User Digital Output:3 on Pin:gpio.21]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.14 Dir:gpio.12 CS:gpio.16 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.25 Dir:gpio.26 CS:gpio.17 Disable:NO_PIN Index:-1 R:0.110]
[MSG:INFO:  Y Neg Limit gpio.39]
[MSG:INFO:  Y All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.32:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.22 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:########]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.2.1.91]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.0 [FluidNC v3.0.x (noGit) (wifi+bt) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:WARN: Low memory: 14992 bytes]
[MSG:WARN: Low memory: 14744 bytes]
[MSG:WARN: Low memory: 12268 bytes]
[MSG:WARN: Low memory: 12068 bytes]
[MSG:WARN: Low memory: 11940 bytes]
[MSG:WARN: Low memory: 11880 bytes]
[MSG:WARN: Low memory: 11552 bytes]
[MSG:WARN: Low memory: 11504 bytes]
[MSG:WARN: Low memory: 11376 bytes]
[MSG:WARN: Low memory: 11248 bytes]
[MSG:WARN: Low memory: 11120 bytes]
[MSG:WARN: Low memory: 11072 bytes]
[MSG:WARN: Low memory: 10996 bytes]
[MSG:WARN: Low memory: 10864 bytes]
[MSG:WARN: Low memory: 10736 bytes]
[MSG:WARN: Low memory: 10608 bytes]
[MSG:WARN: Low memory: 10420 bytes]
[MSG:WARN: Low memory: 10292 bytes]
[MSG:WARN: Low memory: 10240 bytes]
[MSG:WARN: Low memory: 9876 bytes]
[MSG:WARN: Low memory: 8692 bytes]
[MSG:WARN: Low memory: 8680 bytes]
[MSG:INFO: Channel auto report interval set to 50 ms]
[MSG:WARN: Low memory: 7532 bytes]
[MSG:WARN: Low memory: 7372 bytes]
[MSG:WARN: Low memory: 7220 bytes]
[MSG:WARN: Low memory: 6200 bytes]
[MSG:WARN: Low memory: 6008 bytes]
```

MitchBradley, 2023-06-26:
Try removing the SD card to see if the memory that is used to read it could be related.

Also try issuing the $Heap command to see how much memory is available after things settle down.

MitchBradley, 2023-06-26:
And set $message/level=debug in case that shows something interesting.

MitchBradley, 2023-06-27:
The warning message was added in 3.7.1.  Whatever is using the memory has probably not changed; the difference is that it is now being reported while previously it went unnoticed.

I am looking for what could be using the memory.  This is a tricky problem because the likely culprit is something deep inside the network stack, beneath two layers of third-party code.  It is also likely to depend on details of what else is happening on the wifi network, thus very difficult to replicate in another environment or to reproduce at will.

MitchBradley, 2023-06-28:
I think I have a way to repro the problem.  If I flood-ping the FluidNC controller with ` sudo ping -l 96 -s 1420 -f 192.168.68.139` from a Linux system, the Low memory warning happens immediately, dropping to below 3K in less than one second, and hanging the ESP32 network stack so it stops responding to pings.

I have tried variations of that command with different preload and packet sizes, leading me to think that the network stack is being presented with a lot of packets in a short time.  It has to allocate memory for each until it is dealt with, then the memory can be returned back to the heap.  $HEAP shows that, indeed, the memory is being properly returned, but there is an interval when a lot of memory is being used up by network packets that have not yet been processed.

A similar thing could happen on a network that is not being subjected to a ping flood.  There are some network services that use broadcasting, where every node must receive the packet and do something with it. Other services or agents might do address-scanning, periodically sending an inquiry to every possible address.  I know that some mesh routers do something like that; in fact we had to add a filter in the FluidNC HTTP server to detect and discard inquiries from Google Mesh routers which use them to automatically discover when a new router has been added to the network.

"Fixing" this is going to be very difficult, since it is happening deep inside the SDK code which we normally try not to touch - because modifying it requires a lot of work to re-plumb things and creates a difficult maintenance problem going forward.

The conditions on the WiFi network are not under our control. We can't stop another computer from sending arbitrary amounts of random stuff that the ESP32 network stack will have to look at before it can release the memory it used to hold the incoming packets.

We can make the "WARN" message go away; for example we could move it to DEBUG level so most people will not see it.  But that could just hide an underlying problem.  If the memory gets too low, even temporarily, FluidNC might crash or hang, without any warning or indication why.

One mitigation might be to put FluidNC behind an extra router that would filter traffic so FluidNC does not have to see all the chitchat that is happening on the main network.

MitchBradley, 2023-06-28:
Also note that this test was performed without WebUI.  The starting point was the wifi version immediately after reboot, with no network connections active, just FluidTerm monitoring the serial port.  WebUI will probably make it die a bit sooner due to the modest amount of memory that is necessary to maintain the WebUI websocket connection - but if you push the ESP32 network stack hard enough, it will run out of memory with or without WebUI.

The same thing happens with not only 3.7.1, but also some development versions that try to be extra-careful about memory use and flow control for websockets.  It boils down to "if the wifi network presents enough packets in a short time to the underlying ESP32 network stack, it will allocate memory for them until it runs out".

MitchBradley, 2023-06-29:
I found a way to configure the memory usage of the WiFi module.  Try building from the latest version of the WebSocketsNoStall branch https://github.com/bdring/FluidNC/tree/WebSocketsNoStall .  Use $heap periodically to monitor heap usage.  Hopefully the low-water mark will stay above the warning threshold.

SparkyDan555, 2023-06-29:
Thanks for the reply and apologies for the delayed response. I have tried the WebSocketsNoStall branch but unfortunately cannot report good news. I still get the warning messages and the captive portal _'File index.html. gz is missing, please upload it'_ page is unresponsive at times. When I try to upload the files, the upload freezes and the log reports:

```
[MSG:WARN: Low memory: 720 bytes]
[MSG:INFO: Upload cancelled]
[MSG:WARN: Low memory: 620 bytes]
```

This is with the wifibt build (which is the reason I am using a 16 MB board). So I tried just a wifi build and the problem is actually worse. The captive portal page barely loads at all, although no warning messages are seen in the console? Neither build was stable enough to upload index.html.gz and reach the main home page.

MitchBradley, 2023-06-29:
I am not sure that we can ever stabilize the wifibt build because BT uses so much memory.  Doubling the FLASH does not help with RAM.

Can you upload the index.html.gz with FluidTerm?

SparkyDan555, 2023-06-29:
I can upload it via FluidTerm but still can't access the main page due to stability.

SparkyDan555, 2023-06-29:
Also, just to note that the wifibt build FluidNC v3.6.8 (main-33576ba2) (wifi+bt) is responsive and stable.

MitchBradley, 2023-06-30:
The most invasive change that occurred since 3.6.8 was the change from using Arduino String to C++ std::string . To see if that change triggered your problem, please test with builds from #6444b46 (before)  and #ddb5628a (after).

MitchBradley, 2023-07-04:
The precompiled Bluetooth library in the Arduino Espressif32 framework uses a *lot* of RAM.  The WebSocketsNoStall branch has code to release the Bluetooth RAM if $Bluetooth/Enable=Off .

https://github.com/MitchBradley/BTLib is a setup to compile an alternative version of libbt.a that omits unnecessary features like BLE, audio, and whatnot, thus saving a significant amount of RAM even when Bluetooth is enabled.

I spent more that two full days working on this, benefiting no other users that I know of.  Please consider becoming a sponsor.

SparkyDan555, 2023-07-04:
The change from using Arduino String to C++ std::string doesn't appear to impact **MSG:WARN: Low memory:**

### HEAD is now at c444b463... Merge branch 'Devt' into DeathToStrings:
```
Grbl 3.6 [FluidNC v3.6.7 (HEAD-c444b463-dirty) (wifi+bt) '$' for help]
[MSG:ERR: '$H'|'$X' to unlock]
[MSG:INFO: WiFi Disconnected]
```
Cannot connect to WebUI

### HEAD is now at ddb5628a... Start timers after creation, with error messages:
```
Grbl 3.6 [FluidNC v3.6.7 (HEAD-ddb5628a-dirty) (wifi+bt) '$' for help]
[MSG:ERR: '$H'|'$X' to unlock]
[MSG:ERR: mount_to_vfs failed code 0x0x101]
[MSG:ERR: mount_to_vfs failed code 0x0x101]
[MSG:ERR: mount_to_vfs failed code 0x0x101]
```
Can connect to WebUI

The latest **WebSocketsNoStall** branch still gives warnings:
```
Grbl 3 [FluidNC v3.7,2-pre3 (WebSocketsNoStall-9da58e98-dirty) (wifi+bt) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:WARN: Low memory: 5816 bytes]
```
and with the new libbt.a...

```
Grbl 3 [FluidNC v3.7,2-pre3 (WebSocketsNoStall-9da58e98-dirty) (wifi+bt) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
[MSG:WARN: Low memory: 5804 bytes]
```

MitchBradley, 2023-07-04:
Do you have both bluetooth and wifi enabled at the same time?  You need to turn one of them off, say with $bluetooth/enable=off or $wifi/mode=off

SparkyDan555, 2023-07-06:
I have made sure that bluetooth is disabled which helped with the WebSocketsNoStall build but not the main build, I still see warning messages.

MitchBradley, 2023-07-06:
There is no point testing the main build, as it does not have any of the numerous fixes that I have done recently.  Do you still see warning messages with the WebSocketsNoStall build?  Your message above was unclear.

SparkyDan555, 2023-07-06:
No warning messages with WebSocketsNoStall after $bluetooth/enable=off

closed 2024-12-12 by bdring