bdring/FluidNC #952: Problem:

kami0011, 2023-06-30:
### Controller Board

6-pack

### Machine Description

 ESP32 Dev V1 Board 30 pin. 6 axis cnc card

using 6 axis card . 

### Input Circuits

```yaml
Iputs are xyzabc home limits and overtravel limits
  safety_door_pin: gpio.34:low
  reset_pin: gpio.35:low
  feed_hold_pin: gpio.36:low
  cycle_start_pin: gpio.39:low
```

### Configuration file

```yaml
name: 6 Axis_Board_For_Pioneer
board: 6-pack

kinematics:

stepping:
  engine: I2S_stream
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.18:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.0
        step_pin: i2so.1
        disable_pin: i2so.13
    motor1:
      null_motor:
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0.000
      positive_direction: true
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.19:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.2
        step_pin: i2so.3
        disable_pin: i2so.14
    motor1:
      null_motor:
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 0.000
      positive_direction: false
      settle_ms: 250.000
      seek_mm_per_min: 2000.000
      feed_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 5.000
    motor0:
      limit_all_pin: gpio.21:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.15
    motor1:
      null_motor:
  a:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:
  b:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:
  c:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.27
  data_pin: gpio.25
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.34:low
  reset_pin: gpio.35:low
  feed_hold_pin: gpio.36:low
  cycle_start_pin: gpio.39:low

coolant:
  delay_ms: 1000.000

probe:
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:
  digital1_pin: gpio.4
  digital2_pin: gpio.17
  digital3_pin: gpio.16

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
starts fine but seems like FluidNC default works on  ESP32 Dev V4 38 pin.

My card is deisgned with ESP32 Dev V1 30 pin.
How can i run FluidNC on this ESP Dev V1 30 pin as i cant change my hardware.

Please help me resolve this problem. could it be th reason of card not working. 
Right now i am testing on card as card is not connected in machine and is on bench.
on GRBL ESP32 old firmware it runs M62 P1 and out put LED lightsup.
But on new FluidNC firm that doesnot work and gives an error
```

### User Interface Software

WebUI

### What happened?

Hi,
have installed FluidNC on ESP32 Dev V1 Board 30 pin.
My cnc card is designed on this one and i am upgrading from GRBL ESP32 to FluidNC
I have edited YAML file .
But I am confused if the FluidNC is by default designed for ESP32 Dev Controller V4 or could it be used on  ESP32 Dev V1.
If yes then how can I use it.?
Please suggest ,
Best Regards,

bdring, 2023-06-30:
Edit your post to add the startup messages from FluidTerm

MitchBradley, 2023-06-30:
http://wiki.fluidnc.com/en/support/requesting_help#fluidnc-startup-messages

kami0011, 2023-06-30:
thanks for your message. I am really sorry to bother you again and again. I am new to this all as i have already got a board and GRBL ESP32 programmed by my friend . all i am trying to upgrade to fluid.
Please tell me if i can use it for ESP32 Dev V1 or not .
i read the link you send me but sorry couldn't understand and couldn't get the answer .
Thanks a lot

kami0011, 2023-06-30:
and if yes then how what do i have to do before compiling to use ESP32 Dev v1 board

bdring, 2023-06-30:
You should should install FluidNC yourself following the wiki. We cannot explain any better than the wiki.

http://wiki.fluidnc.com/en/installation#using-pre-compiled-files

MitchBradley, 2023-06-30:
Maybe your friend can help.

kami0011, 2023-06-30:
thanks alot for your time

MitchBradley, 2023-06-30:
It should work.  There is no need to compile.  You just need to learn about FluidNC.

kami0011, 2023-06-30:
thank you for all your help
great work

closed 2023-06-30 by MitchBradley