bdring/FluidNC #953: Problem: Slow engraving speed when running 300dpi with lightburn only with FluidNC firmware

phoenixfire425, 2023-07-05:
### Wiki Search Terms

Slow engraving speed when running 300dpi with lightburn

### Controller Board

custom MKS board with ESP32-U2

### Machine Description

ESP32-U2 on a custom board made by MKS
has two TMC2209 (not running in UART)

[image]

### Configuration file

```yaml
name: "P2 33W"
board: "ESP32 TTAN LTS"
meta: Corey McFarland 06-12-2023

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.33:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 40000
    acceleration_mm_per_sec2: 3000
    max_travel_mm: 400
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    
    motor0:
      limit_all_pin: gpio.36:low
      stepstick:
        direction_pin: gpio.26:high
        step_pin: gpio.25
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 40000
    acceleration_mm_per_sec2: 3000
    max_travel_mm: 400
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_all_pin: gpio.39:low
      stepstick:
        direction_pin: gpio.5:low
        step_pin: gpio.27
    motor1:
      null_motor:
  

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.35
  
control:
  safety_door_pin: gpio.23:low
  #gpio.23:low <- flame
  #reset_pin: NO_PIN
  feed_hold_pin: gpio.22:low
  #cycle_start_pin: NO_PIN
  #macro0_pin: NO_PIN 
  #gpio.22 <- tilt sensor pin
  #macro1_pin: NO_PIN
  #macro2_pin: NO_PIN
  #macro3_pin: NO_PIN
  
Laser:
  pwm_hz: 10000
  output_pin: gpio.32
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  #Beeper connector. Command: "M62 P0" turn beeper On. 
  #"G4 P0.5" wait for 0.5s. "M63 P0" turn beeper OFF
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  
start:
  must_home: false
  deactivate_parking: false
  check_limits: false
  
coolant:
  flood_pin: gpio.4
  #mist_pin: NO_PIN
  delay_ms: 0
```

### Startup Messages

```text
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine P2 33W]
[MSG:INFO: Board ESP32 TTAN LTS]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:gpio.35 freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.33]
[MSG:INFO: Axis X (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.26 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.36:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,400.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.5:low Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: safety_door_pin gpio.23:low]
[MSG:INFO: feed_hold_pin gpio.22:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.32 Freq:10000Hz Period:4095]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant gpio.4]
[MSG:INFO: Connecting to STA SSID:Far-Lands]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.4.51]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.7 [FluidNC v3.7.1 (wifi) '$' for help]
ok
[VER:3.7 FluidNC v3.7.1:]
[OPT:PHS]
[MSG: Machine: P2 33W]
[MSG: Mode=STA:SSID=Far-Lands:Status=Connected:IP=192.168.4.51:MAC=0C-8B-95-F0-09-90]
```

### User Interface Software

LightBurn

### What happened?

Before upgrading to fluidNC I was able to run a 24000 mm/min 300 dpi file pretty easly and fast.  (about 50mmx50mm).  I just tested this with fluidNC and it seems that the laser drymatically slows down when I am doing an image on stucki at 300 dpi.  is this a buffer issue?  no over scanning.  I have also tried running a native dpi image 212 and its still the same.

TLDR:  same gcode,  
FluidNC runs slow.  
Factory firmware or GRBL_ESP32 runs fast as normal.

FluidNC the same gcode took 11 minutes to complete.  before fluidNC it was <5 minutes.

### GCode File

[lightburnFile_P2_FluidNC_Issue.lbrn2.txt](https://github.com/bdring/FluidNC/files/11952933/lightburnFile_P2_FluidNC_Issue.lbrn2.txt)

[gcode_P2_FluidNC_Issue.lbrn2.nc.txt]
(https://github.com/bdring/FluidNC/files/11952935/gcode_P2_FluidNC_Issue.lbrn2.nc.txt)

### Other Information

https://discord.com/channels/780079161460916227/1120886677641105538/1120887946057359441

MitchBradley, 2023-07-05:
The problem is caused by the fix to #486 .  That fix should have applied to M3 mode only, but instead it affects M4 mode too.  The transition between G0 and G1 modes causes a planner synchronization which breaks the smooth motion transition.  I am working on a fix.

MitchBradley, 2023-07-05:
Try this pre-release https://github.com/bdring/FluidNC/releases/tag/v3.7.2-pre3

closed 2023-07-12 by bdring

distebia, 2023-08-28:
@phoenixfire425 Salve, ho una mks dlc32 e anche io riscontro problemi di velocità eseguendo una foto. ho aggiornato alla 3.7.6 ma i problemi persistono. vorrei provare il firmware originale ma non riesco a trovare quello che hai indicato tu, ossia Firmware: V4.00(8M.H35.20230104). mi potresi dare un link?

commit 19ed9a23 (2023-07-05): Fix #953 - inappropriate planner syncs in laser mode