bdring/FluidNC #957: Problem: fluidnc does not work with mastercam gcode

angelmoi, 2023-07-08:
### Wiki Search Terms

With other programs, the firmware fluidnc works perfectly eg. vectric, flatcam, eaglepcb

### Controller Board

3 motors usig on a cnc router 

### Machine Description

Gantry router with three steper motors and spindle

### Input Circuits

```yaml
N/A
```

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 638
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 800
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: true
    
    motor0:
      limit_all_pin: gpio.17:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 638
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 800
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 632
    max_rate_mm_per_min: 600
    acceleration_mm_per_sec2: 200
    max_travel_mm: 200
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.16:low:pu
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 10000

i2c0:
   sda_pin: gpio.21
   scl_pin: gpio.22
        
oled:
   i2c_num: 0
   i2c_address: 60
   width: 128
   height: 64
   radio_delay_ms: 1000
        
probe:
  pin: gpio.32:low:pu
  
#control:
  #feed_hold_pin: gpio.34:low:pu

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 10000=100%
```

### Startup Messages

```text
[MSG:ERR: 20 (Unsupported GCode command) in /sd/T.NC at line 2]
[MSG:ERR: 24 (Gcode axis command conflict) in /sd/T.NC at line 9]
```

### User Interface Software

Fluidterm

### What happened?

I was tryied to use mastercam gcode but fluidnc show several problems with the code

### GCode File

%
O0000(T)
(DATE=DD-MM-YY - 08-07-23 TIME=HH:MM - 11:29)
(MCX FILE - T)
(NC FILE - C:\USERS\ALS\DESKTOP\T.NC)
(MATERIAL - ALUMINUM MM - 2024)
( T215 |    6. FLAT ENDMILL | H215 )
N100 G21
N102 G0 G17 G40 G49 G80 G90
N104 T215 M6
N106 G0 G90 G54 X10.513 Y0. A0. S16000 M3
N108 G43 H215 Z10.
N110 G1 Z1. F79.5
N112 X11. Z.974
N114 X5. Z.66
N116 X11. Z.346
N118 X5. Z.031
N120 X11. Z-.283
N122 X5. Z-.598
N124 X11. Z-.912
N126 X5. Z-1.227
N128 X11. Z-1.541
N130 X5. Z-1.856
N132 X11. Z-2.17
N134 X5. Z-2.484
N136 X11. Z-2.799
N138 X5. Z-3.113
N140 X11. Z-3.428
N142 X5. Z-3.742
N144 X11. Z-4.057
N146 X5. Z-4.371
N148 X11. Z-4.686
N150 X5. Z-5.
N152 X.563 F1000.
N154 G3 X-1.125 Y1.688 I-1.688 J0.
N156 X-2.813 Y0. I0. J-1.688
N158 X1.125 Y-3.938 I3.938 J0.
N160 X5.063 Y0. I0. J3.938
N162 X-1.125 Y6.188 I-6.188 J0.
N164 X-7.313 Y0. I0. J-6.188
N166 X1.125 Y-8.438 I8.438 J0.
N168 X9.563 Y0. I0. J8.438
N170 X-1.125 Y10.688 I-10.688 J0.
N172 X-11.813 Y0. I0. J-10.688
N174 X-9.828 Y-6.885 I12.938 J0.
N176 X0. Y-12. I9.828 J6.885
N178 X12. Y0. I0. J12.
N180 X0. Y12. I-12. J0.
N182 X-12. Y0. I0. J-12.
N184 X-9.828 Y-6.885 I12. J0.
N186 G0 Z10.
N188 M5
N190 G91 G28 Z0.
N192 G28 X0. Y0. A0.
N194 M30
%

### Other Information

N/A

MitchBradley, 2023-07-08:
This is not a problem with FluidNC.  It is a problem with the Mastercam post processor that you are using.  A post processor is a program that generates GCode that is acceptable for a particular type of machine controller.  The GCode language has different dialects. Some commands, for example the O0000 command in your example, are not implemented on all controllers. Sometimes a command will mean one thing on one controller and something else on a different controller.  To use Mastercam, you will need to select a post processor that is compatible with the GRBL dialect of GCode.

I did a web search on your behalf, searching for "mastercam grbl post processor".  You should do the same search.  What I found was that there is no strict GRBL post processor for mastercam, but some people think that the Fanuc post processor is close enough that it might work.

I am closing this issue.  Any further investigation should be done on mastercam forums.

closed 2023-07-08 by MitchBradley

angelmoi, 2023-07-08:
Ok Mitch, thanks a lot for your opportune and assertive answer