bdring/FluidNC #959: Problem: Reboot caused by Guru Meditation Error?

ianwestwood, 2023-07-11:
### Wiki Search Terms

I searched for 'errors' and 'reboots' here http://wiki.fluidnc.com/en/support/troubleshooting_config_files and am asking for help & advice please.

### Controller Board

'FluidNC Pen Laser Controller TMC2209' purchased from [tindie](https://www.tindie.com/products/33366583/fluidnc-penlaser-cnc-controller-tmc2209/) & delivered last week along with  the 'ESP32 W/PCB Antenna + FluidNC Programming' option. 

### Machine Description

I'm using the controller for a ZenXY V2, designed by the folks over at V1 engineering.  Build details here https://docs.v1e.com/zenxy/.  

Currently running on a bench PSU at 12V.

Photos of setup here:

[image]

[image]

[image]

[image]

[image]

### Input Circuits

```yaml
I don't think this is limit switch related, but for completion I'm using optical switches connected to gpio 36 & 39 (yellow wires are 'sig' in the photo). 

They appear to be working as expected and it homes OK.
```

### Configuration file

```yaml
board: FluidNC Pen Laser TMC2209
name: V1 Engineering ZenXY V2 hello world
meta: Optical end stops enabled

kinematics:
  corexy:

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0
  
uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

start:
  must_home: false
[... 205 lines cut ...]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:ShaManorHouse]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.86.27]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

This is a screenshot from the FluidNC UI after crash:

[image]

MitchBradley, 2023-07-11:
Try the 3.7.2-pre3 version

ianwestwood, 2023-07-12:
Hi Mitch, thank you for taking the time to look into this for me.... it is much appreciated.  I have been playing with 3.7.2-pre3 and have some (hopefully helpful) observations, comments & questions to share.  

If you find it helpful, I can leave my ZenXY test rig set up until we are done chatting so I can try different things for you.
 Please let me know.

Apologies for any use of incorrect terminology - happy to clarify anything I have said here (I'm not a coder, but some how mostly manage to flail around with decent success in many of my projects)

_The good (0)_: As you say above, auto polling mostly causes crashes. For me, 50ms never completes on anything more complex than straight line 'wipe' patterns, Since all my kit is new I guess I can discount bad SD card contacts.

I don't have a quality response to the computer sleep question.  I was mostly working in other tabs and doing other things in other apps during this process, so I imagine most scenarios were met at some point during this time.  

Having said that, once I set Poll to '3 sec' it ran fine over a period of a couple hours and ran 8 jobs back to back (the final one this morning after sitting overnight), and the random pauses did not occur.

Fluidterm returned:

> Grbl 3 [FluidNC v3.7,2-pre3 (wifi) '$' for help]
> [MSG:INFO: Channel auto report interval set to 50 ms]
> [MSG:INFO: Channel auto reporting turned off]
> [MSG: /sd/linear wipe 45 F5000.gcode file job succeeded]
> [MSG: /sd/linear wipe neg 45 F5000.gcode file job succeeded]
> [MSG: /sd/Clover F5000.gcode file job succeeded]
> [MSG: /sd/Rose polygon F5000.gcode file job succeeded]
> [MSG: /sd/linear wipe 0 F5000.gcode file job succeeded]
> [MSG: /sd/linear wipe 90 F5000.gcode file job succeeded]
> [MSG: /sd/linear wipe 0 F5000.gcode file job succeeded]
> [MSG: /sd/linear wipe neg 45 F5000.gcode file job succeeded]

_The Bad (1)_: It appears this release breaks SD card uploading via the fluidNC UI. It feels like this causes some kind of crash that is not reported in the UI or via fluidterm.  On the surface everything looks OK but the UI becomes kind of 50% responsive 50% unresponsive and everything seems to get into a weird state that I can't quite articulate.

_The Ugly (2)_: Once in this state, it's actually very difficult to get back into a working condition.  Power cycling doesn't seem to be the best way, because fluidterm either looks good displaying a healthy startup or just stops displaying altogether.  In this state, very often, the end stops fail on pull_off.  Restart with the UI doesn't always help.  It *feels like* some bogus version of the config file has been loaded.  Sometimes (but not always!!!), resting current draw rises to about 1.5A making the TMC drivers warm up.

But closing the FluidUI tab, closing fluidterm, pressing EN on the ESP32, starting fluidterm, opening the UI and then restarting via the UI (so you can see fresh data in fluidterm), does seem to give the best & most consistent path to a good working state..... which is how I came to run 8 successful jobs back to back!  Lets call this a 'session'

_The Ugly (3)_: Any type of halt, crash, alarm, pause or whatever basically hoses that 'session'.  For example, if I exceed _max_travel_mm_ when _soft_limits: true_, I get a warning that I am unable (or more accurately, can't for the life of me figure out how) to recover from, and I have to reset the session.

*Question (4)* Is there a relationship between homing _pulloff_mm_ and _max_travel_mm_?  I would love some clarification because running 

> $HY
> $HX
> G90
> G92 Y0 X0
> G00 X0.0000 Y0.0000

at the beginning of my gcode doesn't seem to actaully set home to Y0 X0.  My current work around is to increase _max_travel_mm_ by 3mm to prevent the alarm from happening.

*Question (5)* How do I use _Config items_ in FluidNC Settings?  I would have thought they would be for quick tweaks to the config when you are dialing in a machine or going through a difficult process you didn't gcode properly for!  But you can't change any values during a job because *Error 500 :Error: Another interface is busy*.  But because of _The Ugly (1)_ you have to bail out of the session, so you may as well just edit config.yaml.  What am I missing?

*Question (6)* Playlists.  Is it possible to implement gcode playlists?  This would be perfect for artsy projects like this zen table as you could just upload gcode and have it 'play' in some kind of order.  Could we finagle a macro for this purpose?

*Question (7)* Incidentally, previous to testing this release, I noticed you couldn't upload to the SD card while a job is running.  Is this a limitation of the ESP32?  Per *Question (6)*, if we implemented gPlaylists it would be awesome if you could slip a gcode into the playlist of a busy machine and reorder them!!!  

Sorry about the scope creepy questions!!!  Ha ha.

Anyways, hope this makes sense, is constructive, helpful.... and I look forward to hearing from and testing for you soon.

ianwestwood, 2023-07-12:
Incidentally, here is a picture of the completed rose polygon FYI:
[image]

MitchBradley, 2023-07-12:
I fixed the SD card problem yesterday.

ianwestwood, 2023-07-12:
Agreed, that's why I numbered them!  I'll bear that in mind for the future.

MitchBradley, 2023-07-12:
> Question (4)

The pulloff is supposed to be outside the travel envelope.  So after homing, the machine position is supposed to be at mpos_mm, and from there you can travel max_travel_mm in the direction implied by positive_direction without triggering a soft limit.

The limits are in machine coordinates.  When you say G92 X0 Y0, that sets the work coordinate position.  If your work 0 and machine 0 are different, then a motion to max_travel_mm in work coordinates might exceed the limit which is in machine coordinates.

Homing inside a GCode file is not recommended, or more precisely it is not the way things are commonly done, so we do not spend any time thinking about how it should work / making sure it works.  The common use case for CNC machines is to perform homing as a separate step outside of any GCode file.  Then establish a work offset that makes sense for the job.  Typically that work offset is referenced to a place on the stock.  Obviously, that doesn't exactly capture the sand table use case.  CNC grew up in the milling world and art machines are an afterthought.

Furthermore, G92 is not considered best practice anymore.  Best practice is to set the offset from machine zero to work zero in the G54 (or G55 or ...) coordinate system, then choose e.g. G54 in the GCode file.

MitchBradley, 2023-07-12:
> Question (5) 
Tweaking the config while running is definitely not supported.  The config is supposed to be "set it once and leave it".  You can change some config items on the fly while in idle state, for the purpose of tuning things like acceleration, speed, steps_per_mm, etc, but then you are expected to put the resulting values in the config file and be done with it.

Making this stuff tunable while running would be a can of worms of the worst kind.

MitchBradley, 2023-07-12:
6. Playlists are not implemented.  I have some experimental code that might eventually lead to making them possible, but I have been so swamped fixing bugs in the wifi stack, many of which were introduced by an upgrade to a new version of the Arduino framework, that the experimental code has languished.  And there is another problem.  For playlists to work well, you would need to be able to have multiple SD files open at once.  Each open SD file chews up a lot of RAM for buffers, which risks crashing the system because the ESP32 does not have an infinite amount of RAM.  And as soon as we enable such a feature, you can guarantee that someone will try to use it to several levels, and then complain when the machine crashes.

MitchBradley, 2023-07-12:
7. Uploading while running a job would be an excellent way to mess up your job.  The first problem is that it would use up extra RAM for the upload buffers (see 6).  The second problem is that it would take away cycles from the hard-real-time task of running the job.

This is not a big computer with multi-gigahertz 16-core CPU and many Gigabytes of RAM.  It is a tiny MCU that costs less than $10 and barely has enough resources for the stuff we are coaxing it to do.

MitchBradley, 2023-07-12:
Regarding "it would be awesome if ..." - it would be awesome if I had half the time necessary to keep this software running well considering all the stuff it does already.  Also it would be awesome if even a small fraction of the users sponsored the project.

ianwestwood, 2023-07-12:
> Furthermore, G92 is not considered best practice anymore. Best practice is to set the offset from machine zero to work zero in the G54 (or G55 or ...) coordinate system, then choose e.g. G54 in the GCode file.

Did some reading.  Interesting stuff, thanks for the pointers.

ianwestwood, 2023-07-12:
> I fixed the SD card problem yesterday.

Let me know if you need someone to test!  Happy to support you.

ianwestwood, 2023-07-12:
> Also it would be awesome if even a small fraction of the users sponsored the project.

Ack

MitchBradley, 2023-07-12:
A new release just dropped.  https://github.com/bdring/FluidNC/releases/tag/v3.7.3

ianwestwood, 2023-07-12:
> A new release just dropped. https://github.com/bdring/FluidNC/releases/tag/v3.7.3

I'll give it a whirl today.  Thanks for sharing.

ianwestwood, 2023-07-12:
Quick question if I may....

I ran install-wifi.bat, but it looks like wifi is turned off.  Am I terrible at following instructions?  

Fluidterm says:

> Grbl 3.7 [FluidNC v3.7.3 (wifi) '$' for help]
>
> ets Jul 29 2019 12:21:46
> 
> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1184
> load:0x40078000,len:13220
> ho 0 tail 12 room 4
> load:0x40080400,len:3028
> entry 0x400805e4
> 
> [MSG:INFO: FluidNC v3.7.3]
> [MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
> [MSG:INFO: Local filesystem type is littlefs]
> [MSG:INFO: Configuration file:TMC2209_ZenXY_V1.yaml]
> [MSG:INFO: Machine V1 Engineering ZenXY V2 hello world]
> [MSG:INFO: Board FluidNC Pen Laser TMC2209]
> [MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:115200]
> [MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
> [MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
> [MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
> [MSG:INFO: Axis count 3]
> [MSG:INFO: Shared stepper disable gpio.13]
> [MSG:INFO: Axis X (0.000,653.000)]
> [MSG:INFO:   Motor0]
> [MSG:INFO:     tmc_2209 UART1 Addr:0 Step:gpio.14 Dir:gpio.12 Disable:NO_PIN R:0.110]
> [MSG:INFO:  X Pos Limit gpio.39]
> [MSG:INFO:   Motor1]
> [MSG:INFO: Axis Y (0.000,438.000)]
> [MSG:INFO:   Motor0]
> [MSG:INFO:     tmc_2209 UART1 Addr:1 Step:gpio.25 Dir:gpio.26 Disable:NO_PIN R:0.110]
> [MSG:INFO:  Y Pos Limit gpio.36]
> [MSG:INFO:   Motor1]
> [MSG:INFO: Axis Z (-1000.000,0.000)]
> [MSG:INFO:   Motor0]
> [MSG:INFO: X Axis driver test passed]
> [MSG:INFO: Y Axis driver test passed]
> [MSG:INFO: Kinematic system: CoreXY]
> [MSG:INFO: Using spindle NoSpindle]
> [MSG:INFO: WiFi off]
> 
> Grbl 3.7 [FluidNC v3.7.3 (wifi) '$' for help]

MitchBradley, 2023-07-12:
read the Commands and Settings page of the wiki .  Search that page for wifi.

MitchBradley, 2023-07-12:
Use $wifi/mode, $sta/ssid, and $sta/password directly.

ianwestwood, 2023-07-12:
Yes, I tried many wifi related commands, but they didn't work after running install-wifi.bat for 3.7.3.  Should I be doing a complete erase and reinstall?

I ran the following into fluidterm:

> $ss
> [MSG:INFO: FluidNC v3.7.3]
> (edited down for clarity)
> $wifi/mode
> error:3
> $sta/ssid
> error:3
> $sta/password
> error:3
> $sta
> <Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>

Then rolled back to pre3 and got:

> $ss
> [MSG:INFO: FluidNC v3.7,2-pre3]
> (edited down for clarity)
> $wifi/mode
> $WiFi/Mode=STA>AP
> ok
> $sta/ssid
> $Sta/SSID=XXXXXXXXXX
> ok
> $sta/passowrd
> error:3
> $sta
> $Start/Message=Grbl \V [FluidNC \B (\R) \H]
> $Report/Status=1
> $Sta/SSID=XXXXXXXXXX
> $Sta/Password=******
> $Sta/MinSecurity=WPA2-PSK
> $Sta/IPMode=DHCP
> $Sta/IP=0.0.0.0
> $Sta/Gateway=0.0.0.0
> $Sta/Netmask=0.0.0.0
> ok

bdring, 2023-07-12:
It works for me....

```
$wifi/mode=sta>ap
ok
Resetting MCU  ; me sending CTRL+R in FluidTerm

[MSG:INFO: FluidNC v3.7.2]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:6P_ss_XYZ.yaml]
[MSG:INFO: Machine 6 Pack StepStick XYZ]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
```

ianwestwood, 2023-07-12:
Thanks for replying dbring!  And thanks for making the PCB's for people like me!

I had a clean install of v3.7,2-pre3 previously and only ran install-wifi.bat from fluidnc-v3.7.3-win64.zip to get my results above... Is this the correct way, or did I miss some steps?

bdring, 2023-07-12:
You may need a new copy of the WebUI.

Did you see the release notes?

https://github.com/bdring/FluidNC/releases/tag/v3.7.3

I am not sure that will solve your problem, but you should do that.

ianwestwood, 2023-07-12:
Neat, I didn't realize I could do that with fluidterm!  I had read the release notes, but without wifi I thought I was stuck.

Looks like I successfully updated the WebUI:

> $Xmodem/Receive=index.html.gz
> [MSG:INFO: Received 116668 bytes to file /littlefs/index.html.gz]

but the wifi still shows off in the start up messages and gives errors such as:

> $wifi/mode
> error:3

When I try to do anything wifi related.

bdring, 2023-07-13:
Is your SSID really `XXXXXXXXXX` ?

Here is my `$sta`  and `$I`

```
$sta
$Start/Message=Grbl \V [FluidNC \B (\R) \H]
$Report/Status=2
$Sta/SSID=Barts-WLAN
$Sta/Password=******
$Sta/MinSecurity=WPA2-PSK
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
ok
$I
[VER:3.7 FluidNC v3.7.2:]
[OPT:PHS]
[MSG: Machine: 6 Pack StepStick XYZ]
[MSG: Mode=STA:SSID=Barts-WLAN:Status=Connected:IP=192.168.1.33:MAC=30-AE-A4-1B-EF-B8]
ok
```

ianwestwood, 2023-07-13:
> Is your SSID really `XXXXXXXXXX` ?

He he, no.  I try not to publicize things I don't have to!

> Here is my `$sta` 

Cool, looks pretty much like mine on v3.7,2-pre3.

I'm sure Mitch will chime in when he is able, but if I were a betting man I'd wager there are some missing files or references brought on as a result of my path from v3.7,2-pre3  to v3.7.3.

Time will tell!

On another note, I imagine doing a project like this has its ups and downs, especially when people are $hitty & needy.  

I'd like to take the opportunity, while I have the attention of Mitch and yourself, to say many of us out there appreciate the time & effort you donate.  

I hope that by being here on this thread, I can help work through some things with you guys and hopefully make things less $hitty as a result.

MitchBradley, 2023-07-13:
Sure

MitchBradley, 2023-07-13:
There is a bug in the new release that made some settings disappear