bdring/FluidNC #961: Problem: wifi off

norventa, 2023-07-13:
### Wiki Search Terms

wifi off
wifi_mode

### Controller Board

MKS DLC32 v2.1

### Machine Description

3 axis cnc.

### Input Circuits

```yaml
n/a.
```

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: K40 Laser
meta: (01.01.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
Resetting MCU
ets Jun  8 2016 00:&&ºËSH¿HH═ÐÚ0x1 (POWERON_RESET),bÙïÚ0x1b (SPI_FAST_FLASH_BOOU®Hh¢╣ÖÑØ═Ñ┴Ú 0, SPIWÑé┬òò
clk_drv:0x00,q_drv'┬┴0,d_drv:0x00,cs0_d.ùé┬┴0,hd_drv:0x00,wp_drv'┬┴0
mode:DIO, clock div:1C!Ù+æÚ0x3fff0030,len:118SHêÙ+æÚ0x40078000,len:13&&j
ho 0 tail 12 room 4
load'┬Ð0080400,len:3028
entr^é┬Ð00805e4

[MSG:INFO: FluidNC v3.0.x (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Using default speed map]
[MSG:INFO: Machine MKS DLC32 XYZ]
[MSG:INFO: Board MKS DLC32 2.0]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.22 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.2]
[MSG:INFO: Calling wifi_config.]
[MSG:INFO: WiFiConfig::begin<<]
[MSG:ERR: default]
[MSG:INFO: WiFi off]
[MSG:INFO: Called wifi_config.]
```

### User Interface Software

FluidTerm

### What happened?

I downloaded the source and built with PlatformIO. Flashed using PlatformIO. Got [MSG:INFO: WiFi off] message as shown in Startup Messages above.

Added some debug output to WiFiConfig::begin() shown in Other Information.

Discovered the switch statement jumped to default: (I added this).

This shows the wifi_mode->get() is returning a value that is invalid i.e. not in wifiModeOptions enum.

### GCode File

n/a.

### Other Information

`bool WiFiConfig::begin() {
        log_info("WiFiConfig::begin<<");
        //stop active services
        wifi_services.end();

        switch (wifi_mode->get()) {
            case WiFiOff:
                log_info("WiFi is disabled");
                return false;
            case WiFiSTA:
                log_info("WiFiSTA");
                if (StartSTA()) {
                    goto wifi_on;
                }
                goto wifi_off;
            case WiFiFallback:
                log_info("WiFiFallback");
                if (StartSTA()) {
                    goto wifi_on;
                }
                // fall through to fallback to AP mode
            case WiFiAP:
                log_info("WiFiAP");
                if (StartAP()) {
                    goto wifi_on;
                }
                goto wifi_off;
            default:
                log_info("default");
        }
`

norventa, 2023-07-13:
Ok, I have sorted it. Basically the WebUI::WiFiConfig creator function isn't getting called before the call to begin(). My solution was to replace the creator with a static void init() function and remove the destructor function. I call WiFiConfig::init() from setup() and replace all references of WebUI::wifi_config. with WebUI::WiFiConfig:: In many ways this is more appropriate as the class is really a singleton.

I have had this problem before using gcc with other projects and it is a known problem if not strictly a bug.

MitchBradley, 2023-07-13:
The problem is that PR #950 caused an initialization order dependency.  We are working on it.

MitchBradley, 2023-07-13:
Your fix, while similar to an approach I was trying, does not go far enough.  It also needs to be applied to BTConfig, NotificationsService, WebServer, and TelnetServer.  Just hang tight while we resolve it.

MitchBradley, 2023-07-14:
v3.7.4 resolves the wifi issues

closed 2023-07-14 by MitchBradley