bdring/FluidNC #962: Problem: wifi dosn't work after upgrading to v3.7.3

mstfyghm, 2023-07-13:
### Wiki Search Terms

wifi

### Controller Board

esp32 mini

### Machine Description

gantry co2 laser with steppers

### Configuration file

```yaml
no config (Using default configuration)
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:WARN: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: WiFi off]

Grbl 3.7 [FluidNC v3.7.3 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
$wifi
error:3
```

### User Interface Software

Fluidtem

### What happened?

After downloading _fluidnc-v3.7.3-win64_ and upgrading my machine to v3.7.3 It doesn't connect to wifi. so I tried with a fresh installation of fluidnc. The startup massage in fluidterm shows that the wifi is off by default. So I tried to configure wifi but it seems it doesn't know the $wifi command!

manukieli, 2023-07-13:
I've the same problem.

bdring, 2023-07-13:
@MitchBradley has confirmed the problem. The way the settings are being created needs to be tweaked a little. With C++ "tweaked a little" can be like tweaking a live hand grenade, so be patient.

closed 2023-07-14 by mstfyghm