bdring/FluidNC #969: Feature: guide for bluetooth

Stewie-040882, 2023-07-16:
### Machine Context

Installed bluetooth version, uploaded index.html.gz, then paired it to my phone (and shows a pc image in phone settings). but then cant do anything else.  

### Feature Description

FluidNC Wiki needs to be updated so there is more information available for bluetooth usage. currently the only information available is that you can use wifi or bluetooth, and that bluetooth could be an alternitive to wifi. but nothing on how to use FluidNC with bluetooth

### Other Approaches

unknown

### How I Can Help

approach FluidNC as a noob. Then try to find or follow "any" guide on controlling device via bluetooth (as i cant find one)

Ams-Laser, 2023-07-16:
I dont know if webuis is available from bluethoot but as a paired device it adds a serial port you can use as of it where conecte directly to pc

melvinisken, 2023-07-17:
I don't think you can use it via mobile device. As Ams mentioned, the Bluetooth-connection has an entirely different meaning than connecting to the laser. It creates a serial port, and you would need an APP that can control a laser via sending serial commands. Better is to use WiFi-Version, connect both devices to the same network and use the web interface.

MitchBradley, 2023-07-17:
Bluetooth cannot be used with WebUI, but there are apps for sending gcode over bluetooth.  Search for "bluetooth gcode sender".  When you find a solution that you like, write up how to use it and we will add that to the wiki.

Stewie-040882, 2023-07-18:
Thanks all, it would be nice to find that info in the wiki though. Theres plenty of info regarding wifi setup ect, but vague when it comes to bluetooth.
   I only wanted to use bluetooth as i was connecting via Wifi AP mode, and figured if i use bluetooth, i would still have internet access on my laptop/phone. I suppose its a good backup option, but i think i can get by for 10 minutes of no internet access lol

MitchBradley, 2023-07-18:
It would be nice to have an infinite amount of time and money to document everything as well as it should be documented.  But we have neither.

Stewie-040882, 2023-07-19:
no problem mitch, great effort on the work you've done and being able to keep it open source

FrankHovis, 2023-07-23:
This might be asking the obvious in the wrong place, but can the Wifi version appear as a Serial Port for things like Candle to connect to?

bdring, 2023-07-23:
This is not a feature of FluidNC. There may be some 3rd party apps that convert a Wi-Fi port to serial, but I have not used one in a long time.

Regardless, it will likely be slower than serial due to the protocol being optimized for a serial hardware.

KyoshoTom, 2023-11-04:
I am trying to use FluidNC on an MKS DLC32 to control a laser pointer turret. It only uses 2 steppers and I would like to use a PS3 controller (or similar) to control the turret. I chose the DLC32 because it has BT capabilities, but not even sure if what I am trying to do is possible. If anyone has any tips or ideas, Please let me know.

MitchBradley, 2023-11-04:
FluidNC does not directly support PS3, but you could use a combination of FluidNC as the controller and CNCjs as the sender.  The PS3 would be connected to a raspberry pi running CNCjs.  See https://github.com/balthisar/cncjs-pendant-gamepad .  Please be advised that the FluidNC team cannot provide any support for issues you might encounter with CNCjs installation or usage.

closed 2023-11-04 by MitchBradley

KyoshoTom, 2023-11-04:
Thank you Mitch

MitchBradley, 2023-11-04:
According to [this thread](https://millrightcnc.proboards.com/thread/3248/ps3-joystick-success), it is also possible to use a PS3 with the UGS sender, and possibly with other senders that support key mapping.  UGS is better-supported than CNCjs.

KyoshoTom, 2023-11-04:
I wonder if using FluidNC with wifi would be an easier approach? Using an Android app to control the turret. I knew I was getting in over my head with this project, but I want to try :) I also have an Arduino Uno R3 ESP32 based board. I am just trying to find the simplest option to make this work.

bdring, 2023-11-04:
Yes, of course. That is what the WebUI is for. We have been pushing that solution for a long time.

MitchBradley, 2023-11-05:
WebUI runs in a browser so you do not need a special app on the Android device.  The CNC machines in my production shop use FluidNC with wifi connections to a 10" Android tablet running WebUI in Chrome.  It is also possible to run WebUI on a phone or a PC, again via a browser.  I find the tablet to be ideal for production use.

KyoshoTom, 2023-11-05:
Perfect! I really appreciate the help from you guys, it means a lot. I been studying FluidNC, watching a lot of videos and going through the wiki page. It really is amazing. The way you edit the yaml and upload is next level (perfect for beginners like me). I think I just need to edit for the setup I have (which is very basic). Would I need to comment out the features I don't need or just leave them as they are?

MitchBradley, 2023-11-05:
Look at example config files

KyoshoTom, 2023-11-05:
Do you guys know if it is possible to upload code to the MKS-DLC32 via Arduino IDE (or VSC)? I assumed since it is based on the ESP32 that this was possible. I can't seem to find any info on this. I have read the DLC32 firmware pages here on github and they used the ESP32 Flash tool and upload a .bin file. I have done this before, but can I simply use Arduino IDE (since I am familiar with that)? Thank you

MitchBradley, 2023-11-06:
http://wiki.fluidnc.com/en/installation tells how to install precompiled binaries and how to use vscode+platformio for development.  Most people do not have to compile.

We have better installation tools and we do not support Arduino IDE.

KyoshoTom, 2023-11-07:
I put the laser pointer robot to the side for now, However, I do have another project that would be perfect for this MKS DLC32 using FluidNC on a CoreXY pen plotter. I think the WiFi set up would be best for this setup. Would I just need to edit an example_config file (MKS DLC32_2.1_XYZ) for my machine and then upload it?

MitchBradley, 2023-11-08:
yes