bdring/FluidNC #970: Problem:

Ams-Laser, 2023-07-16:
### Wiki Search Terms

homing cycle

### Controller Board

MKS Tynybee 1.0

### Machine Description

Gantry external stepper drivers

### Input Circuits

```yaml
n/a
```

### Configuration file

```yaml
board: AMS V1.1 XYYZA
name: AMS-Laser
planner_blocks: 128
kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2
  card_detect_pin: gpio.34:low
  frequency_hz: 5000000

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    steps_per_mm: 20
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 5000.000
    max_travel_mm: 1230.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      hard_limits: false
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

  y:
    steps_per_mm: 20
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 800.000
    max_travel_mm: 610.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low:pu
      hard_limits: false
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3

    # use Z driver for 2nd Y axis motor
    motor1:
      limit_neg_pin: gpio.22:low:pu
      hard_limits: false
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8:low
        disable_pin: I2SO.6

  z:
    steps_per_mm: 40.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 2000.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9

control:
  safety_door_pin: NO_PIN
  # on MT_DET connector
  reset_pin: NO_PIN
  # on TH1 connector
  feed_hold_pin: NO_PIN
  # on TB connector
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  # Heated Bed Terminal Block
  flood_pin: i2so.16
  # HE0 Terminal Block
  mist_pin: i2so.17
  delay_ms: 0

Laser:
  pwm_hz: 5000
  # on 3D Touch connector (has pdwn + PWM)
  output_pin: gpio.2:high:pd
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%

start:
  must_home: false
```

### Startup Messages

```text
no problem
```

### User Interface Software

webui

### What happened?

jog x 100mm pushed the HomeX it sets the Xw and Xm to 0
jog y 100mm pused the HomeY it sets Ym to 0 but not the Yw

### GCode File

na

MitchBradley, 2023-07-16:
"no problem" is not an acceptable answer for Startup Messages.  We want to see the actual messages.

MitchBradley, 2023-07-17:
Homing does not set work coordinates.  It sets machine coordinates.  If homing x happens to set the work coordinates to 0, it is because the work coordinate system that you are using has work x = machine x.

closed 2023-07-17 by MitchBradley

Ams-Laser, 2023-07-18:
was expecting this to happen acording to wiki at http://wiki.fluidnc.com/en/config/axes  :

A value of -1 means the machine will not move, but the current machine position (mpos) position of the axis will be set to the mpos_mm value for the axis. This can be used for axes that don't have any switches.

thnks

MitchBradley, 2023-07-18:
There is nothing in this report to suggest that a homing operation was performed.  The thing about -1 setting the mpos position applies to a homing operation.  For you config, if you send $HY, the y mpos should be set to 0, with Y work position set to the corresponding offset that is stored in the G54 offset (or G55 .. G59 depending on which work coordinate system is selected).

Ams-Laser, 2023-07-18:
might have an ofsett set, will look into it, thanks
i tought it should be both 0
Ed