bdring/FluidNC #973: n/a

FrankNz, 2023-07-17:
### Wiki Search Terms

n/a

### Controller Board

n/a

### Machine Description

n/a

### Input Circuits

```yaml
n/a
```

### Configuration file

```yaml
n/a
```

### Startup Messages

```text
n/a
```

### User Interface Software

Fluidterm

### What happened?

Running on MacOs Ventura 13.4.1(c)
The problem below was occurring before on 13.4.0

- Fluidterm >= 3.7.1 crashes after asking for a port, see attached image. I can still use Ctrl+] to quit Fluidterm.
- Fluidterm = 3.7.0 works OK, see attached image.

The images are of connections to the same controller. Not simultaneously.

### GCode File

na

### Other Information

[image]
[image]

MitchBradley, 2023-07-17:
Please use a more useful title than "n/a".

The change in fluidterm.py between 3.7.0 and 3.7.1 occurred as a result of #911 and #927, authored by @Alex-CodeLab .  As the author of the most recent change, it is his responsibility to address the problems.

FrankNz, 2023-07-17:
I stand corrected, sorry.

FrankNz, 2023-07-17:
I should have added I'm on Python 3.11.4, pip 23.1.2

Alex-CodeLab, 2023-07-21:
@FrankNz Does that happen all the time, or sometimes?

There seems to be some strange racecondition in the threads.

Here I run the same code 3 times, all with different results 
(1st: lock, 2nd: crash, 3rd: working)
[image]

Alex-CodeLab, 2023-07-21:
Adding halve a second delay when starting the `reader()` seems to fix it. 
@FrankNz could you try if this helps? 
https://github.com/Alex-CodeLab/FluidNC/commit/bd9aca0ec47dc48d65c1a466fcc5751f931575c0

FrankNz, 2023-07-21:
HI,
For me it happens every time.
I have attached a clip of my Mac below.
I have received your subsequent email 

Thank you for providing a potential fix, I’ll try your it later today and get back to you.

Regards 
Frank

￼

FrankNz, 2023-07-21:
HI,
For me it happens every time.
I have attached a clip of my Mac below.
I have received your subsequent email

Thank you for providing a potential fix, I’ll try your it later today and get back to you.

Regards
Frank
[image]

FrankNz, 2023-07-21:
It is working now, Thank you.

closed 2023-07-21 by FrankNz