bdring/FluidNC #974: UGS NOT CONNECTING

arlimbad222, 2023-07-17:
### Wiki Search Terms

nun

### Controller Board

CUSTOM

### Machine Description

4 AXIS mini milling machine

### Configuration file

```yaml
name: "4AXIS"
board: "4AXIS"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 3.5
  pulse_us: 3.5
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 317
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 400
      seek_mm_per_min: 3500
      settle_ms: 250
      seek_scaler: 2
      feed_scaler: 2
    
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.16:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.18
    motor1:
      null_motor:

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 250
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 400
      seek_mm_per_min: 3500
      settle_ms: 250
      seek_scaler: 2
      feed_scaler: 2

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.17:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 140
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      mpos_mm: 0
      positive_direction: true
      feed_mm_per_min: 500
      seek_mm_per_min: 3500
      settle_ms: 250
      seek_scaler: 1.5
      feed_scaler: 1.5
      

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.4:low:pu
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

  a:
    steps_per_mm: 8.889
    max_rate_mm_per_min: 31400
    acceleration_mm_per_sec2: 20000

    motor0:
      limit_all_pin: NO_PIN
      stepstick:
        direction_pin: gpio.13
        step_pin: gpio.12
    motor1:
      null_motor:        

coolant:
  flood_pin: gpio.21
  mist_pin:  gpio.23

        
probe:
  pin: gpio.32:low:pu

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1: 
  macro2:
  macro3:

start:
  must_home: false
  deactivate_parking: true
  check_limits: false

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.5:low:pu
  cycle_start_pin: gpio.19:low:pu
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

PWM:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 24000=100%
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.0.x (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:ISSUE.YAML]
[MSG:INFO: Machine 4AXIS]
[MSG:INFO: Board 4AXIS]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:4us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-317.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.18 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-250.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-140.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.13 Disable:NO_PIN]
[MSG:INFO:   Motor1]
[MSG:INFO: feed_hold_pin gpio.5:low:pu]
[MSG:INFO: cycle_start_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.22 Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant gpio.21]
[MSG:INFO: Mist coolant gpio.23]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: BT Started with AL600_TEST]
ok
```

### User Interface Software

UGS

### What happened?

I am using platform.io to compile fluid nc, when i try to connect ugs using fluidnc the ugs give this error 
(*** Expected a 'FluidNC 3.3.0' or later but got 'FluidNC 3.0.0'
*** Could not establish connection with the controller
*** Connection close)

& i am using v3.6.5 of fluid nc i jut dont know how to change this line of startup messege from code. here is the line.
$ss
[MSG:INFO: FluidNC v3.0.x (noGit)]

with grbl normal it is working but the only problem is homing button is not working with grbl selection .

### Other Information

[image]
[image]
[image]

BlueOrangeLive, 2023-07-17:
Hello . In the file git-version.py in line 13 + 23 simply enter the new version v3.7.4 then it works with UGS as well. Recompile.

arlimbad222, 2023-07-17:
Hi @BlueOrangeLive 

i change line that line now ugs connect to fluid nc but there is one problem with normal grbl selection A Rotary axis works well but when i select fluid nc taht a axis is gone i share some screenshots here. 

[image]
with normal grbl selection

[image]
with fluid nc 

i use same board and config file.

MitchBradley, 2023-07-17:
Another way to change the version number in the startup message is with $start/message http://wiki.fluidnc.com/en/features/commands_and_settings#settings

breiler, 2023-07-17:
We are looking into the problem.

UGS uses the `$Config/Dump` to determine which axes are available. But it can't currently not parse the response of the command as FluidNC doesn't seem to finish the response with an "ok" anymore.

We could probably fix this in UGS to detect the axis using the status response instead. But I would like to use the $Config/Dump command for other purposes, so this needs to be fixed anyway.

breiler, 2023-07-18:
Ok, the error was in UGS and was introduced when I consolidated GRBL stuff with FluidNC.  

I have fixed this in https://github.com/winder/Universal-G-Code-Sender/pull/2260 and should be available in the latest nightly build here: https://github.com/winder/Universal-G-Code-Sender#downloads

closed 2023-07-18 by arlimbad222