bdring/FluidNC #975: PWM settings not using the correct speed_map values from config_yaml

kartom, 2023-07-18:
### Wiki Search Terms

PWM_HZ

### Controller Board

Custom

### Machine Description

DM542 steppers, 10V PWM VFD spindle

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "3 Axis Router"

start:
  must_home: false
  deactivate_parking: false
  check_limits: true

control:
  safety_door_pin: gpio.32:pu
  cycle_start_pin: gpio.33:low:pu
probe:
  pin: gpio.35:low
 
stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 5
  pulse_us: 3
  disable_delay_us: 0

axes: 
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 380.0
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 380.000
      seek_mm_per_min: 400
      feed_mm_per_min: 50
    motor0:
      limit_pos_pin: gpio.34
      standard_stepper:
        step_pin: gpio.22
        direction_pin: gpio.21:low
  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 595.0
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 595.000
      seek_mm_per_min: 400
      feed_mm_per_min: 50
    motor0:
      limit_pos_pin: gpio.39
      standard_stepper:
        step_pin: gpio.17
        direction_pin: gpio.16:low
  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 115.000
    soft_limits: true
    homing:
      cycle: 1
      mpos_mm: 115.000
      seek_mm_per_min: 200
      feed_mm_per_min: 50
    motor0:
      limit_pos_pin: gpio.36
      standard_stepper:
        step_pin: gpio.4
        direction_pin: gpio.15:low

PWM:
  pwm_hz: 800
  output_pin: gpio.13:low
  enable_pin: gpio.2
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0.000% 24000=100.000%
  
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Using default speed map]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board 3 Axis Router]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:5us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,380.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.22 Dir:gpio.21:low Disable:NO_PIN]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,595.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.17 Dir:gpio.16:low Disable:NO_PIN]
[MSG:INFO:  Y Pos Limit gpio.39]
[MSG:INFO: Axis Z (0.000,115.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.4 Dir:gpio.15:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.36]
[MSG:INFO: safety_door_pin gpio.32:pu]
[MSG:INFO: cycle_start_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.2 Out:gpio.13:low Dir:NO_PIN Freq:1000Hz Period:65535]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.35:low]
[MSG:INFO: Connecting to STA SSID:Hemma]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.20]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

The PWM/spped_map in the configuration file does not affect the prm to pwm mapping.
An rpm setting of 5000 rpm gives 50% PWM for the following speed_map: 0=0.000% 24000=100.000% which is obviously not correct.

kartom, 2023-07-18:
I have measured the actual PWM duty cycle coming out from pthe ESP chip so the converter is not relevent in this case and the setting of 5000 rpm gives 50% duty cycle, and the setting of 100 rpm gives 1% duty cycle so it seems to me that the code is using some default 10000 rpm as the scaling from rpm du PWM duty % and not the actual setting in the spped_map which in this case should have given a PWM duty of 5000/24000=20,83% and not 50% for 5000 rpm.

kartom, 2023-07-18:
I made a dump of the settigs in the machine and as i suspected, the speed_map is not correctly interpreted:

$cd
board: 3 Axis Router
name: ESP32 Dev Controller V4
meta: 
stepping: 
  engine: RMT
  idle_ms: 250
  pulse_us: 3
  dir_delay_us: 5
  disable_delay_us: 0
  segments: 12
spi: 
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard: 
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
kinematics: 
  Cartesian: 
axes: 
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 320.000000
    max_rate_mm_per_min: 5000.000000
    acceleration_mm_per_sec2: 200.000000
    max_travel_mm: 380.000000
    soft_limits: true
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 380.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 400.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      standard_stepper: 
        step_pin: gpio.22
        direction_pin: gpio.21:low
        disable_pin: NO_PIN
  y: 
    steps_per_mm: 320.000000
    max_rate_mm_per_min: 5000.000000
    acceleration_mm_per_sec2: 200.000000
    max_travel_mm: 595.000000
    soft_limits: true
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 595.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 400.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      standard_stepper: 
        step_pin: gpio.17
        direction_pin: gpio.16:low
        disable_pin: NO_PIN
  z: 
    steps_per_mm: 400.000000
    max_rate_mm_per_min: 4000.000000
    acceleration_mm_per_sec2: 200.000000
    max_travel_mm: 115.000000
    soft_limits: true
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 115.000000
      feed_mm_per_min: 50.000000
      seek_mm_per_min: 200.000000
      settle_ms: 250
      seek_scaler: 1.100000
      feed_scaler: 1.100000
    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.36
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      standard_stepper: 
        step_pin: gpio.4
        direction_pin: gpio.15:low
        disable_pin: NO_PIN
control: 
  safety_door_pin: gpio.32:pu
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: gpio.33:low:pu
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
coolant: 
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
probe: 
  pin: gpio.35:low
  toolsetter_pin: NO_PIN
  check_mode_start: true
macros: 
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 
start: 
  must_home: false
  deactivate_parking: false
  check_limits: true
parking: 
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000
user_outputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN
PWM: 
  pwm_hz: 1000
  direction_pin: NO_PIN
  output_pin: gpio.13:low
  enable_pin: gpio.2
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0.000000% 10000=100.000000%
  off_on_alarm: false
arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
ok

MitchBradley, 2023-07-18:
Duplicate of problem reported on Discord.  Fixed by #971

closed 2023-07-18 by MitchBradley

kartom, 2023-07-19:
Great!