bdring/FluidNC #977: M4, laser inline commands have poor performance.

V1EngineeringInc, 2023-07-19:
### Wiki Search Terms

lightburn, M4, inline, Uart, RMT, RTM, i2s_static

### Controller Board

Jackpot

### Machine Description

MPCNC, Dual X, Dual Y, Single Z
TMC 2209, New UART code

### Configuration file

```yaml
board: Jackpot TMC2209
name: Jackpot TMC2209
meta: 06-20-2023 B. Dring

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
uart1:
  txd_pin: gpio.0
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  baud: 115200
  mode: 8N1

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 50.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 180.000
    max_travel_mm: 1220
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 1200.000
      seek_mm_per_min: 300.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    #X
    motor0:
      limit_neg_pin: gpio.25:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5.000
      tmc_2209:
        uart_num: 1
        addr: 0
        cs_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.500
        hold_amps: 0.400
        microsteps: 8
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
[... 328 lines cut ...]

### User Interface Software

lightburn, or wifi microsd-esp3dui

### What happened?

Extremely poor performance of inline commands. With the all new setup I am using (new hardware, and new firmware), inline commands are causing severe stuttering and obvious slow down at each power level change. ("S" command). ((It looks almost like it is accelerating for the power changes, this exact thing happen in Marlin at one point. I am unsure if it is this or the new hardware))

Same resolution setting in other firmware have proved to work at 20mm/s and 80mm/s. This is showing issues even at 5mm/s.

Lightburn settings.
[image]

### GCode File

[raster test.zip](https://github.com/bdring/FluidNC/files/12096691/raster.test.zip)

### Other Information

I have verified all the same settings as the FluidNC lightburn settings, Vector burn files work fine.
I tried different microstepping. 8 and 16th
I tried mapping the power level standard 1000, and 255.

The version shown in the MSG section seems odd but it is from last night's DevT branch, the only one with the UART settings needed for this board

V1EngineeringInc, 2023-07-19:
If I run from the microSD, 
"[MSG:WARN: Low memory: 12884 bytes]
[MSG:INFO: Channel auto report interval set to 50 ms]"

Shows up in the terminal, I suspect that is normal.

V1EngineeringInc, 2023-07-19:
Changing the mapping back to 1000 per discord user suggestion.

MitchBradley, 2023-07-19:
First of all, let's setup some procedures to make debugging possible.  The "nogit" in the version message suggests that the source code was downloaded as a zip ( I wish github could be configured to turn off that feature).  It is much better to fork the repository and make your own working branch.  Every time that you report a result, make sure that any code changes are committed and pushed to the fork branch. Then the version number will reflect the source code via the commit hash, and you can point to your fork so we can see your actual code.  If we can't see your code, every hypothesis has to be sandwiched with "I wonder if .... but there is no quick way to find out", which is inefficient to the point of being intolerable.

Also let's keep this entirely on GitHub.  Having to consult two different channels is similarly inefficient.

V1EngineeringInc, 2023-07-19:
Okay, understood. Let me get a few orders out and then I can spend some more time and focus again this afternoon.

V1EngineeringInc, 2023-07-20:
" It is much better to fork the repository and make your own working branch."

I am not really understanding what it takes to make this happen. I am not understanding everything that needs to be modified in the workflow file. Other than the version number, does something else change. I am pretty fluent in compiling several types of firmware and making some edit but workflows are still a bit of a magical to me.

V1EngineeringInc, 2023-07-20:
Is there another way I can confidently compile the devt branch to prove I am using the right one?

V1EngineeringInc, 2023-07-20:
" Every time that you report a result, make sure that any code changes are committed and pushed to the fork branch."
At this point I have made no changes, Bart has made all the UART changes.

MitchBradley, 2023-07-20:
The first step is to create a fork with the Fork button.  That will make a copy of the git repository in your github account.

[image]

The next step depends on the tools that you use.  Do you use the VsCode IDE?  If so you can do the next step directly from VsCode.

MitchBradley, 2023-07-20:
Assuming VsCode, do File>New Window and then under Start, hit Clone Git Repository, then Clone from GitHub, then start typing v1engineering and choose v1engineering/FluidNC when it comes up.

V1EngineeringInc, 2023-07-20:
I use vscode, I have forked this repo. I am just not understanding how to get the workflow file to use the devt branch and spit out all the files to my fork not try to use your and barts repos.

MitchBradley, 2023-07-20:
The workflow file is irrelevant.  You need to select the Devt branch in the Git extension, then use the platformio upload button to build and upload to the esp32

MitchBradley, 2023-07-20:
Or better yet, "Create new branch from ..." and choose origin/Devt as the starting point.  Give it a new name for your work.

MitchBradley, 2023-07-20:
The workflows are for creating release bundles.  You don't need to do that at all.

V1EngineeringInc, 2023-07-20:
Instead of just downloading, and compiling it? Okay that will have to wait for tomorrow I guess. I am not at that computer that is near the connected board. 
I just downloaded and compiled then I tried pio direct to the board, fluidterm ctrl-u, then ctrl-r, and the update from esp3dui. If you think it will make a difference, I will do it that way. Learning something new is always good.

V1EngineeringInc, 2023-07-20:
I am confident I did compile and flash the right branch, but I understand wanting to see the correct numbers for troubleshooting’s sake.

I am just concerned the new UART controls are causing some issues, since no one seems to be having issues with M4 other than me.

V1EngineeringInc, 2023-07-20:
I should have plenty of time tomorrow. I had pushed a lot of work aside to work on the new board, then a tradeshow, and set me back a few days. Shipping orders has to be done in a timely manor but his is my next highest priority. I appreciate the help.

MitchBradley, 2023-07-20:
Lots of people are having problems with laser rasters.

If you just download and compile, I have no confidence about what you actually did.  I can take your word for it, but I cannot double-check anything.  If we do have to change anything, we quickly lost track of what is what.  Every professional SW engineer uses git or some equivalent version management system religiously, because they have experienced the confusion that inevitably results when they don't.  git itself is confusing, but the alternative of not tracking changes carefully is much worse.  VsCode has pretty good git integration, so you don't have to do much extra work to use it.  But you do have to learn which buttons to push.

V1EngineeringInc, 2023-07-20:
The issue still persists.

V1EngineeringInc, 2023-07-20:
If I remove the direction pin as shown in the wiki, http://wiki.fluidnc.com/en/config/config_spindles#laser, it makes a single moves and stops. $32=1 seems to have no effect, although it does boot to $32=0.

MitchBradley, 2023-07-20:
Now that we have a trackable build setup, I will look into it - after breakfast.

V1EngineeringInc, 2023-07-20:
Cool, thanks.

MitchBradley, 2023-07-20:
> If I remove the direction pin as shown in the wiki ...

Do you mean the enable pin?  That section of the wiki does not mention direction as far as I can see

> $32=1 seems to have no effect, although it does boot to $32=0.

FluidNC does not use $32 for laser mode.  If there is a laser: section, it behaves like $32=1.  If the spindle section is pwm:, it behaves like $32=0.

MitchBradley, 2023-07-20:
Just so we can eliminate variables - can you use STA mode instead of AP mode for wifi?  AP puts a lot of load on the ESP32 and we have also seen hard-to-debug evidence of memory leaks in the AP-mode wifi stack.  We do not recommend using it except for initial setup.

MitchBradley, 2023-07-20:
Now that the compiling setup confusion is resolved, I have dug deeper and see that you have a pwm: section instead of a laser: section.  For pwm spindles, laser mode is off.  They are treated like milling machines/routers where speed changes synchronize the planner, ensuring that the spindle has spun up/down prior to motion at the new speed.  Laser power, of course, can be modulated in fractions of a millisecond, while rotating spindles can take seconds to reach a new target speed.

MitchBradley, 2023-07-20:
Okay I now see what you mean about removing the direction pin - your pwm: section has a direction pin.  In non-laser mode, i.e. when the section is pwm: instead of laser, M4 means counterclockwise rotation instead of the normal clockwise rotation of M3.  If you have a pwm: section with no direction pin, FluidNC disables M4 because, lacking a direction pin, it has no way to establish the CCW direction.

Change the pwm: section to laser: and see if your problem goes away.

V1EngineeringInc, 2023-07-20:
Much much better.

I am seeing glitches at higher speeds, let me try with STA mode.

Here is the last few lines of a glitch to check out while I change settings.
`G1 X0.189S781.2
G1 X0.19S108.2
G1 X9.0ok
98S80
G1 X4.5S0
G1 Y0.18ok
9S0
G1 X-4.5S0
G1 X-9.098S80
G1 X-0.19S820
G1 X-0.189S192.8
G1 X-0.19S80
G1 X-0.19S830.5
G1 X-0.189S80`

MitchBradley, 2023-07-20:
Define "glitch" and "higher speeds".  How high?  Have you done a bandwidth analysis to see if the serial rate can keep up with the motion?

MitchBradley, 2023-07-20:
If the "glitch" is the "ok" in the middle of the echoed command, that could be due to the sender program having separate threads for displaying transmitted characters vs received characters.  Without knowing details of how you got that listing, it's hard to say.

V1EngineeringInc, 2023-07-20:
Sorry I just turned off AP STA. I will try again.

The machine was running smooth at about 70mm/s (as compared to grbl-hal hits 80 at the same settings on a different board). The freeze happened all the way down to 20mm/s it would just say busy after a a few mm worth of burn. It does seem like a memory thing but what do I really know. Testing again with AP/STA off from the lightburn panel.

MitchBradley, 2023-07-20:
Is WebUI running at the same time as lightburn?  If so, is auto-reporting enabled?  Auto-reporting at high refresh rates can use up a lot of I/O bandwidth.  It is okay for milling but not for lasering.

MitchBradley, 2023-07-20:
Are you using lightburn over USB or TCP?  TCP sucks compared to USB because of packet latency.

MitchBradley, 2023-07-20:
TCP has good bandwidth for large packets, but rather high latency.  For small packets the bandwidth drops off severely.  The GRBL protocol was heavily tuned toward the characteristics of serial ports - very small packet sizes due to limitations in AVR serial receive buffers, coupled with low and predictable latency for single characters.

V1EngineeringInc, 2023-07-20:
USB. 

So it looks like I can run about 35-38mm/s with smooth running no signs of stuttering at power changes with STA off. (same .19line interval 134dpi). Marlin hits 22 at best with 8 bit and 26mm/s with a 32 bit chip. That is great.
I can try turning AP back on with auto reporting off...just need to figure out how to enable it again.

V1EngineeringInc, 2023-07-20:
What else might affect the speed? I have 8th microstepping, figured less pulses was better, what about"pwm_hz: 5000" try dropping to 2000?

Just looking to see an ultimate max here before I set up a normal parameters for users.

MitchBradley, 2023-07-20:
38 mm/sec is 26 ms/mm or 5 ms/ .19 mm .  5 ms is 57 characters at 11500 baud (87 us/char).  So, unless there is some inefficiency in the way that lightburn manages buffers, the UART bandwidth is probably not the limiting factors.  You could test that by shortening the GCode lines.  After one G1 line, the next line need not say G1, so 
```
G1 X0.19S108.2
G1 X9.098S80
G1 X4.5S0
G1 Y0.189S0
G1 X-4.5S0
G1 X-9.098S80
G1 X-0.19S820
G1 X-0.189S192.8
G1 X-0.19S80
G1 X-0.19S830.5
G1 X-0.189S80```
could be shortened to:
```
G1
X0.19S108.2
X9.098S80
X4.5S0
Y0.189S0
X-4.5S0
X-9.098S80
X-0.19S820
X-0.189S192.8
X-0.19S80
X-0.19S830.5
X-0.189S80
```
In hand-coded GCode, it is no unusual to first say
```
G1 F1000
```
to establish the mode and feedrate, then on following lines you just mention the x and y coordinates.

Another trick is to ensure that lines end with only linefeed, not CR-LF, thus saving one character per line.

Yet another trick is to set the maximum S value to a small integer like 9, using the speed map to associate that with the desired pwm values.  Then instead of saying S80, S192.8, S830.5, etc, you can say S1, S2, S9

A final trick would be to cheat on the steps_per_mm so that instead of saying "X-.189", you could say "X-1".  With all those tricks you could reduce the typical line length from 16 characters to 6 - e.g. "X-1S5<newline>"

But even without doing all the tricks, you can find out if line bandwidth is a limiting factor.  Just do the easy tricks like omitting the "G1 " and see if you can go a little faster.

MitchBradley, 2023-07-20:
Another thing to try is increasing planner_blocks http://wiki.fluidnc.com/en/config/top_level_config_items

V1EngineeringInc, 2023-07-20:
Direct from the MicroSD in STA mode should be the fastest then? Skip the whole USB thing and no AP mode Memory bottleneck?

MitchBradley, 2023-07-20:
In principle, yes.  Give it a try.

V1EngineeringInc, 2023-07-21:
If I enable STA mode, still running off USB, It will eventually freeze. Depending on speed in can take 30 seconds or less.  I don't think I can test it from the Micro SD.

I am going to turn up the buffer size, 32, maybe 48. Not sure what else to try if that doesn't help.

V1EngineeringInc, 2023-07-21:
Doubling the planner got me above 100mm/s, not bad at 120mm/s, before stuttering with the wifi still off.

V1EngineeringInc, 2023-07-21:
Well Mitch I have done a ton of learning today. Thanks for being so patient with me.

How about some great news....in AP mode I am still getting very little hesitation with M4 from the Micro SD. So from lightburn wired wifi off and AP mode direct from MicroSD is the same.

The only issues are M4 with wifi on while directly connected to lightburn (a common use as it has all the laser calibrations built in). Maybe add the command to temporarily turn it off to the wiki $WiFi/Mode=off? I might be totally wrong.

When I disconnected from the AP is hesitated for a long while then resumed. I am guessing STA mode fixes that? Last thing for me to test as I get a poor signal by my CNC.

Should I close this issue?

V1EngineeringInc, 2023-07-21:
I guess I need to retest to see how much added planner/memory helps, but it definitely does with M4. If there is not a huge downside, I will leave the public config set to 32 for my users.

closed 2023-07-23 by V1EngineeringInc