bdring/FluidNC #982: Critical issue for laser application & perhaps other applications - Speed_map not correctly read from config

Avataar120, 2023-07-26:
### Wiki Search Terms

speed_map

### Controller Board

Personal design - but not relevant for this issue

### Machine Description

Not relevant for this issue

### Input Circuits

```yaml
Not relevant for this issue
```

### Configuration file

```yaml
name: K40_AvaShield
board: 4_Ava_Shield_8.x_XYA_OLED

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 310.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 78.74
    max_rate_mm_per_min: 20000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 210.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2500.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        ms3_pin: I2SO.7
        step_pin: I2SO.6
        direction_pin: I2SO.5:low
        disable_pin: I2SO.4
        

  a:
    #4 micro-steps for 1.8 degrees
    #If 16 teeth on motor / 20 teeth on the roller : 2.778 (4/1.8*20/16)
    #If same teeth on motor and roller : 2.222
    steps_per_mm: 2.778
    max_rate_mm_per_min: 53200.000
    acceleration_mm_per_sec2: 100000.000
    max_travel_mm: 360.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 100.000
      settle_ms: 250.000
      seek_scaler: 1.500
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: gpio.32:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 0.000
      stepstick:
        ms3_pin: I2SO.15
        step_pin: I2SO.14
        direction_pin: I2SO.13
        disable_pin: I2SO.12

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

i2c0:
   sda_pin: gpio.25
   scl_pin: gpio.33

oled:
   i2c_num: 0
   i2c_address: 60
   width: 128
   height: 64
   radio_delay_ms: 1000
   
control:
  safety_door_pin: gpio.27:low:pu
  reset_pin: gpio.12:low:pu
  feed_hold_pin: gpio.14:low:pu
  cycle_start_pin: gpio.13:low:pu

coolant:
  mist_pin: gpio.4
  delay_ms: 50.000

probe:
  pin: gpio.36:low
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  tool_num: 0
  speed_map: 0=0% 1000=100%
  output_pin: gpio.15
  enable_pin: gpio.16:low
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 15000

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.3]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Using default speed map]
[MSG:INFO: Machine K40_AvaShield]
[MSG:INFO: Board 4_Ava_Shield_8.x_XYA_OLED]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: I2C SDA: gpio.25, SCL: gpio.33, Freq: 100000, Bus #: 0]
[MSG:INFO: OLED I2C address:0x3c width: 128 height: 64]
[MSG:ERR: OLED is not responding]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (0.000,310.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.39:low]
[MSG:INFO: Axis Y (0.000,210.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.6 Dir:I2SO.5:low Disable:I2SO.4]
[MSG:INFO:  Y Neg Limit gpio.34:low]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis A (-360.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.14 Dir:I2SO.13 Disable:I2SO.12]
[MSG:INFO:  A Neg Limit gpio.32:low:pu]
[MSG:INFO: safety_door_pin gpio.27:low:pu]
[MSG:INFO: reset_pin gpio.12:low:pu]
[MSG:INFO: feed_hold_pin gpio.14:low:pu]
[MSG:INFO: cycle_start_pin gpio.13:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.16:low Out:gpio.15 Freq:15000Hz Period:4095]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: WiFi off]
```

### User Interface Software

Fluidterm

### What happened?

When booting, there is this warning : 
[MSG:INFO: Using default speed map]

However, speed map is well present in config file : 
laser:
  speed_map: 0=0% 1000=100%

If I enter $cd, I obtain : 
  speed_map: 0=0.000000% 255=100.000000%
which is obviously incorrect

If I enter manually 
$speed_map=0=0% 1000=100%
I get OK and if I issue $cd command, I can see it is taken into account

  speed_map: 0=0.000000% 1000=100.000000%

But if I issue $cd=config.yaml to save it to config file and if I reboot the board, 

Same warning message at start & map come back to 255

This is critical for lasers as it can lead to overcurrent in the tube and to fry the tube.

if I look to std::vector<speedEntry> Parser::speedEntryValue(), I do not see any speed_entries affectations in the code if it can help ...

MitchBradley, 2023-07-26:
[image]

closed 2023-07-26 by MitchBradley

Avataar120, 2023-07-26:
Oups ... Sry I was thinking to be on the latest version ... my bad ... I close it