bdring/FluidNC #983: Problem: File upload via webui to SD card fails, infoscreen shows:  Error Upload failed

Wieringerwerf, 2023-07-28:
### Wiki Search Terms

file upload sd card cancelled

### Controller Board

MKS Tinybee 1.03

### Machine Description

LowRider v3 CNC 
Gigastone 32GB MLC SD card / FAT32 

### Configuration file

```yaml
board: MKS TinyBee V1.0 XYYZZ
name: LR3CNC-ESP32

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
#try lower frequency
   frequency_hz: 6000000
#  card_detect_pin: gpio.34
  
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 1250
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 250.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      hard_limits: true
      pulloff_mm: 4.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0

 

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 2500
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 250.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:high
      hard_limits: true
      pulloff_mm: 34.000
      stepstick:
        step_pin: I2SO.4
        direction_pin: I2SO.5
        disable_pin: I2SO.3

   # use E0 driver for 2nd Y axis motor
    motor1:
      limit_neg_pin: gpio.32:high
      hard_limits: true
      pulloff_mm: 34.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9
 
  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 50.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.22:high
      hard_limits: true
      pulloff_mm: 10.000
      stepstick:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6
        
   # use E1 driver for 2nd Z axis motor
    motor1:
      limit_neg_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 6.100
      stepstick:
        step_pin: I2SO.13
        direction_pin: I2SO.14
        disable_pin: I2SO.12

probe:
  pin: gpio.36:low
#  pin: NO_PIN
  
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
#  feed_hold_pin: gpio.36:low
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.7.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:DBG: Spiffs mount failed: ESP_FAIL]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:ERR: Skipping key frequency_hz indent 3 this indent 2]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine LR3CNC-ESP32]
[MSG:INFO: Board MKS TinyBee V1.0 XYYZZ]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1250.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.39]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO: Axis Y (0.000,2500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.33]
[MSG:DBG:  Y Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO:  Y2 Neg Limit gpio.32]
[MSG:DBG:  Y2 Neg Limit 0]
[MSG:INFO: Axis Z (-50.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Z Neg Limit gpio.22]
[MSG:DBG:  Z Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.14 Disable:I2SO.12]
[MSG:INFO:  Z2 Neg Limit gpio.35]
[MSG:DBG:  Z2 Neg Limit 0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: Connecting to STA SSID:Noordeloos]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.0.1.245]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://LR3CNC-ESP32.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:DBG: config.yaml hash "3B0675131E32BB7E5C1ACA63A52D26133B608212A172237CE0936A2320052161"]
[MSG:DBG: favicon.ico hash "B8B2871A343CA0F9A7A130213226962AD2BE4DFD28D7D8B756C3557569CB876D"]
[MSG:DBG: index.html.gz hash "94B77CC427B19E67041AC4B3891318595E3B4D0E1337479D1A0DC8F51244A6A6"]
[MSG:DBG: macro1.g hash "5A77EF10BC6A8501817D2FA644A1E18AD9CC02D7AE6500A920CC0CD31FF657FE"]
[MSG:DBG: macrocfg.json hash "45D36D4485FC3400AC4DBB4096C7EC9655B8E3B6922467E63947296F197E9892"]
[MSG:DBG: preferences.json hash "556C763CBE5A9F44A4AD6F26D65810F91E41589E1511004438BAD7D788C0653B"]
ok
```

### User Interface Software

ESP3D for FluidNC

### What happened?

Try to upload 10mb file to see speed, randomly fails between 6-7mb mark

Screen shows: ERROR: Upload Failed
Log shows: [MSG:INFO: Upload cancelled]

MitchBradley, 2023-07-28:
[MSG:ERR: Skipping key frequency_hz indent 3 this indent 2]

Wieringerwerf, 2023-07-28:
Saw that as well,

not this is the $ss output:

`[MSG:INFO: FluidNC v3.7.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:DBG: Spiffs mount failed: ESP_FAIL]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine LR3CNC-ESP32]
[MSG:INFO: Board MKS TinyBee V1.0 XYYZZ]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:6000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1250.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.39]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO: Axis Y (0.000,2500.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.4 Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.33]
[MSG:DBG:  Y Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO:  Y2 Neg Limit gpio.32]
[MSG:DBG:  Y2 Neg Limit 0]
[MSG:INFO: Axis Z (-50.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Z Neg Limit gpio.22]
[MSG:DBG:  Z Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.14 Disable:I2SO.12]
[MSG:INFO:  Z2 Neg Limit gpio.35]
[MSG:DBG:  Z2 Neg Limit 0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: Connecting to STA SSID:Noordeloos]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.0.1.245]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://LR3CNC-ESP32.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:DBG: config.yaml hash "E0D15FF04DCC9C4D59E76DAEAA3E998E3F2D4575A0BDD6532B2F3F3D9645508E"]
[MSG:DBG: favicon.ico hash "B8B2871A343CA0F9A7A130213226962AD2BE4DFD28D7D8B756C3557569CB876D"]
[MSG:DBG: index.html.gz hash "94B77CC427B19E67041AC4B3891318595E3B4D0E1337479D1A0DC8F51244A6A6"]
[MSG:DBG: macro1.g hash "5A77EF10BC6A8501817D2FA644A1E18AD9CC02D7AE6500A920CC0CD31FF657FE"]
[MSG:DBG: macrocfg.json hash "45D36D4485FC3400AC4DBB4096C7EC9655B8E3B6922467E63947296F197E9892"]
[MSG:DBG: preferences.json hash "556C763CBE5A9F44A4AD6F26D65810F91E41589E1511004438BAD7D788C0653B"]
ok`

MitchBradley, 2023-07-28:
Does it still fail with the frequency fixed?

Wieringerwerf, 2023-07-28:
just tried, got a bit further but at 9mb, same error, first in log:
[MSG:INFO: Upload cancelled]
then pop-up: error: Upload failed

Wieringerwerf, 2023-07-28:
testing lowering to 5xxx

Wieringerwerf, 2023-07-28:
failed at 6.14MB

MitchBradley, 2023-07-28:
I just uploaded a 10.52 MB file to SD at 8000000 Hz, with no problems, on one of my controllers.  I will try with a Tinybee.

Wieringerwerf, 2023-07-28:
tried at 5 and at 10, random issues, tried different cards as well, tried different browsers as well, will try different computer (wired)

MitchBradley, 2023-07-28:
I tried with a Tinybee V1.0 using the config file at https://github.com/bdring/FluidNC/blob/main/example_configs/MKS_TinyBee_1_XYZAB.yaml

I uploaded a 10.52 MB .tgz file with WebUI, no problems.  The SD card is a SanDisk Ultra 16GB.

MitchBradley, 2023-07-28:
I tried it with your config file (corrected frequency_hz indent, frequency 6000000).  Worked.

My Tinybee board has nothing connected other than the power supply, the USB, and the SD card.  No stepsticks, no switches, ...

MitchBradley, 2023-07-28:
Unplugged USB; still works.

The firmware is 3.7.5.  The power supply is a bench supply set to 14.5V with a 2.2A current limit.  When uploading, the supply says the current is 0.09A.

Wieringerwerf, 2023-07-28:
tried, 4.5mb > no problem
tried 5mb binary, renamed with .gcode, no problem
10mb binary renamed with .gcode, problems. 

Will try a different sd brand tomorrow and with the other config file as well.

For now, thanks for the help, wife is calling, off to bed.

MitchBradley, 2023-07-28:
The USBPWR jumper was OFF.  I changed it to ON, turned off the bench supply, and reconnected the USB.  It still works.

Wieringerwerf, 2023-07-30:
tried different powersupply, same error, will try other config this afternoon

Wieringerwerf, 2023-08-25:
retested it with a new Jackpot board... No issues at all. Seems to be hardware related.

Wieringerwerf, 2023-08-25:
Can close this thread.

closed 2023-08-25 by Wieringerwerf