bdring/FluidNC #989: Problem: when no_pin is given endstop is triggered but no error is given

erikcramerjr, 2023-08-06:
### Wiki Search Terms

Six pack, no pin  

### Controller Board

6-pack

### Machine Description

XYYZ homing on all axis  TMC5160 Drivers

### Configuration file

```yaml
motor0:
      limit_neg_pin: gpio.4:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN

    motor0:
      limit_neg_pin: gpio.4:high
      limit_pos_pin: NO_PIN
      limit_all_pin: 

board: 6 Pack
name: 6 Pack TMC5160  XYYZ homing
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.27:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
                      
        r_sense_ohms: 0.110
        run_amps: 0.75
        hold_amps: 0.50
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
[... 497 lines cut ...]
[MSG:DBG:  Y Neg Limit 0]
```

### What happened?

When you don't set one of the pins, and don't set NO-PIN, no error is given during start-up but the endstop for the motor is set to high automatically. What i would expect that would happen is a error at start up that the config is not good, same as when you use a pin twice.

  motor0:
      limit_neg_pin: gpio.4:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
VS

    motor0:
      limit_neg_pin: gpio.4:high
      limit_pos_pin: NO_PIN
      limit_all_pin: 

### GCode File

no gcode used

### Other Information

.

bdring, 2023-08-06:
# 1

I don't understand your "What Happened?"

# 2

If I start a config file like yours ...

```yaml
motor0:
      limit_neg_pin: gpio.4:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN

    motor0:
      limit_neg_pin: gpio.4:high
      limit_pos_pin: NO_PIN
      limit_all_pin: 

board: 6 Pack
name: 6 Pack TMC5160  XYYZ homing
...
```

I get a bunch of errors that you don't. You must show start messages from the included config or I am wasting my time.

```
[MSG:WARN: Ignored key motor0]
[MSG:ERR: Skipping key limit_neg_pin indent 6 this indent 0]
[MSG:ERR: Skipping key limit_pos_pin indent 6 this indent 0]
[MSG:ERR: Skipping key limit_all_pin indent 6 this indent 0]
[MSG:ERR: Skipping key motor0 indent 4 this indent 0]
[MSG:ERR: Skipping key limit_neg_pin indent 6 this indent 0]
[MSG:ERR: Skipping key limit_pos_pin indent 6 this indent 0]
[MSG:ERR: Skipping key limit_all_pin indent 6 this indent 0]
[MSG:INFO: Machine 6 Pack StepStick XYZ]
[MSG:INFO: Board 6 Pack]
```
# 3

If you don't set a pin to NO_PIN it will do it for you.

This ...

```
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: 
```

is the same as this ...

```
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
```
Both result in this

```
  motor0:
    limit_neg_pin: NO_PIN
    limit_pos_pin: NO_PIN
    limit_all_pin: NO_PIN
```

This is true for all pins that are not required.

I use the PWM spindle for example

```
pwm:
  output_pin: NO_PIN
```

or 

```
pwm:
  output_pin: NO_PIN
```
will give a error like this and force a default config.

```
[MSG:ERR: PWM output pin not defined]
```

# 4

Pins can have attributes. If you do not provide them defaults will be used. They is active high and no pull up or pull down.

MitchBradley, 2023-08-06:
Fixed by #991

erikcramerjr, 2023-08-07:
Do you still need the start messages, or can i mark it as fixed ?

commit a5bfcb08 (2023-08-07): Merge pull request #991 from bdring/Fix989

Fix #989 - treat empty pin specification like NO_PIN
closed 2023-08-07 by MitchBradley