bdring/FluidNC #990: Problem: When homing Y axis with tmc 5160 and 2 motors motors dont home

erikcramerjr, 2023-08-06:
### Wiki Search Terms

motor TMC hangs 5160

### Controller Board

6-pack

### Machine Description

XYYC 5160 config

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack TMC5160  XYYZ homing
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.27:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        r_sense_ohms: 0.110
        run_amps: 0.75
        hold_amps: 0.50
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        cs_pin: i2so.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
[... 570 lines cut ...]
[MSG:DBG: Homing limited X]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from 149.000,150.000,150.000]
[MSG:DBG: Planned move to 150.000,150.000,150.000 @ 100.000]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was 150.000,150.000,150.000]
[MSG:DBG: mpos becomes 150.000,150.000,150.000]
[MSG:DBG: mpos transformed 150.000,150.000,150.000]
[MSG:DBG: X Axis StealthChop]
[MSG:DBG: CHOPCONF: 0x0x13408158]
[MSG:DBG: COOLCONF: 0x0x0]
[MSG:DBG: THIGH: 0x0x0]
[MSG:DBG: TCOOLTHRS: 0x0x0]
[MSG:DBG: GCONF: 0x0x4]
[MSG:DBG: PWMCONF: 0x0xc40c001e]
[MSG:DBG: IHOLD_IRUN: 0x0x150e]
[MSG:DBG: Homing done]
[MSG:DBG: Homing Cycle Y]
[MSG:DBG: Y Axis StealthChop]
[MSG:DBG: CHOPCONF: 0x0x13408158]
[MSG:DBG: COOLCONF: 0x0x0]
[MSG:DBG: THIGH: 0x0x0]
[MSG:DBG: TCOOLTHRS: 0x0x0    (this is the last message after this nothing happens)

MitchBradley, 2023-08-06:
The hang looks like a problem that we are tracking down on the Discord thread "help-forum / Low memory warning"

bdring, 2023-08-06:
Try it using FluidTerm and WiFi off

erikcramerjr, 2023-08-13:
Without the wifi how do you control the output, I'm sorry but i've only used the web interface sofar

bdring, 2023-08-13:
> Without the wifi how do you control the output, I'm sorry but i've only used the web interface sofar

How does this question relate to the topic of this issue? If it does not relate, start a new issue and be more clear about what thing like output mean.

erikcramerjr, 2023-08-13:
> > Without the wifi how do you control the output, I'm sorry but i've only used the web interface sofar
> 
> How does this question relate to the topic of this issue? If it does not relate, start a new issue and be more clear about what thing like output mean.

You asked me to test without Wifi, How do you control the motors interface without the web interface..

bdring, 2023-08-13:
Using the serial port

erikcramerjr, 2023-11-23:
Issue was tested and could not be reproduced anymore in version v3.7.9 can be marked as resolved.

closed 2023-11-23 by erikcramerjr