bdring/FluidNC #993: Feature: Added ULN2003 stepper motor drive board control code

Walkerlkr, 2023-08-10:
### Machine Context

Based on the WallPlotter structure, I am making a convenient four-motor wall painting robot. If I want to apply it on any surface, I want to integrate the stepper motor with the main control, so that the four corners are fixed and the motor moves with the pen, so I can only use the smaller 28byj48 stepper motor.

### Feature Description

Can we add ULN2003 stepper motor drive board +28byj-48-5v stepper motor to control it?

### Other Approaches

Although it is possible to use a 0.9 degree round cake motor, I feel that the cost will be very high.

### How I Can Help

I can finance you to buy 28byj48 stepper motor +ULN2003 drive board.

breiler, 2023-08-10:
Don't bother with the ULN2003, use a step/direction based stepper motor driver instead that is already supported by the firmware.

The developers will probably not prioritize adding support for ULN2003 as there are so many cheap hardware alternatives that are already supported by the ordinary "standard stepper"-method using step/direction motor drivers: http://wiki.fluidnc.com/en/config/axes#standard-stepper

For instance, I have run 28byj-48-5v with the EasyDriver v4 in the past. The precision I got from those motors are pretty bad though (a lot of backlash).

bdring, 2023-08-10:
We have a unipolor motor. It is not documented in the wiki yet.

If you donate to the project, I will document it.

https://github.com/bdring/FluidNC/blob/main/FluidNC/src/Motors/UnipolarMotor.h

I used it back in the GRBL_ESP32 days for this project.

https://www.youtube.com/watch?v=XTdoDOojJFQ

Walkerlkr, 2023-08-14:
> 我们有一个单极电机。维基百科上还没有记录它。
> 
> 如果您向该项目捐款，我会记录下来。
> 
> https://github.com/bdring/FluidNC/blob/main/FluidNC/src/Motors/UnipolarMotor.h
> 
> 我在 GRBL_ESP32 天的这个项目中使用了它。
> 
> https://www.youtube.com/watch?v=XTdoDOojJFQ

Walkerlkr, 2023-08-14:
> 我们有一个单极电机。维基百科上还没有记录它。
> 
> 如果您向该项目捐款，我会记录下来。
> 
> https://github.com/bdring/FluidNC/blob/main/FluidNC/src/Motors/UnipolarMotor.h
> 
> 我在 GRBL_ESP32 天的这个项目中使用了它。
> 
> https://www.youtube.com/watch?v=XTdoDOojJFQ

I am a Chinese user without an American credit card, can I ask the project to support Alipay sponsorship?

closed 2023-08-14 by Walkerlkr

bdring, 2023-08-14:
Thanks, but we only support these donation methods.

http://wiki.fluidnc.com/#donations

Walkerlkr, 2023-08-15:
I'm really sorry that donation is not supported in my area. I hope I can help you in other places.