bdring/FluidNC #1001: Problem: Axis TMC driver not detected - expected 0x0x11 got 0x0xff]

mkubi, 2023-08-16:
### Wiki Search Terms

 TMC driver not detected 

### Controller Board

4 Axis SPI Daisy Chain Controller

### Machine Description

Diode laser in XYYZ configuration

### Configuration file

```yaml
board: 4 Axis SPI Daisy
name: 4 Axis SPI Daisy Test
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        spi_index: 2
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.15
        direction_pin: gpio.2
        disable_pin: NO_PIN

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        spi_index: 3
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26
        disable_pin: NO_PIN

  a:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2130:
        spi_index: 4
        r_sense_ohms: 0.110
        run_amps: 0.750
        hold_amps: 0.750
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.33
        direction_pin: gpio.32
        disable_pin: NO_PIN

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false
```

### Startup Messages

```text
Grbl 3.7 [FluidNC v3.7.6 (wifi) '$' for help]
$ss
[MSG:INFO: FluidNC v3.7.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 4 Axis SPI Daisy Test]
[MSG:INFO: Board 4 Axis SPI Daisy]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.110]
[MSG:INFO: Axis Y (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:2 R:0.110]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:3 R:0.110]
[MSG:INFO: Axis A (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:4 R:0.110]
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0xff]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0xff]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0xff]
[MSG:ERR: A Axis TMC driver not detected - expected 0x0x11 got 0x0xff]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is ]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
```

### What happened?

issue hapened after upgrading  firmware from GRBL_ESP32 to FLUIDNC

bdring, 2023-08-16:
I just ran my unit on v3.7.5 and it works fine. 

Sorry for the stupid question, but is main power on? . . . not just USB.

```
[MSG:INFO: FluidNC v3.7.5]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:TMC2130_SPI_Daisy.yaml]
[MSG:INFO: Machine 4 Axis SPI Daisy Test]
[MSG:INFO: Board 4 Axis SPI Daisy]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.110]
[MSG:INFO: Axis Y (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.110]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:3 R:0.110]
[MSG:INFO: Axis A (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:4 R:0.110]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.5 (wifi) '$' for help]
```

Here is my config

```yaml
board: 4 Axis SPI Daisy
name: 4 Axis SPI Daisy Test
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
[... 148 lines cut ...]

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false
```

mkubi, 2023-08-17:
I applied 12v power, no USB is connected.  I measured 12.4V between input terminas, however directly on driver there is just 3.06V

mkubi, 2023-08-17:
Additional tests, flashed - 2nd ESP32 with old "GRBL_ESP32" firmware and motors are working. When swithing to FLUIDNC (tried 3.7.5 and 3.7.6) no. Propably some not supported drivers in FLUIDNC?

bdring, 2023-08-17:
You can send $MI at any time rerun the test.. Try that a few times.

mkubi, 2023-08-17:
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:extremeAP]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.54]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok

mkubi, 2023-08-17:
It's interesting even $MI says driver not detected, but axes are moving when 1st test passed

mkubi, 2023-08-17:
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:extremeAP]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.54]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:DBG: config_bdring.yaml hash "4496817D71AEF660C0494E8A284F466562FA6C69BA43EE7B9FB514D5A8600043"]
[MSG:DBG: index.html.gz hash "7A9C15E55C5CC41861C898408921F1A0ABCA5FE91381867F024BA92E00DC9E40"]
[MSG:DBG: config.yaml hash "F71A10C06657D7D71CD0DB1A4204480FF285A8CA53DC427FABBCFEBC2129858A"]
[MSG:INFO: Telnet started on port 23]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
ok
$mi
[MSG:ERR: X Axis TMC driver not detected - expected 0x0x11 got 0x0x0]
[MSG:INFO: Y Axis driver test passed]
[MSG:DBG: Y Axis Coolstep]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:DBG: Y2 Axis Coolstep]
[MSG:INFO: Z Axis driver test passed]
[MSG:DBG: Z Axis Coolstep]
ok

bdring, 2023-08-17:
I am not sure what is going on. Both Grbl_ESP32 and FluidNC use the same 3rd party library (TMCStepper) for all motor communications. My 4x SPI Daisy controller always reports on startup an after each $MI.

I initially thought that turn on timing might com into play, but the $MI should not have timing issues because that is well after turn on.

It is possible that the response is more sensitive on your controller. With diasy chain all responses have to go through all motors. Try moving the motors a small amount. If they move change the microstepping and do the same move to see if the setting has changed to move amount.

mkubi, 2023-08-17:
seems the microsteping has no effect. Looks like motor is moving same amount no matter what I configure.

FabioRM, 2023-08-18:
My personal experience with Makerbase MKS DLC32 v2.1 and three TMC2209 drivers: whenever I add a third TMC2209 driver no one driver is recognized. I solved using a smaller 470 ohm between TX and RX lines. Despite being NOT STANDARD nor referenced on the TMC2209 datasheet, it has been now running since 3 weeks without issues. It sould be nice if someone can double check this behaviour and report back.

mkubi, 2023-08-30:
I don't think this is HW related, reverted back to GRBL_ESP32 and it works fine.

bdring, 2024-12-12:
Closing inactive issue

closed 2024-12-12 by bdring