bdring/FluidNC #1005: Feature: New addition to the Existing Hardware page?

V1EngineeringInc, 2023-08-25:
### Machine Context

Open source, yes, https://oshwlab.com/allted/4layer-desktop-rc1_copy, but a link that will stay current with any revisions is here, https://docs.v1e.com/electronics/jackpot/#license-and-source

### Feature Description

Wiki - I can write up a page, link or cut and paste from here, https://docs.v1e.com/electronics/jackpot/

### Other Approaches

Project supporter - I think I qualify, not sure what the standards are for this one currently I have a small monthly reoccurring donation.

For sale - yes, https://www.v1e.com/products/jackpot-cnc-controller

Description, 6x TMC2209 driver ports, 7 inputs, 2x 5V outputs, 2x input level outputs, one expansion module socket, and a MicroSD card slot.

### How I Can Help

[image]
 New picture just in case.

bdring, 2023-08-25:
You can add it.

http://wiki.fluidnc.com/en/wiki_contributions

V1EngineeringInc, 2023-08-29:
Okay,  I think I am ready. Did you want me to make a wiki page on the fluid site or link my existing page to the existing hardware blurb?

I will link mine, if you want me to change it, I think I can pretty quickly make a page on your wiki. Not sure which way you wanted it.

V1EngineeringInc, 2023-08-29:
Oh and I am not sure about the order you want it in so I will just put it above the "do not use".

MitchBradley, 2023-08-29:
If it were me, I would put some summary information in the FluidNC wiki and include a link to where a user can find detailed information.

V1EngineeringInc, 2023-08-29:
Done, thanks!

closed 2023-08-29 by V1EngineeringInc