bdring/FluidNC #1007: Problem: Macros pin error

FlashMiks, 2023-08-30:
### Wiki Search Terms

macros pin trouble, macros, macros button

### Controller Board

DIY ESP32-WROOM

### Machine Description

3-axies simple test board

### Configuration file

```yaml
name: "REPUBLIC CNC"
board: "ESP32 Dev Controller V4"

start:
  must_home: false

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 630
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 450
    max_travel_mm: 450
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.17:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 630
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 450
    max_travel_mm: 300
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false

    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 630
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 20
    max_travel_mm: 27
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: true

    motor0:
      limit_all_pin: gpio.16:low:pu
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.32:low:pu
  macro1_pin: gpio.25:low:pu
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: G0X0Y0
  macro1: G0X15Y15
  macro2:
  macro3:
  after_homing:
  after_reset:
  after_unlock:

coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN
        
Laser:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: gpio.22
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 0=3.500% 1000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:3axis_v4h2_test.yaml]
[MSG:INFO: Machine REPUBLIC CNC]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-27.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: macro0_pin gpio.32:low:pu]
[MSG:INFO: macro1_pin gpio.25:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.22 Out:gpio.2 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:Inmart2]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 10.10.101.225]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.6 (wifi) '$' for help]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008d038  PS      : 0x00060930  A0      : 0x8010f61d  A1      : 0x3ffb20c0
A2      : 0x00030a9b  A3      : 0x00030a97  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x801eab20  A9      : 0x3ffb2090
A10     : 0x00000001  A11     : 0x00000020  A12     : 0x00000003  A13     : 0x00000000
A14     : 0x3ffb8954  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00030a9b  LBEG    : 0x4008b650  LEND    : 0x4008b666  LCOUNT  : 0xffffffff

Backtrace: 0x4008d035:0x3ffb20c0 0x4010f61a:0x3ffb20d0 0x40119511:0x3ffb20f0 0x400e4662:0x3ffb2110 0x400e481d:0x3ffb2180 0x400f1e61:0x3ffb21b0 0x400f1ee9:0x3ffb21e0 0x400f2266:0x3ffb2200 0x400f2515:0x3ffb2230 0x400e55ed:0x3ffb2260 0x4011af91:0x3ffb2290

ELF file SHA256: 1d1d266e85df2483

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Inmart2]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 10.10.101.225]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.6 (wifi) '$' for help]
```

### What happened?

When I tried connect 25 or 32 pin to the ground then board have error "Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled." and reboot. Can you tell me how I can use macros on input pins? What am I doing wrong?

bdring, 2023-08-30:
I tried loading your config and it worked.

```yaml
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:3axis_v4h2_test.yaml]
[MSG:INFO: Machine REPUBLIC CNC]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-27.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: macro0_pin gpio.32:low:pu]
[MSG:INFO: macro1_pin gpio.25:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.22 Out:gpio.2 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:Bill's Wi-Fi Network]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.0.1.171]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.6 (wifi) '$' for help]
```

I'll try looking at the backtrace

bdring, 2023-08-30:
Here is the backtrace

```
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:43
0x4010f61a
C:/Users/barto/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:67
0x40119511
C:/Users/barto/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:89
0x400e4662
C:\Users\barto\Documents\GitHub\FluidNC/FluidNC/src/Configuration/../MyIOStream.h:33
0x400e481d
C:\Users\barto\Documents\GitHub\FluidNC/FluidNC/src/Machine/Macros.cpp:15
0x400f1e61
C:\Users\barto\Documents\GitHub\FluidNC/FluidNC/src/Protocol.cpp:1110
0x400f1ee9
C:\Users\barto\Documents\GitHub\FluidNC/FluidNC/src/Protocol.cpp:805
0x400f2266
C:\Users\barto\Documents\GitHub\FluidNC/FluidNC/src/Protocol.cpp:339
0x400f2515
C:\Users\barto\Documents\GitHub\FluidNC/FluidNC/src/Protocol.cpp:269
0x400e55ed
C:\Users\barto\Documents\GitHub\FluidNC/FluidNC/src/Main.cpp:147
0x4011af91
C:/Users/barto/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

bdring, 2023-08-30:
I added a section to the wiki about understanding backtraces.

http://wiki.fluidnc.com/en/installation#understanding-backtraces

I am not sure what is going on, but my first guess would be a bouncy switch. Is the switch good quality and does the circuit have filtering like a simple RC filter.

http://wiki.fluidnc.com/en/support/controller_design_guidelines#rc-filtering

You could try running to the macro via command to make sure that works.

http://wiki.fluidnc.com/en/config/macros#run-via-command

FlashMiks, 2023-09-01:
I tried to assemble an empty circuit, as in the picture:

https://drive.google.com/file/d/1Hk_2rQ3_JJOxXEhRf0cXhIB0MjvvLw91/view?usp=sharing

But the result is still the same. When I close the pin to GND - I get an error and reboot

Grbl 3.7 [FluidNC v3.7.6 (wifi) '$' for help]
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:3axis_v4h2_test.yaml]
[MSG:INFO: Machine REPUBLIC CNC]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-27.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: macro0_pin gpio.32:low:pu]
[MSG:INFO: macro1_pin gpio.25:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:gpio.22 Out:gpio.2 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:Inmart2]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connection failed]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.6 (wifi) '$' for help]
[MSG:INFO: Running macro0]
[MSG:INFO: Running macro macro0: G0X0Y0]
ok
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008d038  PS      : 0x00060930  A0      : 0x8010f61d  A1      : 0x3ffb20c0
A2      : 0x02030a9b  A3      : 0x02030a97  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x801eab20  A9      : 0x3ffb2090
A10     : 0x00000001  A11     : 0x00000020  A12     : 0x00000000  A13     : 0x3ffcb7bd
A14     : 0x3ffb8954  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
EXCVADDR: 0x02030a9b  LBEG    : 0x4008b650  LEND    : 0x4008b666  LCOUNT  : 0xffffffff

Backtrace: 0x4008d035:0x3ffb20c0 0x4010f61a:0x3ffb20d0 0x40119511:0x3ffb20f0 0x400e4662:0x3ffb2110 0x400e481d:0x3ffb2180 0x400f1e61:0x3ffb21b0 0x400f1ee9:0x3ffb21e0 0x400f2266:0x3ffb2200 0x400f2515:0x3ffb2230 0x400e55ed:0x3ffb2260 0x4011af91:0x3ffb2290

ELF file SHA256: 1d1d266e85df2483

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.6]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Inmart2]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.10.101.225]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

PR #1010 (2023-09-02): Fix crash when running numbered macros
Fixes #1007
closed 2023-09-05 by FlashMiks