bdring/FluidNC #1009: Problem: 3.7.6 fail with my config, but 3.7.5 ok

andrewthecat, 2023-09-01:
### Wiki Search Terms

configuration format changelog

### Controller Board

CNC32 ( MKS DLC32 schematics )

### Machine Description

generic 3018

### Configuration file

```yaml
board: MKS-DLC32
name: K40 MOD
meta: 2022-12-27 by Tong

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 254
  pulse_us: 6
  dir_delay_us: 10
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 1500
    acceleration_mm_per_sec2: 500
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:high

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 1500
    acceleration_mm_per_sec2: 500
    max_travel_mm: 183
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 6000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
[... 146 lines cut ...]
[MSG:INFO: Axis Z (-80.000,0.000)]                                                                                                                                                                                
[MSG:INFO:   Motor0]                                                                                                                                                                                              
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]                                                                                                                                                   
[MSG:INFO:  Z Pos Limit gpio.34:low]                                                                                                                                                                              
[MSG:INFO: feed_hold_pin gpio.4:low:pu]                                                                                                                                                                           
[MSG:INFO: cycle_start_pin gpio.22:low:pu]                                                                                                                                                                        
[MSG:INFO: macro0_pin gpio.33:low:pu]                                                                                                                                                                             
[MSG:INFO: Kinematic system: Cartesian]                                                                                                                                                                           
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.32 Dir:NO_PIN Freq:5000Hz Period:8191]                                                                                                                                 
[MSG:INFO: Using spindle PWM]                                                                                                                                                                                     
[MSG:INFO: Connecting to STA SSID:andrewthecat_2G]                                                                                                                                                                
[MSG:INFO: Connecting.]                                                                                                                                                                                           
[MSG:INFO: Connecting..]                                                                                                                                                                                          
[MSG:INFO: Connecting...]                                                                                                                                                                                         
[MSG:INFO: Connected - IP is 192.168.1.129]                                                                                                                                                                       
[MSG:INFO: WiFi on]                                                                                                                                                                                               
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]                                                                                                                                                        
[MSG:INFO: SSDP Started]                                                                                                                                                                                          
[MSG:INFO: HTTP started on port 80]                                                                                                                                                                               
[MSG:INFO: Telnet started on port 23]
```

### What happened?

I updated firmware to version 3.7.6. After that, the settings page in webui stopped opening because the controller is returning an invalid(truncated) json. Machine does not respond to home. Moving works, but not correctly. I tried flash erase and flash again, same result. Then I rolled back to 3.7.5 and the problem disappeared. https://imgur.com/a/JceVPVt

andrewthecat, 2023-09-06:
3.7.8 works but Z doesn't move until cycle: 0. So I changed cycle to 3-2-1 (XYZ) and it worked

MitchBradley, 2023-09-06:
Did you read what the wiki says about homing?

[image]

andrewthecat, 2023-09-06:
Yes. It is not written here that this will lead to incorrect movement of the axis (Moves less than requested). I used commands $HZ,  $HX, HY, but that didn't solve the problem

closed 2024-12-19 by bdring