bdring/FluidNC #1012: Feature: GCODE stream from telnet or websocket or serial

kshitijgarg2609, 2023-09-04:
### Machine Context

I am trying to make desktop application in java swing, used telnet and websocket protocol. I have send .nc file line by line and I am getting ok but sometimes late ok reply and I also found one more thing that streaming line by line GCODES in IDLE mode gives inaccuracy.

### Feature Description

simply just need one example for sending gcode line by line and result in acuracy.

### Other Approaches

I can see the starting nc file from SD works fine but want to know how it works the same command stream I want to do via telnet or serial or websocket

### How I Can Help

I am trying to stream gcode in after typing $X but getting errors, but this mode IDLE, I came to know the state by typing '?' so how to get output like this <RUN|Wpos and proper way of streaming gcode. Or I missed any documentation. Please do help me for this.

breiler, 2023-09-04:
> I also found one more thing that streaming line by line GCODES in IDLE mode gives inaccuracy.

What do you mean by inaccuracy?

First you need to read about the GRBL streaming protocol:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#writing-an-interface-for-grbl

There are plenty of senders that does this, I suggest that you look at them. For Java specifically you  can look at UGS. There are a couple of layers with their own thread workers and can be quite complex  to follow. But here are the key components:
* Lowest layer for TCP connections: https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/connection/TCPConnection.java
* On top of that there is a connection which keeps track of the commands sent/received: https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/communicator/BufferedCommunicator.java
* And then we have the protocol implementation here: https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/firmware/fluidnc/FluidNCController.java

kshitijgarg2609, 2023-09-04:
There is no inaccuracy in fluidnc, inaccuracy is occuring when I am streaming commands over telnet protocol or websocket protocol. My motive is to stream .nc file instead of uploading. First I am sending $X to unlock then sending command line by line, sending one command and getting ok as response but I am sending next line only if I am receiving ok response. I just that whether this is the only way to stream or any document I missed.

breiler, 2023-09-04:
Did you read the link describing the GRBL streaming protocol?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#writing-an-interface-for-grbl

Is there a reason that you don't want to use an existing sender?
http://wiki.fluidnc.com/en/support/gcode_senders

crxksa, 2023-09-08:
> There is no inaccuracy in fluidnc, inaccuracy is occuring when I am streaming commands over telnet protocol or websocket protocol. My motive is to stream .nc file instead of uploading. First I am [...]

code you post the jave code here

MitchBradley, 2023-09-08:
The java code is in the UGS repository.  Search for "ugs github" to find it.