bdring/FluidNC #1016: Problem:

crxksa, 2023-09-06:
### Wiki Search Terms

TMC2209 4 Axis Controller, config.yaml not working

### Controller Board

TMC2209 4 Axis Controller

### Machine Description

hi,
I have made this board, I flashed the latest fluidnc-v3.7.8 and the appropriate yaml file wiki, but no movement.

http://wiki.fluidnc.com/en/hardware/official/TMC2209_4_Axis 

please help!!!

### Configuration file

```yaml
name: "TMC2209 XYZA  Huany"
board: "FluidNC 4X 2209"

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 240

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        uart:
          txd_pin: gpio.16
          rxd_pin: gpio.4
          rts_pin: NO_PIN
          cts_pin: NO_PIN
          baud: 115200
          mode: 8N1

        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0

  y:
    steps_per_mm: 53.33
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 950
    max_travel_mm: 410
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 240

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 200
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 200
      feed_mm_per_min: 100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34:low
      limit_all_pin: NO_PIN
      tmc_2209:
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8

  a:
    steps_per_mm: 400
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 200
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 200
      feed_mm_per_min: 100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      tmc_2209:
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

10V:
  forward_pin: gpio.25
  reverse_pin: gpio.26
  pwm_hz: 5000
  output_pin: gpio.27
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
no error show
```

### User Interface Software

webui

### What happened?

no movement

MitchBradley, 2023-09-06:
1. Please use FluidTerm for debugging, instead of WebUI.  It can be difficult to see error messages in WebUI because some types of errors can happen before WebUI can start.
2. In the Machine Description section, you told us only about the controller board, not the complete machine. Provide photos of your machine and its connections.
3. Please provide the startup messages, even though there are no errors.  If there are errors, you should resolve them before filing an issue.
5. Tell us what you did to try and make the machine move.  "no movement" does not tell us what you did.  The best way to test motion is to issue a GCode command like "G0 X100" in FluidTerm.

bdring, 2023-09-06:
1. Does the power LED come on?
2. Do the motors lock when power is applied.
3. Startup messages are critical. You may have no errors, but you might be accidentally using a default config. $ss will show the startup messages in the WebUI, but Mitch is right. You should be using FluidTerm to debug.

crxksa, 2023-09-07:
> 1. Please use FluidTerm for debugging, instead of WebUI.  It can be difficult to see error messages in WebUI because some types of errors can happen before WebUI can start.

**[MSG:INFO: FluidNC v3.7.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key uart]
[MSG:ERR: Skipping key txd_pin indent 10 this indent 8]
[MSG:ERR: Skipping key rxd_pin indent 10 this indent 8]
[MSG:ERR: Skipping key rts_pin indent 10 this indent 8]
[MSG:ERR: Skipping key cts_pin indent 10 this indent 8]
[MSG:ERR: Skipping key baud indent 10 this indent 8]
[MSG:ERR: Skipping key mode indent 10 this indent 8]
[MSG:ERR: Initialization error at /axes/x/motor0/tmc_2209: TrinamicUartDriver must set uart_num: ]
[MSG:ERR: Initialization error at /axes/y/motor0/tmc_2209: TrinamicUartDriver must set uart_num: ]
[MSG:ERR: Initialization error at /axes/z/motor0/tmc_2209: TrinamicUartDriver must set uart_num: ]
[MSG:ERR: Initialization error at /axes/a/motor0/tmc_2209: TrinamicUartDriver must set uart_num: ]
[MSG:ERR: Configuration is invalid]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:ERR: Initialization error at /stepping: I2SO bus must be configured for this stepping type]
[MSG:ERR: Configuration is invalid]
[MSG:INFO: Connecting to STA SSID:hamdan wifi main]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.46]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.8 (wifi) '$' for help]**

> 2. In the Machine Description section, you told us only about the controller board, not the complete machine. Provide photos of your machine and its connections.

**just the TMC2209 4 Axis Controller with the 2 stepper motor on the x and y, for now.**

> 3. Please provide the startup messages, even though there are no errors.  If there are errors, you should resolve them before filing an issue.

** above mentioned**

> 4. Tell us what you did to try and make the machine move.  "no movement" does not tell us what you did.  The best way to test motion is to issue a GCode command like "G0 X100" in FluidTerm.

**Grbl 3.7 [FluidNC v3.7.8 (wifi) '$' for help]
G0 X100
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008b7c8  PS      : 0x00060f30  A0      : 0x800eeb94  A1      : 0x3ffb1ff0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000048  A5      : 0x00000000
A6      : 0x00003a97  A7      : 0x00000004  A8      : 0x800f2550  A9      : 0x3ffb2200
A10     : 0x00000000  A11     : 0x3ffcbe94  A12     : 0x00000000  A13     : 0x3ffcb729
A14     : 0x00060f20  A15     : 0x00000001  SAR     : 0x00000014  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x4008b7c8  LEND    : 0x4008b7d3  LCOUNT  : 0x00000003

Backtrace: 0x4008b7c5:0x3ffb1ff0 0x400eeb91:0x3ffb2000 0x400e57a5:0x3ffb20b0 0x400ddb4d:0x3ffb20d0 0x400de4f2:0x3ffb20f0 0x400e57e6:0x3ffb2110 0x400db3fa:0x3ffb2130 0x400f16a9:0x3ffb2200 0x400f254d:0x3ffb2230 0x400e565f:0x3ffb2260 0x4011b0e9:0x3ffb2290

ELF file SHA256: feacfa732f9d9de9

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4**

crxksa, 2023-09-07:
> 1. Does the power LED come on?
Yes

> 2. Do the motors lock when power is applied.
No

> 3. Startup messages are critical. You may have no errors, but you might be accidentally using a default config. $ss will show the startup messages in the WebUI, but Mitch is right. You should be using FluidTerm to debug.

**[MSG:INFO: FluidNC v3.7.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key uart]
[MSG:ERR: Skipping key txd_pin indent 10 this indent 8]
[MSG:ERR: Skipping key rxd_pin indent 10 this indent 8]
[MSG:ERR: Skipping key rts_pin indent 10 this indent 8]
[MSG:ERR: Skipping key cts_pin indent 10 this indent 8]
[MSG:ERR: Skipping key baud indent 10 this indent 8]
[MSG:ERR: Skipping key mode indent 10 this indent 8]
[MSG:ERR: Initialization error at /axes/x/motor0/tmc_2209: TrinamicUartDriver must set uart_num: ]
[MSG:ERR: Initialization error at /axes/y/motor0/tmc_2209: TrinamicUartDriver must set uart_num: ]
[MSG:ERR: Initialization error at /axes/z/motor0/tmc_2209: TrinamicUartDriver must set uart_num: ]
[MSG:ERR: Initialization error at /axes/a/motor0/tmc_2209: TrinamicUartDriver must set uart_num: ]
[MSG:ERR: Configuration is invalid]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:ERR: Initialization error at /stepping: I2SO bus must be configured for this stepping type]
[MSG:ERR: Configuration is invalid]
[MSG:INFO: Connecting to STA SSID:hamdan wifi main]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.46]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.8 (wifi) '$' for help]**

crxksa, 2023-09-07:
> I see a lot of error messages in the config.
> 
> I can try and load it after breakfast. 
> Please consider donating to the project so we can continue to provide free support.

Thanks for keeping grbl alive. 
Definitely!

boboxx, 2023-09-07:
There seems to be alot of issues related to config  trim leading/trailing whitespace. would it not make sence to  trim whitespace prior to parsing?

bdring, 2023-09-07:
You need to define a `uart` at the top level for TMC2209. This was changed a few months ago to allow better control of the uart used and allow drivers to share a uart or use separate uarts.

http://wiki.fluidnc.com/en/config/uart_sections

This part of the wiki shows a 2209 example.

http://wiki.fluidnc.com/en/config/axes#tmc2209

Here is your corrected config

```yaml
name: "TMC2209 XYZA  Huany"
board: "FluidNC 4X 2209"

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

uart1:
  txd_pin: gpio.16
  rxd_pin: gpio.4
  baud: 115200
  mode: 8N1
  
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 240

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 32
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0

  y:
    steps_per_mm: 53.33
    max_rate_mm_per_min: 18000
    acceleration_mm_per_sec2: 950
    max_travel_mm: 410
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 3000
      feed_mm_per_min: 240

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:      
        uart_num: 1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7

  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 200
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 200
      feed_mm_per_min: 100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34:low
      limit_all_pin: NO_PIN
      tmc_2209:        
        uart_num: 1
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8

  a:
    steps_per_mm: 400
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 200
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 200
      feed_mm_per_min: 100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      tmc_2209:
        uart_num: 1
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

10V:
  forward_pin: gpio.25
  reverse_pin: gpio.26
  pwm_hz: 5000
  output_pin: gpio.27
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  off_on_alarm: false
```

crxksa, 2023-09-07:
yes, now it works, its alive!! Thank you so much.

Here is my updated config file, i am not using 10V, so removed it.
Could you please guide me to set up StallGuard for all axises
I made some changes to the below code from this wiki http://wiki.fluidnc.com/en/support/stallguard_tuning

```javascript
name: "TMC2209 XYZA  Huany"
board: "FluidNC 4X 2209"

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

uart1:
  txd_pin: gpio.16
  rxd_pin: gpio.4
  baud: 115200
  mode: 8N1
  
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false

      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        uart_num: 1
        addr: 0
      r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
      
        use_enable: false
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:      
        uart_num: 1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
      
        use_enable: false
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34:low
      limit_all_pin: NO_PIN
      tmc_2209:        
        uart_num: 1
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
      
        use_enable: false
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8

  a:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      tmc_2209:
        uart_num: 1
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
      
        use_enable: false
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
```

when i home y axis 

i get

```
$HY
[MSG:INFO: ALARM: Homing Fail Pulloff]
ALARM:8
[MSG:ERR: Macro can only be used in idle state]
ok
```

bdring, 2023-09-07:
The wiki has everything I know about Stallguard.

The wiki has some information here on that homing alarm.

http://wiki.fluidnc.com/en/config/homing_and_limit_switches#homing-fail-pulloff

I don't know why you got the macro error. I don't see any macros defined.

crxksa, 2023-09-08:
> The wiki has everything I know about Stallguard.
> 
> The wiki has some information here on that homing alarm.
> 
> http://wiki.fluidnc.com/en/config/homing_and_limit_switches#homing-fail-pulloff
> 
> I don't know why you got the macro error. I don't see any macros defined.

Thanks got the Stallguard for all axes.

Here is my current TMC2209 4 Axis Controller config.yaml for anyone who needs it.

```C
name: "TMC2209 XYZA  Huany"
board: "FluidNC 4X 2209"

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 4
  disable_delay_us: 0

uart1:
  txd_pin: gpio.16
  rxd_pin: gpio.4
  baud: 115200
  mode: 8N1
  
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false

      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        uart_num: 1
        addr: 0
      r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
      
        use_enable: false
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0

  y:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: false
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:      
        uart_num: 1
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
      
        use_enable: false
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:        
        uart_num: 1
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
      
        use_enable: false
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8

  a:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 750
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      feed_mm_per_min: 200.000
      seek_mm_per_min: 200.000

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      tmc_2209:
        uart_num: 1
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.000
        hold_amps: 0.500
        microsteps: 16
        stallguard: 15
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StallGuard
      
        use_enable: false
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
   
start:
  must_home: false
```

closed 2023-09-08 by crxksa