bdring/FluidNC #1017: Will M266 be supported sometime ?

Ams-Laser, 2023-09-06:
### Machine Context

Building Embroidery machine that needs M266 to wait for needle up sensor before next move

### Feature Description

the m266 wait for input trigger will permit the next move to happen only when the needle is out of the fabric

### Other Approaches

im currently using a joystick hack to trigger next move on sender, supporting the M266 code will make more easy to implement and further develop embroidery mode

### How I Can Help

i can donate, and help test

breiler, 2023-09-08:
Hi @Ams-Laser, 

I suspect that you mean the Marlin-specific gcode described here:
https://marlinfw.org/docs/gcode/M226.html

In that case I think the answer is "no" (http://wiki.fluidnc.com/en/features/supported_gcodes):
> Our GCode is strongly based on [LinuxCNC](http://linuxcnc.org/docs/stable/html/) ([G](https://linuxcnc.org/docs/2.8/html/gcode/g-code.html) and [M](https://linuxcnc.org/docs/2.8/html/gcode/m-code.html) codes). If you are thinking about suggesting a new command, look to see what LinuxCNC does. Saying "...like Marlin" will get you nowhere.

The M66 command described at LinuxCNC sounds like it would suit you needs: https://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m66.

However it is also requested here https://github.com/bdring/FluidNC/issues/586

I think that we can close this as a duplicate.

bdring, 2023-09-08:
Closing. 

Continue discussion on the other issue.

closed 2023-09-08 by bdring