bdring/FluidNC #1019: Feature: outputs to indicate status:

kami0011, 2023-09-08:
### Machine Context

I want to get output of status like alarm on output pin. How can I define in YAML file 

### Feature Description

I want to get output of status like alarm on output pin. How can I define in YAML file 

### Other Approaches

I want to get output of status like alarm on output pin. How can I define in YAML file 

### How I Can Help

I want to get output of status like alarm on output pin. How can I define in YAML file

MitchBradley, 2023-09-08:
FluidNC currently does not support this feature.  You could do it with an external microprocessor connected via a UART channel.  That microprocessor would have to monitor status reports, looking for Alarm state.

kami0011, 2023-09-08:
Is the any ready made solution from third party

MitchBradley, 2023-09-08:
There is some discussion in the #displays_pendants channel of the FluidNC Discord server.

bdring, 2023-09-08:
See this post.

https://discord.com/channels/780079161460916227/1149787482183974965

closed 2023-09-08 by bdring