bdring/FluidNC #1022: Problem:

strangeweirdo19, 2023-09-11:
### Wiki Search Terms

ARC error , Error 33 ,

### Controller Board

A custom PCB designed using ESP32 38pin that plugs into an Arduino CNC sheild 
Attached below 
[image]

### Machine Description

2 DRV8825 Stepper Drivers (1/32-step) with two Motors in coreXY configuration, RC servo for Z axis ,Limit all switches on both sides of axis (X,Y)

### Configuration file

```yaml
name: "Arduino Sheild"
board: "Arduino Sheild"

stepping:
  engine: RMT
  idle_ms: 25
  dir_delay_us: 1
  pulse_us: 10
  disable_delay_us: 0

kinematics:
  corexy:
  
axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 162.5
    max_rate_mm_per_min: 8000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 300
    homing:
      cycle: 2
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      positive_direction: false
      allow_single_axis: true
      seek_scaler: 5.0
      feed_scaler: 5.0
    
    motor0:
      limit_all_pin: gpio.17:high:pu
      hard_limits: true
      pulloff_mm: 5.000
      stepstick:
        direction_pin: gpio.26
        step_pin: gpio.12
    motor1:
      null_motor:

  y:
    steps_per_mm: 162.5
    max_rate_mm_per_min: 8000
    acceleration_mm_per_sec2: 5000
    max_travel_mm: 300
    homing:
      cycle: 2
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      seek_scaler: 5.0
      feed_scaler: 5.0
[... 910 lines cut ...]
<Run|MPos:2.283,-5.138,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:4.778,-4.932,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:7.363,-4.717,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:9.948,-4.502,-10.000|FS:4243,0|Ov:100,100,100|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:12.538,-4.286,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:15.169,-4.068,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:17.618,-3.865,-10.000|FS:4243,0|WCO:0.000,-5.329,-10.000|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:20.265,-3.643,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:22.748,-3.437,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:25.357,-3.222,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:27.828,-3.015,-10.000|FS:4243,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:30.366,-2.803,-10.000|FS:0,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:32.855,-2.597,-10.000|FS:0,0|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Idle|MPos:33.486,-2.545,-10.000|FS:0,0|Ov:100,100,100|SD:0.29,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:33.486,-2.545,-9.980|FS:2000,0|SD:0.38,/sd/gc/4xiDraw_0003.gcode>
[GC:G1 G54 G17 G21 G91 G94 M5 M9 T0 F2000 S0]
<Run|MPos:33.486,-2.545,-8.490|FS:2000,0|SD:0.48,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:33.486,-2.545,-6.760|FS:2000,0|WCO:0.000,-5.329,-10.000|SD:0.48,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:33.486,-2.545,-5.100|FS:2000,0|SD:0.48,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:33.486,-2.545,-3.400|FS:800,0|SD:0.48,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:33.486,-2.545,-1.730|FS:0,0|SD:0.48,/sd/gc/4xiDraw_0003.gcode>
<Run|MPos:33.486,-2.545,-0.110|FS:0,0|SD:0.48,/sd/gc/4xiDraw_0003.gcode>
<Idle|MPos:33.486,-2.545,0.000|FS:0,0|SD:0.48,/sd/gc/4xiDraw_0003.gcode>
[GC:G1 G54 G17 G21 G91 G94 M5 M9 T0 F3000 S0]
[MSG:ERR: 33 (Gcode invalid target) in /sd/gc/4xiDraw_0003.gcode at line 9]

breiler, 2023-09-11:
I haven't analyzed the gcode in detail, but this is what it looks like when rendering it in NCViewer:

[image]

And this is how it looks in UGS:
[image]

What is it supposed to look like?

strangeweirdo19, 2023-09-11:
[image]
This is how it should look and here is the reupload of the GCODE  
![image](https://github.com/bdring/FluidNC/assets/62467232
[SD_4xiDraw_0003.txt](https://github.com/bdring/FluidNC/files/12574009/SD_4xiDraw_0003.txt)
/e613b9b2-b5a3-4778-ae18-8a4a523ac2f9)
[image]
View From UGS

breiler, 2023-09-11:
I think that the gcode is poorly structured and missing basic setup gcode state. For instance it should at least have G90 for absolute mode and G21 for millimeters in the beginning.

The real problem is that the gcode contains GRBL specific jog commands `$J=G91 G21 Z10 F2000`. These commands was never intended to be a part of a gcode program. By replacing the jog command with this I think it would work better:
```
G91 G0 Z10
G90 G1
```

Example here: [SD_4xiDraw_0003_without_jog.txt](https://github.com/bdring/FluidNC/files/12574508/SD_4xiDraw_0003_without_jog.txt)

In my opinion I don't think that this is a problem with FluidNC but rather an issue with the plugin generating the gcode.

strangeweirdo19, 2023-09-11:
I would like to point that the initial code I had sent already had these changes done yet threw me Error 33 
Even when I manually run these 9 commands via the console error is thrown on Line 9
**Initial Code**
G90
G21
G1 F3000
G1 X33.4878 Y2.7849
G4 P0.30000001192092896
G91 G21 Z10 F2000
G4 P0.30000001192092896
G1 F3000.000000
G2 X32.6692 Y3.7393 I0.7622 J1.4819
G2 X32.83 Y4.7559 I1.0592 J0.3535

breiler, 2023-09-11:
This line will put the controller in "Incremental mode" (G91): 
`G91 G21 Z10 F2000`

But you never change back to "Absolute mode" (G90) afterwards.

Jog commands will will automatically revert back to the gcode state before. But since it isn't valid gcode you can't use it like that.

closed 2023-09-11 by strangeweirdo19