bdring/FluidNC #1026: Problem: Setting up FluidNC for PenPlotter

GordonLutz, 2023-09-15:
### Wiki Search Terms

PenPlotter, Homing Issue

### Controller Board

MKS DLC32

### Machine Description

Bachin TA-4 PenPlotter with 3 Stepper Motors for X,Y,Z. 2 Limit switches for X and Y and DRV8825 driver. 
[image]

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: K40 Laser
meta: (01.01.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 18.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 195.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 3000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 1000
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 2000.000
    max_travel_mm: 300.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 3000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 4000.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 500.000
      seek_mm_per_min: 3000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
```

### Startup Messages

```text
INFO: FluidNC v3.7.8
INFO: Compiled with ESP32 SDK:v4.4.4
INFO: Local filesystem type is littlefs
INFO: Configuration file:MKS_DLC32_21_XYZ.yaml
INFO: Machine K40 Laser
INFO: Board MKS-DLC32 V2.1
INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21
INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12
INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000
INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms
INFO: Axis count 3
INFO: Shared stepper disable I2SO.0
INFO: Axis X (0.000,325.000)
INFO:   Motor0
INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN
INFO:  X Neg Limit gpio.36
INFO: Axis Y (0.000,220.000)
INFO:   Motor0
INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN
INFO:  Y Neg Limit gpio.35
INFO: Axis Z (0.000,80.000)
INFO:   Motor0
INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN
INFO:  Z Neg Limit gpio.34
INFO: macro0_pin gpio.33:low:pu
INFO: Kinematic system: Cartesian
INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191
INFO: Using spindle Laser
INFO: Probe Pin: gpio.22
INFO: STA SSID is not set
INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1
INFO: AP started
INFO: WiFi on
INFO: Captive Portal Started
INFO: HTTP started on port 80
[MSG:INFO: FluidNC v3.7.8]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:MKS_DLC32_21_XYZ.yaml]
[MSG:INFO: Machine K40 Laser]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,325.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok
INFO: Telnet started on port 23
```

### User Interface Software

UGS, ESP3D for FluidNC

### What happened?

Im using the Bachin TA-4. Unfortunately im having trouble getting it to work. I flashed FluidNC onto the MKS DLC32. Then i connected the plotter with the board and the board with my PC. After opening Up universal GCode Sender I at least could control the axis via the Jog Controller. But im having trouble homing the maschine and telling it where the maschine home is. I have 2 Limit Switches Connected that also work since i see the Alarm X and Y go off in UGS when i press them. But whenever im trying to home the maschine does one of two things happens. First the maschine does a real small step into X+ and Y+ and then stops. Or im getting this Error Message in UGS: 

"[MSG:INFO: ALARM: Homing Fail Pulloff]
ALARM:8
INFO: ALARM: Homing Fail Pulloff
ERR: Macro can only be used in idle state"

So the plotter doesnt home and just makes small steps to the top right ignoring limit switches and giving me the error messsage. Another problem I encountered is that i can job the PenHolder left, right up and down but only down for Y20. After that the Plotter ignores other commands in Y- direction i assume because of soft Limits eventho i set the softlimit for Y to 300 and UGS says it it as Y20. When i jog the PenHolder in Y+ Direction it moves but completly ignoring the Y Limit switch and bashing against the frame. 

[image]

The picture shows the error message when trying homing as well as the movement down to Y-20 that worked. Then after the second Y-20 it wont go further but going up to Y + 40 where the build Plate is already over

breiler, 2023-09-15:
I think that the setting you are looking for is the `pulloff_mm`. It is currently set to 2mm, try increasing it:
http://wiki.fluidnc.com/en/config/axes#pulloff_mm

> After that the Plotter ignores other commands in Y- direction i assume because of soft Limits eventho i set the softlimit for Y to 300 and UGS says it it as Y20. 

Indeed, your soft limit settings will prevent the machine from moving outside the configured boundries. But this only work if your machine has been properly homed. So you need to take care of that problem first

GordonLutz, 2023-09-15:
This should be my current config File: 

board: MKS-DLC32 V2.1
name: K40 Laser
meta: (01.01.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 325.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: false
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 157.750
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4

The Plotter still just moves for a short distance up right and im gettign the same alarm warning. I noticed tho that the switches seem to be active all the time in UGS and just go away after i manually trigger them. Can i it be that the switches are inverted? 

I wonder why the hardlimites are turned of in the config tho or doesnt that matter?

GordonLutz, 2023-09-15:
This should be my config file rn: 

`board: MKS-DLC32 V2.1
name: PenPlotter TA-4
meta: (01.01.2022) by Skorpi

kinematics:
  Cartesian:

stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 195
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
  macro2:
  macro3:

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.22
  check_mode_start: true

Laser:
  pwm_hz: 5000
  #L on Beeper / IN on TTL
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%
# 135=0mA 270=5mA 400=10mA 700=16mA
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

# 5,18,19,22,23,25,26,27,32,33,39,I2SO.7
# SDA 0 / SCL 4
`

I noticed that under motor0 the hardlimits where false. In UGS the limit switches got triggerd evertime even tho there where not activated and the alarm only turned of when i manually triggered the switches. I dont know if that is supposed to be like that but i switched the x and y motor to hard limits true and the pulloff distance to 5mm.

Now when i connect to UGS the switch alarm is gone but im getting the Error: 5 when i try homing as well as my Steppers are not reacting to any inputs. In UGS the command gets send and the visualizer says it moved but the plotter didnt move at all in no direction. Before i changed the limit switches to true and the pulloff distance jog worked

am i missing somthing? 
[image]

bdring, 2023-09-15:
Test switches per the wiki and report the results.

http://wiki.fluidnc.com/en/support/help_with_switch_problems

MitchBradley, 2023-09-15:
It is easier to debug with FluidTerm because there is less complexity between you and the controller.  If you send "$message/level=debug" FluidNC will issue messages describing the detailed progress of homing.  You can then send "$H" and watch the progress.

Re: the limit switches got triggerd evertime even tho there where not activated

It is possible that your limit switches are configured incorrectly.  They might need :low specifiers to change the logic sense.  Please perform the switch test as bdring said.  Also please spell-check your postings.

GordonLutz, 2023-09-15:
Sorry for my bad spelling english is not my first language but i try my best. 

@MitchBradley When i tried your "$message/level=debug" solution im was getting the following messages: 

[MSG:DBG: Homing Cycle XY]
[MSG:DBG: Starting from 30.000,10.000,0.000]
[MSG:DBG: Planned move to 35.000,15.000,0.000 @ 424.264]
[MSG:DBG: CycleStop PrePulloff]
[MSG:INFO: ALARM: Homing Fail Pulloff]
ALARM:8
[MSG:ERR: Macro can only be used in idle state]
ok

The results of the limit switch test are the following: 

? command : <Idle|MPos:30.000,10.000,0.000|FS:0,0|Pn:PXYZ>

$Limits (after i manually triggered the X and Y Switches): 
:              xyz         P
:               yz          P
:               yz          P
:                z           P

This led me to believe that my switches are inverted somehow. I modified my config in the following parts:

x:
    steps_per_mm: 100
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 195
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: **100**
      seek_mm_per_min: **200**
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: **gpio.36:low**
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 12000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 220.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: **true**
      mpos_mm: 0.000
      feed_mm_per_min: **100**
      seek_mm_per_min: **200**
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: **gpio.35:low**
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        
So now when i start UGS i only get an Switch Alarm for Z and Probe but i dont know how to get rid of them. Maybe that not important because they dont interfere with anything. Since I set  positive_direction: **true** for the Y Axis my Plotter actually moves to the top left corner where my switches are located when i press Homing. After adjusting a few values the homing seems to work.

Is there a way to ignore the Z and Probe Switches that are apperently active?

MitchBradley, 2023-09-15:
Did  you try adding :low to the Z limit pin and the probe pin?

GordonLutz, 2023-09-15:
Somehow I did not think about that... Thanks @MitchBradley.

closed 2023-09-15 by GordonLutz