bdring/FluidNC #1027: Problem: M8 delay_ms still delays even if M8 is already enabled

AussieMakerGeek, 2023-09-18:
### Wiki Search Terms

M8 Delay

### Controller Board

MKS Tinybee

### Machine Description

Machine independent but this is a converted Gravograph LS800 laser cutter

### Input Circuits

```yaml
N/A
```

### Configuration file

```yaml
board: MKS-Tinybee
name: LS800
meta: 
stepping: 
  engine: I2S_static
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2
  segments: 12

i2so: 
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi: 
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard: 
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  baud: 115200
  mode: 8N1

uart_channel1:
  uart_num: 1

kinematics: 
  Cartesian: 

axes: 
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 59.500000
    max_rate_mm_per_min: 80000.000000
    acceleration_mm_per_sec2: 12000
    max_travel_mm: 610.000000
    soft_limits: true
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 1.000000
      feed_mm_per_min: 1500.000000
      seek_mm_per_min: 3000.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 4.000000
      stepstick: 
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.0
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y: 
    steps_per_mm: 120.000000
    max_rate_mm_per_min: 20000.000000
    acceleration_mm_per_sec2: 2000
    max_travel_mm: 305.000000
    soft_limits: true
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 304.00
      feed_mm_per_min: 1500.000000
      seek_mm_per_min: 3000.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 4.000000
      stepstick: 
        step_pin: I2SO.4
        direction_pin: I2SO.5:low
        disable_pin: I2SO.3
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  z: 
    steps_per_mm: 1200.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 500.000000
    max_travel_mm: 250.000000
    soft_limits: false
    homing: 
      cycle: 0
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 300.000000
      seek_mm_per_min: 500.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: gpio.22
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 15.000000
      stepstick: 
        step_pin: I2SO.7
        direction_pin: I2SO.8:low
        disable_pin: I2SO.6
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  a: 
    steps_per_mm: 337.000000
    max_rate_mm_per_min: 8000.000000
    acceleration_mm_per_sec2: 60.000000
    max_travel_mm: 80.000000
    soft_limits: false
    homing: 
      cycle: -1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 300.000000
      seek_mm_per_min: 500.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000000
      stepstick: 
        step_pin: I2SO.10
        direction_pin: I2SO.11:low
        disable_pin: I2SO.9
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

control: 
  safety_door_pin: gpio.39:low
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36:low
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant: 
  #flood_pin: I2SO.16
  flood_pin: I2SO.17
  mist_pin: I2SO.18
  delay_ms: 7000

probe: 
  pin: NO_PIN
  toolsetter_pin: NO_PIN
  check_mode_start: true

macros: 
  startup_line0: 
  startup_line1: 
  macro0: $H
  macro1: $HZ
  macro2: G91&G0F1000A+10
  macro3: G90&G0X-45&G0X100

start: 
  must_home: true
  deactivate_parking: false
  check_limits: false

parking: 
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000

user_outputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

Laser: 
  pwm_hz: 10000
  output_pin: gpio.2:pd
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000000% 100=95.000000%
  off_on_alarm: true

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
```

### Startup Messages

```text
N/A
```

### User Interface Software

Lightburn

### What happened?

When multiple M8 commands are in the same gcode file, the code is still paused for delay_ms even if M8 is already active.

I'm not sure if this is standard in all gcode processors but I would have thought the state of M8/7 would be tracked so that a subsequent M8/7 would not cause a delay since there is no need for it. Or possibly give me the option to control this behavior's.

I can remove the extra M8 commands but i'd prefer not to have to do manual post processing of gcode.

### GCode File

; LightBurn 1.3.00
; GRBL device profile, current position
; Bounds: X0 Y-3.7 to X12.7 Y0
G00 G17 G40 G21 G54
G91
M4
; Cut @ 5 mm/sec, 90% power
;Initial M8, using delay_ms
M8
G0 X9.626Y-1.477
M3
; Layer C01
G1 X0.147Y0.272S90F300
G1 X0.196Y0.236
;..........
; Cut @ 5 mm/sec, 90% power
;Second M8 still casuses delay_ms but there is no need as M8 is already active
M8
G0 X-6.475Y-0.183
; Layer C00
G1 X-0.033Y-0.322
G1 X-0.093Y-0.301
G1 X-0.147Y-0.272
;.........
M9
G1 S0
M5
; return to starting pos
G0 X-3.188Y1.639
M2

AussieMakerGeek, 2023-09-18:
Originally, I posted this because it's somewhat annoying but there is actually two facets to my problem. I didn't post the original issue here because I thought it was an issue with LightBurn but it would seem that there is something else, probably in FluidNC.

At the same time of this M8 delay, the laser is not turning off.  Initially I expected that I should see a 'G1 S0' type command but if there is no G1 movement of the machine, isn't S0 implied?

Basically, what is happening at the second M8, the laser is staying on for the delay, then moving at the G0 with it still on. The proper behavior should be that if there is no active G1 command, S should be 0 anyway. This is not what appears to be happening.

boboxx, 2023-09-18:
I have also seen strange behaviour with the spindle control (M4/M5), M8/M9 and G4. I'm also seeing a large pause before my spindle is started up or stoped. 

Here is a sample of code producing the "pause" before the spindle startup/shutdown:
( Made using CamBam - http://www.cambam.co.uk )
( FluidNC Post Processor )
( test 9/16/2023 5:37:40 PM )
( T0 : 0.0 )
G21 G90
G0 Z10.0
( T0 : 0.0 )
T0
( Drill1 )
G17
M8 ; Start coolant
M4 ; Start Spindle
S12000
G4 P10 ; Dwell 10 Sec
G0 X154.457 Y287.3997
G0 Z10.0 X154.457 Y287.3997
G1 Z-6.0 F100.0
G0 Z10.0
G0 Z10.0 X372.4208 Y287.3787
G1 Z-6.0
G0 Z10.0
G0 Z10.0 Y66.3787
G1 Z-6.0
G0 Z10.0
G0 Z10.0 X34.3998 Y66.3997
G1 Z-6.0
G0 Z10.0
M9 ; Stop coolant
M5 ; Stop Spindle
G4 P10 ; Dwell 10 Sec

AussieMakerGeek, 2023-09-18:
> I have also seen strange behaviour with the spindle control (M4/M5), M8/M9 and G4. I'm also seeing a large pause before my spindle is started up or stoped.

I'm not sure if the issue you are experiencing is the same as mine. The problem I am having 2 two fold.  The delay I can handle if i have to (and is somewhat expected because i have delay_ms set) but the issue is that at the end of a G1 move, immediately preceding an M8, the laser does not turn off. 

In laser mode, the laser should only be active in G1/2/3 moves but it would seem in this case, it doesn't turn off for some reason.

It would be nice if multiple M8 commands (without M9 between) didn't cause delay_ms to take effect though. Unless someone has delay_ms set, this probably would not even be noticed and would be specific to laser mode.

boboxx, 2023-09-18:
Yes i'm not sure if it's related but it's strange to have the delay before the G4 is called.

How come you are not using M3/M4/M5 to control the laser? T0 would select laser as per your config and M3/M4 would start and M5 Stop
http://wiki.fluidnc.com/en/features/supported_gcodes#M3-Spindle-CW

bdring, 2023-09-18:
Why do you have a G4?

Laser M4 will not turn on until there is non rapid (G0) motion.

boboxx, 2023-09-18:
Sorry for hijacking, I'm using a VFD and I need to make sure the spindle is up to speed before I make my cuts.

AussieMakerGeek, 2023-09-18:
> Sorry for hijacking

Sorry @boboxx, but i think i'd prefer if you have these problems, that it's a different issue.  Your use case and issue is different from mine.

My gcode does use M3 and M5, but laser is treated totally different from a traditional spindle.

AussieMakerGeek, 2023-09-28:
@bdring - Where would you suggest I look into the code where I might find this 'laser on only during G1/2/3' code? I don't mind looking into it myself as I have coding skills but I started looking and just got lost..

MitchBradley, 2023-09-29:
If you want the laser to be on regardless of modal motion state (G1 G2 G3), declare the spindle as pwm: instead of laser:

MitchBradley, 2023-09-29:
The code that implements it is in Gcode.cpp

AussieMakerGeek, 2023-09-29:
Thanks @MitchBradley i'll review it but that is the opposite of what I want.  There is a problem with the laser implementation that leaves the laser on during M8 delay_ms

bdring, 2023-09-29:
I optimized the delays in a new branch. It will not delay if the state does not change. It will not delay for M9.

I was able to repeat the problem, but I think the gcode is wrong. In M3 mode the laser will stay on in G1 mode. You should have an S0, G1 or M5 before the M8.

AussieMakerGeek, 2023-09-30:
Thanks for the reply and effort @bdring. I did look over the source code but got lost... I tried the gcode as suggested and both changing the command to M4 over M3 or putting a G0 before the M8 does indeed prevent the laser from remaining on during the M8 delay_ms

I looked for a setting in Lightburn to force M4 over M3 but there does not seem to be this option, since it's not spindle aware, this does make sense.

Would it not make sense to just ignore rotation direction anyway and make M3/M4 in laser mode universal?

Test gcode:
`G00 G17 G40 G21 G54
G91
M4
; Cut @ 10 mm/sec, 95% power
M8
G0 X0Y0
M4
; Layer C00
G1 X10S95F500
G1 Y-10
G1 X-10
G1 Y10
G1 X4
; Cut @ 10 mm/sec, 90% power
M8
G0 X11Y0
; Layer C01
G1 X10S90
G1 Y-10
G1 X-10
G1 Y10
G1 X1
M9
G1 S0
M5
; return to starting pos
G0 X-16Y0
M2`

or 
`G00 G17 G40 G21 G54
G91
M4
; Cut @ 10 mm/sec, 95% power
M8
G0 X0Y0
M3
; Layer C00
G1 X10S95F500
G1 Y-10
G1 X-10
G1 Y10
G1 X4
; Cut @ 10 mm/sec, 90% power
G0
M8
G0 X11Y0
; Layer C01
G1 X10S90
G1 Y-10
G1 X-10
G1 Y10
G1 X1
M9
G1 S0
M5
; return to starting pos
G0 X-16Y0
M2`

Does not leave the laser on. I further confirmed leaving out the G0 or leaving M3 does leave the laser on during M8 delay_ms with this exact gcode.

bdring, 2023-09-30:
Most people find that m3 works better for cutting and m4  works better for engraving. 

Power drops on sharp corners in m4 mode and may not cut through. 

We will keep our separate M3 and M4 modes.

Have you talked to the LightBurn people?

MitchBradley, 2023-09-30:
Note that Lightburn is paid software so in principle your money should entitle you to some support. FluidNC, on the other hand, is free, but the time that we spend on it is not. E get a lot of requests to use our uncompensated time to work around Lightburn limitations and we are starting to resent that.

bdring, 2023-09-30:
I created a branch to fix your problem, but I don't think it will be merged. Here is the definition of M3 laser. 

Constant laser power mode simply keeps the laser power as programmed, regardless if the machine is moving, accelerating, or **stopped**. This provides better control of the laser state. With a good G-code program, this can lead to more consistent cuts in more difficult materials.

From https://github.com/gnea/grbl/blob/master/doc/markdown/laser_mode.md

Can I close this issue? I have done a lot of free work on this one.

AussieMakerGeek, 2023-10-01:
Thanks @bdring and @MitchBradley That does clear some stuff up.  I was not aware of the differences between M3 and M4 when it comes to laser mode, 

I did initially raise the issue over in the LB forums but it seemed the issue was with FluidNC.  I'll take it back to them to have that resolved.

I guess that means this issue then breaks out in to two different issues - One being probably Lightburn's poor implementation of the M3 mode and not ensuring the laser is off at the end of a cut but the other nice-to-have would still be for the M7/8 commands to be state aware and not delay if it is already enabled, and no delay on M9.

I don't have any issues closing this issue off, I think you've done great to explain the reason it does it and that it's not a fault with FluidNC, but perhaps you would be ok with me adding the coolant/mist state aware function as a feature request?

bdring, 2023-10-01:
The state aware delays have been added to the devt branch and will be in the next release.

Please consider donating to this project.

closed 2023-10-01 by bdring