bdring/FluidNC #1028: Problem: Lightburn software. FluidNC K40 board ver. 7.7

awp0645, 2023-09-19:
### Wiki Search Terms

Followed Suggestion made in  Wiki : REQUESTING HELP FOR FLUIDNC
$Startup/Show $Startup/Show

[MSG:INFO: FluidNC v5.1.0]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:DBG: Running after-parse tasks]
[MSG:INFO: Axes: using defaults]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:DBG: SPI not defined]
[MSG:DBG: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:Vodafone9E0364]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.148]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok

### Controller Board

FluidNC K40 Ver 7.7 with  Lightburn

### Machine Description

Generic K40 laser. No enhancements other than change of controller.

### Configuration file

```yaml
Do not know how to access this
```

### Startup Messages

```text
Waiting for connection...
error:9
G-code locked out during alarm or jog state.
[VER:v5.1 FluidNC v5.1.0:]
[OPT:PHS]
[MSG: Machine: Default (Test Drive)]
[MSG: Mode=STA:SSID=Vodafone9E0364:Status=Connected:IP=192.168.1.148:MAC=A8-42-E3-59-84-20]
ok
$$
$10=0
ok
$22=1
error:3
Grbl '$' system command was not recognized or supported.
[MSG:ERR: No homing cycles defined]
                                                                                                  ?                                                                                                                                           No matter what $ command I input I get                                                                                                         [MSG:DBG:  Y Neg Limit 1]
[MSG:INFO: Limit switch tripped for Y motor 0]
[MSG:DBG:  X Neg Limit 1]
[MSG:INFO: Limit switch tripped for X motor 0]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limi
:               y
[MSG:DBG:  X Neg Limit 1]
[MSG:INFO: Limit switch tripped for X motor 0]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:INFO: Limit switch tripped for Y motor 0]                                                                                                                                                                                                                                                               OR       xy
:              xy
```

### User Interface Software

Web UI  IP address 198.168.1.148

### What happened?

Powered up the machine. Motors lock on but gantry does not move. The following message is shown in the console:

error:9
G-code locked out during alarm or jog state.
[VER:v5.1 FluidNC v5.1.0:]
[OPT:PHS]
[MSG: Machine: Default (Test Drive)]
[MSG: Mode=STA:SSID=Vodafone9E0364:Status=Connected:IP=192.168.1.148:MAC=A8-42-E3-59-84-20]
ok
[MSG:ERR: No homing cycles defined]
ok

### GCode File

Connot see or access gcode file in Lightburn nor can I see where to do so.  

Calling up Machine settings in Lightburn returns a blank page

### Other Information

I needed to replace a stepper motor so I replaced both with Nema17 .09 steps The colour of the wires are different to the wires coming off the board.  I checked with a multimeter and as the motors are locking in when laser powered up I assume the wiring is correct.
I have sent photos of the board and power pack wiring  to Avataar120 who confirms it to be correct.

I am not familiar with Grbl or dealing with errors so my apologies for incorrect or missing information.

awp0645, 2023-09-19:
$Localfs/Show=config.yamlboard: 4_Ava_Shield_7.x_XYname: K40_AvaShieldmeta:stepping:engine: I2S_staticidle_ms: 255pulse_us: 4dir_delay_us: 1disable_delay_us: 0segments: 12i2so:bck_pin: gpio.22data_pin: gpio.21ws_pin: gpio.17spi:miso_pin: gpio.19mosi_pin: gpio.23sck_pin: gpio.18sdcard:cs_pin: gpio.5card_detect_pin: NO_PINfrequency_hz: 8000000kinematics:Cartesian:axes:shared_stepper_disable_pin: NO_PINshared_stepper_reset_pin: NO_PINx:steps_per_mm: 78.739998max_rate_mm_per_min: 20000.000000acceleration_mm_per_sec2: 2000.000000max_travel_mm: 310.000000soft_limits: truehoming:cycle: 1allow_single_axis: truepositive_direction: falsempos_mm: 0.000000feed_mm_per_min: 200.000000seek_mm_per_min: 2500.000000settle_ms: 250seek_scaler: 1.500000feed_scaler: 5.000000motor0:limit_neg_pin: gpio.39limit_pos_pin: NO_PINlimit_all_pin: NO_PINhard_limits: truepulloff_mm: 5.000000stepstick:step_pin: I2SO.2direction_pin: I2SO.1disable_pin: I2SO.0ms1_pin: NO_PINms2_pin: NO_PINms3_pin: I2SO.3reset_pin: NO_PINy:steps_per_mm: 78.739998max_rate_mm_per_min: 20000.000000acceleration_mm_per_sec2: 2000.000000max_travel_mm: 210.000000soft_limits: truehoming:cycle: 1allow_single_axis: truepositive_direction: falsempos_mm: 0.000000feed_mm_per_min: 200.000000seek_mm_per_min: 2500.000000settle_ms: 250seek_scaler: 1.500000feed_scaler: 5.000000motor0:limit_neg_pin: gpio.34limit_pos_pin: NO_PINlimit_all_pin: NO_PINhard_limits: truepulloff_mm: 5.000000stepstick:step_pin: I2SO.6direction_pin: I2SO.5:lowdisable_pin: I2SO.4ms1_pin: NO_PINms2_pin: NO_PINms3_pin: I2SO.7reset_pin: NO_PINz:steps_per_mm: 1000.000000max_rate_mm_per_min: 200.000000acceleration_mm_per_sec2: 1000.000000max_travel_mm: 20.000000soft_limits: falsehoming:cycle: 0allow_single_axis: truepositive_direction: truempos_mm: 9.500000feed_mm_per_min: 100.000000seek_mm_per_min: 100.000000settle_ms: 250seek_scaler: 1.500000feed_scaler: 5.000000motor0:limit_neg_pin: gpio.35:lowlimit_pos_pin: NO_PINlimit_all_pin: NO_PINhard_limits: truepulloff_mm: 9.500000stepstick:step_pin: I2SO.10direction_pin: I2SO.9:lowdisable_pin: I2SO.8ms1_pin: NO_PINms2_pin: NO_PINms3_pin: I2SO.11reset_pin: NO_PINmotor1:limit_neg_pin: NO_PINlimit_pos_pin: NO_PINlimit_all_pin: NO_PINhard_limits: falsepulloff_mm: 1.000000null_motor:control:safety_door_pin: gpio.27:low:pureset_pin: gpio.12:low:pufeed_hold_pin: gpio.14:low:pucycle_start_pin: gpio.13:low:pumacro0_pin: NO_PINmacro1_pin: NO_PINmacro2_pin: NO_PINmacro3_pin: NO_PINcoolant:flood_pin: NO_PINmist_pin: gpio.4delay_ms: 50probe:pin: gpio.36:lowtoolsetter_pin: NO_PINcheck_mode_start: falsemacros:startup_line0:startup_line1:macro0:macro1:macro2:macro3:start:must_home: truedeactivate_parking: falsecheck_limits: trueparking:enable: falseaxis: Ztarget_mpos_mm: -5.000000rate_mm_per_min: 800.000000pullout_distance_mm: 5.000000pullout_rate_mm_per_min: 250.000000user_outputs:analog0_pin: NO_PINanalog1_pin: NO_PINanalog2_pin: NO_PINanalog3_pin: NO_PINanalog0_hz: 5000analog1_hz: 5000analog2_hz: 5000analog3_hz: 5000digital0_pin: NO_PINdigital1_pin: NO_PINdigital2_pin: NO_PINdigital3_pin: NO_PINdigital4_pin: NO_PINdigital5_pin: NO_PINdigital6_pin: NO_PINdigital7_pin: NO_PINLaser:pwm_hz: 15000output_pin: gpio.15enable_pin: gpio.16:lowdisable_with_s0: falses0_with_disable: falsetool_num: 0speed_map: 0=0.000000% 1000=100.000000%off_on_alarm: falsearc_tolerance_mm: 0.002000junction_deviation_mm: 0.010000verbose_errors: falsereport_inches: falseenable_parking_override_control: falseuse_line_numbers: falseplanner_blocks: 16ok Sent from Mail for Windows From: bdringSent: 19 September 2023 14:43To: bdring/FluidNCCc: awp0645; AuthorSubject: Re: [bdring/FluidNC] Problem: Lightburn software. FluidNC K40 board ver. 7.7 (Issue #1028) The wiki shows you how to get the config file.http://wiki.fluidnc.com/en/support/requesting_help#showing-the-config-file

bdring, 2023-09-19:
Try using Fluidterm until your machine works. I can't read that.

Try jogging to see if the motors work.

http://wiki.fluidnc.com/en/features/jogging#jogging-fluidnc

MitchBradley, 2023-09-19:
The FluidNC version string says 5.1.0.  The latest official release is 3.6.8.  Where did you get a version of FluidNC that claims to be version 5.1.0 ?  Also what board is "FluidNC K40 Ver 7.7"?  There is no such board in our list of supported hardware on the FluidNC wiki.  Please supply a link to where you purchased it.

BlueOrangeLive, 2023-09-19:
https://lasercutting.avataar120.com/en/2022/07/03/brand-new-k40-fluidnc/

https://lasercutting.avataar120.com/en/2022/09/02/update-your-k40-fluidnc-shield/

BlueOrangeLive, 2023-09-19:
https://github.com/Avataar120/FluidNC

MitchBradley, 2023-09-19:
You should ask avaatar120 for help.  He is active on the FluidNC Discord server.

MitchBradley, 2023-09-19:
You should also file an issue on the Avataar120/FluidNC github repo.  The FluidNC core team cannot support modified versions of FluidNC over which we have no control.

Avataar120, 2023-09-19:
Hi,

Please contact me on Discord : Avataar120 or through my email : dric.vivien@gmail.com

As you have a v7.7 board you should never have upgraded it to firmware > 4.3.4 ...
Why is it the 5.1 on your board ? 

Did you upload a custom config.yaml ?
It seems the config file is corrupted or at least FluidNC is not able to read it -> default config file is used.

Can you provide me your order number ? 
I'll send you back the good config.yaml according to your board options.

Note : stop to send $ commands without knowing what you are doing :)
This firmware is not a stock GRBL one, this is FluidNC. For instance $22 is not supported, nor machine config in Lightburn.

Best,

Ava

Avataar120, 2023-09-19:
It seems you bought a XY board.

Please downgrade your firmware to that [one](https://github.com/Avataar120/FluidNC/releases/download/Ava_4.3.4/firmware.bin)
(instructions on my [website](https://lasercutting.avataar120.com/en/2022/09/02/update-your-k40-fluidnc-shield/))

Here the link to good [config.yaml](https://github.com/Avataar120/Ava_Shield_Configs/releases/download/Ava_2.0/4_AvaShield_7.x_K40_XY.yaml)

Please rename it config.yaml and then upload it to the board either using Fluidterm or using the WebUI (instructions on my [website](https://lasercutting.avataar120.com/en/2022/09/02/update-your-k40-fluidnc-shield/))

Ava

awp0645, 2023-09-19:
Hello Mitch

The board is the black Fluid Board shown on the Fluid website.  Ver 7.7 is printed on the board.

The software was pre-loaded and flashed again recently by Cedric  (Avataar120) 

https://lasercutting.avataar120.com/en/￼
Laser Cutting - Lightburn Plug & play shield for K40 laser cutters
lasercutting.avataar120.com

MitchBradley, 2023-09-20:
> The board is the black Fluid Board shown on the Fluid website.

What do you mean by "the Fluid website"?  There is more than one website that relates to FluidNC, and probably some that I don't know about. Do you mean the FluidNC wiki - and if so which page?  Supply a specific URL, not a non-specific brief description.  The FluidNC wiki external hardware page at <http://wiki.fluidnc.com/en/hardware/existing_hardware> shows 12 boards that are black, and none of them is the one you are talking about.

> [https://lasercutting.avataar120.com/en/￼]

That link is malformed and goes to an error page.

Please be a lot more careful in your writing if you expect to get good help.  We don't have to guess what you might mean.

awp0645, 2023-09-20:
Good morning

 The web page is at https://lasercutting.avataar120.com/en/     where details of the board can be seen seen

Thanks

Avataar120, 2023-09-20:
Hi again,

Mitch and the core team cannot help you with this product.
Please contact me directly and/or answer to my above requests.

Thanks,

Ava.

Avataar120, 2023-10-11:
Hi,

Problem has been solved.
You can close that issue :)

Thanks

closed 2023-10-11 by bdring