bdring/FluidNC #1036: FluidNC only reports '$10' on a '$$' request

buster3, 2023-10-06:
### Machine Context

Custom build laser engraver.

### Feature Description

FluidNC should work with LaserGRBL frontend. The report '$$' is not compatible with GRBL. Currently only '$10' is given. '$30' is required.

### Other Approaches

implement a workaround in LaserGRBL

### How I Can Help

I can implement the feature, but need a hint were the '$$' report is generated.

bdring, 2023-10-07:
This is a little complicated in the code due to the way Grbl some setting get mixed into the config file like $30 and some like $10 do not.

I'll look into doing this.

buster3, 2023-10-07:
Thanks, this is the lasergrbl issue: https://github.com/arkypita/LaserGRBL/issues/1885

Avataar120, 2023-10-08:
Hi @buster3, for your information, on my fork I have implemented this answer to $$ in order Lightburn can read settings from the board.
You can easily adapt it to your needs (I know this is a dirty way to implement it and it cannot be reused for all FluidNC applications)

File : ProcessSettings.cpp
New functions : findAxisIndexFromLetter, has_laser, laser_max_speed
Modified function : report_normal_settings

``
static int findAxisIndexFromLetter(char name) {
    int n_axis = config->_axes->_numberAxis;

    for (int axis = 0; axis < n_axis; axis++) {
        if (config->_axes->axisName(axis) == name)
            return axis;
    }

    return -1;
}

static bool has_laser() {
    if (config->_spindles.size())
        for (auto s : config->_spindles)
            if (!strcmp(s->name(), "Laser"))
                return true;

    return false;
}

static float laser_max_speed() {
    if (config->_spindles.size())
        for (auto s : config->_spindles)
            if (!strcmp(s->name(), "Laser"))
                return s->maxSpeed();

    return 0;
}

static Error report_normal_settings(const char* value, WebUI::AuthenticationLevel auth_level, Channel& out) {
    log_to(out, "$0=", config->_stepping->_pulseUsecs);
    log_to(out, "$1=", config->_stepping->_idleMsecs);
    log_to(out, "$2=0");
    log_to(out, "$3=0");
    log_to(out, "$4=0");
    log_to(out, "$5=0");
    log_to(out, "$6=0");
    show_settings(out, GRBL);  // GRBL non-axis settings

    log_to(out, "$11=", config->_junctionDeviation);
    log_to(out, "$12=", config->_arcTolerance);
    log_to(out, "$13=", (config->_reportInches ? 1 : 0));

    log_to(out, "$20=0");
    log_to(out, "$21=0");
    log_to(out, "$22=1");
    log_to(out, "$23=0");
    log_to(out, "$24=200");
    log_to(out, "$25=2000");
    log_to(out, "$26=250");
    log_to(out, "$27=5");
    log_to(out, "$30=", laser_max_speed());
    log_to(out, "$31=0");
    log_to(out, "$32=", (has_laser() ? 1 : 0));

    char axes[7] = "XYZABC";

    for (int i = 0; i < 6; i++) {
        char a = axes[i];

        if (findAxisIndexFromLetter(a) != -1) {
            LogStream ss("");
            ss << "$10" << i << "=" << config->_axes->_axis[findAxisIndexFromLetter(a)]->_stepsPerMm << "\n";
            ss << "$11" << i << "=" << config->_axes->_axis[findAxisIndexFromLetter(a)]->_maxRate << "\n";
            ss << "$12" << i << "=" << config->_axes->_axis[findAxisIndexFromLetter(a)]->_acceleration << "\n";
            ss << "$13" << i << "=" << config->_axes->_axis[findAxisIndexFromLetter(a)]->_maxTravel << "\n";
        }
    }

    return Error::Ok;
}
``

closed 2024-12-12 by bdring