bdring/FluidNC #1037: New Kinematics.

AndreyFly007, 2023-10-08:
### Machine Context

We need a CNC control controller for a laser burner. The laser module is fixedly fixed on top of the workpiece. Opposite the laser there are two mirrors that can be rotated by a stepper motor. Thereby directing the beam to the workpiece.
This kinematics is many times simpler than Delta-Robot kinematics.
The formula for calculating motor steps is very simple.
In the settings you need to specify the height of the mirror from the workpiece (h) and the degree of rotation of the mirror by 1 step (StepDegree). The following is a very simple calculation
Xstep = (ArcTg(Xcoord/h)/StepDegreeX)
Ystep = (ArcTg(Ycoord/h)/StepDegreeY)

### Feature Description

Such kinematics are in great demand, but there is no ready-made solution specifically for CNC. This kinematics will allow you to significantly reduce the cost of mechanics of a CNC machine. No linear guides, screw gears, etc. There are many examples on the network for a laser show, and all of them are not suitable for CNC, since they do not have step adjustments, due to
geometry suffers for this.

Here are several different attempts taken from the network to create a CNC using this kinematics.

https://www.instructables.com/DIY-STEPDIR-LASER-GALVO-CONTROLLER/

https://youtu.be/uNI8IQ32GzQ?si=pykm-O5yps1gFlm4

https://github.com/stanleyondrus/LaserProjectorV2

### Other Approaches

There is currently no ready-made solution

### How I Can Help

Unfortunately, I can't help you here. CNC machine is a hobby for me. But there is confidence that this kinematics is in great demand and there will be people who will appreciate it financially.
My income doesn't even allow me to make a donation for coffee.
Ready to become a beta tester for debugging.

bdring, 2023-10-08:
The only way to prove the great demand is via donations. If you cannot donate, just find someone else from that great demand to donate.

I don't think this is something we will donate our time to. We spent a lot of time to make the kinematics system of FluidNC easy for third parties to customize.

For CNC use, you generally need an expensive lens to adjust the focal distance of the lasers as it moves across the workpiece.

AndreyFly007, 2023-10-08:
The focus of my laser is possible from 100 to 500 mm. Which easily fits into a working field of 200 x 200 mm. The laser above the workpiece can have a distance from 110 mm to 450 mm.

Unfortunately, wages in my country are 6 - 15 dollars per working day. I am unable to collect donations.

You have done a great job creating a really great project. Thank you!

MitchBradley, 2023-10-08:
At the current donation rate, my work on FluidNC is compensated at the rate of less than $1 per day.  If there really is a "great demand", and expertise of the people who represent that demand is consistent with the questions on the instructables article, then that demand would result in even more unskilled users asking for free support.

closed 2023-10-09 by AndreyFly007