bdring/FluidNC #1039: Problem: Require Config File (YAML) for 4axis_external_driver

ASHAPURAAHM, 2023-10-09:
### Wiki Search Terms

4 axis external driver 
config file

### Controller Board

[4axis_external_driver](https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/Machines/4axis_external_driver.h)

### Machine Description

4 axis machine. Router type setup and 4th axis tangential

### Configuration file

```yaml
we require config file
```

### Startup Messages

```text
Need Config file
```

### User Interface Software

Web UI

### What happened?

We require YAML file

bdring, 2023-10-09:
Where did you get the controller?

ASHAPURAAHM, 2023-10-09:
We have Made This

bdring, 2023-10-09:
What type of spindle are you using?

Will you donate to the project?

ASHAPURAAHM, 2023-10-09:
We don't use spindle

Will surely donate to project

bdring, 2023-10-09:
[Donate here.](http://wiki.fluidnc.com/#donations)

Try this config file. It should be a good start. You may need to invert switches, etc on your machine.

```yaml
name: "External 4 Axis Driver Board V2"
board: "External 4 Axis Driver Board V2"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.34
      stepstick:
        direction_pin: gpio.2
        step_pin: gpio.0
    motor1:
      null_motor:

  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.35
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
    motor1:
      null_motor:

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.36
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
    motor1:
      null_motor:

  a:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.39
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  mist_pin:  gpio.21

        
probe:
  pin: gpio.32

```

closed 2024-02-02 by bdring