bdring/FluidNC #1043: GRBL ESP32 conversion issues

pcdonovan, 2023-10-10:
### Wiki Search Terms

Configuration, 5v PWM, No Motion, Examples

### Controller Board

Barts Universal 6-pack with External Driver Chips Opto input module, 5v Module, Mosfet Module

### Machine Description

XYYZ Gantry Router current running GRBL ESP32 with the 5v module controlling a SUPERPID Speed controller on a Makita router. The Mosfet Module controls the Mist coolant solenoid, and the Opto Module controlling the endstops.
4 external 542t drivers.  

All working perfectly on GRBL ESP32

### Configuration file

```yaml
ame: TOPSCNC HEAVY
board: 6-pack
kinematics: {}
stepping:
  engine: I2S_STATIC
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0
axes:
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: true
      settle_ms: 250
      seek_mm_per_min: 2000
      feed_mm_per_min: 200
      seek_scaler: 1.1
      feed_scaler: 5
    motor0:
      limit_all_pin: gpio.33
      hard_limits: false
      stepstick:
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0:low
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: true
      settle_ms: 250
      seek_mm_per_min: 2000
      feed_mm_per_min: 200
      seek_scaler: 1.1
      feed_scaler: 5
[... 372 lines cut ...]
#define Y2_LIMIT_PIN            GPIO_NUM_34

#define DEFAULT_INVERT_LIMIT_PINS       0 // Default for NO Switches
#define DEFAULT_INVERT_PROBE_PIN        0

// 5v Output Module inSocket #2
// https://github.com/bdring/6-Pack_CNC_Controller/wiki/4x-5V-Buffered-Output-Module
#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_2  // 5v PWM Output
#define SPINDLE_ENABLE_PIN      GPIO_NUM_25  // Inverted 5v Enable
#define USER_DIGITAL_PIN_4       GPIO_NUM_39  // 5v Power for SuperPID
#define USER_DIGITAL_PIN_0      GPIO_NUM_36  // M7 on M9 Off Unused

// Quad MOSFET module in socket #3
// https://github.com/bdring/6-Pack_CNC_Controller/wiki/Quad-MOSFET-Module
#define COOLANT_MIST_PIN       GPIO_NUM_26  //24v Mist Solenoid
#define COOLANT_FLOOD_PIN      GPIO_NUM_4    
#define USER_DIGITAL_PIN_1     GPIO_NUM_16   
#define USER_DIGITAL_PIN_2     GPIO_NUM_27

// Open module in socket #4
#define PROBE_PIN               GPIO_NUM_14

// === Default settings
#define DEFAULT_STEP_PULSE_MICROSECONDS I2S_OUT_USEC_PER_PULSE

bdring, 2023-10-10:
I loaded it on a 6 pack and tested a couple of things.

Motion on X axis works.
Laser enable works and the output works in G1 mode.

pcdonovan, 2023-10-10:
ARGHHH ok going to have to see what else the issue could be thanks Bart this was all in prep to swap over to 2.2kw spindle. I'll keep you posted looking forward to getting the 10v module this week

pcdonovan, 2023-10-10:
Does it matter if I use the Laser PWM Vs the Spindle PWM both run through the 5v module?

bdring, 2023-10-10:
It should not matter. The Laser only turns on during feed rate controlled motion. The PWM spindle turns on any time M3 or M4 is on.

http://wiki.fluidnc.com/en/config/config_spindles

bdring, 2023-10-10:
I suggest using module slot #3 or #4 for spindle modules. They are the most versatile.

pcdonovan, 2023-10-10:
I'll give that a shot

bdring, 2023-10-11:
Are you on our Discord server? We can help there too.

https://discord.gg/pPNXnZgN

pcdonovan, 2023-10-11:
Yes, I am! The discord says to head to the Git lol

bdring, 2023-10-11:
Gothub has the advantage of templates. It also has a much larger post size limit. Many people solve problems just following the template. 

Was is your user name on discord.  I'll tag you as a sponsor.

pcdonovan, 2023-10-15:
Problem Solved

closed 2023-10-15 by bdring