bdring/FluidNC #1047: Problem: ESP32 5-axis experiment

yusufferdemm, 2023-10-16:
### Wiki Search Terms

Hello, I wish you a good day,

We designed a board for ESP32 in this way and loaded FluidNC onto the ESP32. I couldn't find a suitable configuration (YAML) file to convert it to 5 axes. I couldn't define A and B axes similar to the 3axis_v4.yaml file in example_configs.

a:
  steps_per_mm: 800
  max_rate_mm_per_min: 2000
  acceleration_mm_per_sec2: 25
  max_travel_mm: 1000
  homing:
    cycle: 1
    mpos_mm: 10
    positive_direction: true
  motor0:
    limit_all_pin: gpio.16:low:pu
    stepstick:
      direction_pin: gpio.33
      step_pin: gpio.27
  motor1:
    null_motor:

When I tried to define them in this way for A and B, the software did not detect the other two axes.

How can I define 5 axes on this board in the configuration file?

Or is there a pre-defined 5-axis configuration file for ESP32?

### Controller Board

ESP32

### Machine Description

CNC Milling Machine

### Configuration file

```yaml
don't have :(
```

### Startup Messages

```text
FluidNC 3.9
```

### User Interface Software

UGS

### What happened?

Somehow, when I define a and b axes in the 3axis_v4 yaml file, a and 5 axes do not appear as defined in UGS.

### Other Information

[image]
[image]
[image]

bdring, 2023-10-16:
There are some example files here.

https://github.com/bdring/6-Pack_CNC_Controller/tree/main/FluidNC_configs

bdring, 2023-10-19:
Show your existing config file that works for the 3 axes.
Show the startup messages too.

bdring, 2024-12-12:
Closing inactive issue

closed 2024-12-12 by bdring