bdring/FluidNC #1049: Problem: Ultimate Bee updated to FluidNC - issues with configuration without errors

NapoleonStallone, 2023-10-17:
### Wiki Search Terms

upgrade CNC XPro V5 with Fluid NC; config.yaml Ultimate Bee; installation FluidNC xPro v5; example config.yaml CNC xPro v5 with VFD Huanyang; router not reacting after FluidNC upgrade; VFD RS485 issue FluidNC

### Controller Board

Spark Concepts CNC XPro V5

### Machine Description

Ultimate Bee by Bulkman3D Kit with Huanyang VFD and 2.2 kW water-cooled spindle; High-torque NEMA 23 stepper motors, CNC xPro v5 controller; X/Y hard limit switches on neg. sid, Z hard limit on pos. side; size 1500x1500; XYYZ set-up, NC-door switch.

### Input Circuits

```yaml
Whole set-up according to guideline provided by Bulkman3D - before FluidNC limits, homing, running on .nc-programs were working.
```

### Configuration file

```yaml
board: XPro V5
name: Bulkman 3d 1515
meta: Config 16.10.2023
stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160.00
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1286.00
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      feed_mm_per_min: 50.00
      seek_mm_per_min: 200.00
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 5.000
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  y:
    steps_per_mm: 160.00
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 1280.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 500.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 85.000
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2
        # disable_pin: NO_PIN
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.050
        run_amps: 1.800
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
        
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 10000000
  
probe:
  pin: gpio.22
  check_mode_start: true

control:
  safety_door_pin: gpio.16:low
  macro0_pin: gpio.13:low
  macro1_pin: gpio.0:low

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

uart1:
  txd_pin: gpio.4
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  baud: 9600
  mode: 8N1
 
Huanyang:
  Uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false
```

### Startup Messages

```text
> $ss
[MSG:ERR: Showing startup log from previous panic]
[MSG:INFO: FluidNC v3.7.9 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Bulkman 3d 1515]
[MSG:INFO: Board XPro V5]
[MSG:INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1286.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35]
[MSG:INFO: Axis Y (0.000,1280.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Neg Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO: Axis Z (-85.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:  Z Pos Limit gpio.39]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.16:low]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:Camper_Imperium-2.4G]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.178.104]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
ok

---------
I removed the SD for checking. Before that this message was delivered as well:
E (8923) sdmmc_io: sdmmc_io_reset: unexpected return: 0x102
[MSG:ERR: sdmmc_card_init failed code 0x102]
```

### User Interface Software

WebUI & CNCjs

### What happened?

Hello!
I tried to solve one issue and created another one.
Initial issue: Manual control of spindle was possible after set-up of the Ultimate Bee - meaning, before running a .nc, I had to manually enable the spindle and turn up the rpm. I tried everything I knew and the internet was suggestint to make the Huanyang VFD run via RS485 connection to the xPro V5, but failed. So I thought FluidNC could be the solution.

Consequent issue: flashed the controller via terminal (Mac-OS) with installfs.sh downloaded from Github according to instruction .pdf. It took some time and checking the wiki as well as Github to get the "working" config.yaml - working = as far as not getting any errors listed anymore (expect for the SD card). Config.yaml uploded via WebUI.

Now, I can send jog commands and can "track" the movement on the CNCjs dashboard or WebUI-command-window - but the router is not moving, not even in the wrong direction, silence. No axis is reacting to commands. Spindle issue of course also not solved as well.

I cannot exclude that I or my lack of expertise is the issue. Anyhow, I would appreciate any suggestion how to get the CNC running again. Thank you for your time in advance!

### GCode File

Example for jog commands that did not lead to jogging:

feeder> G91
feeder> G0 Y10
ok
feeder> G90
ok
ok
feeder> G91
feeder> G0 X10
feeder> G90
ok
ok
ok

NapoleonStallone, 2023-10-17:
I measured the output current from controller to stepper motors, which was zero with the config mentioned above. I removed following parts of the config:

control:
  safety_door_pin: gpio.16:low
  macro0_pin: gpio.13:low
  macro1_pin: gpio.0:low

And now I can jog with the Y2/A axis and X axis. Z and Y-axis are still silent, limit switches are not working as well when triggering manually. 

I am not sure how to interpret these "observations" and if that helps with the root cause...

bdring, 2023-10-17:
I have an xPro. I do not have it hooked up right now.  I don't have time to test your file today.

We have a wiki page for it that might help.

http://wiki.fluidnc.com/en/hardware/3rd-party/xPro_V5

MitchBradley, 2023-10-19:
If you send $message/level=debug, you might get some helpful messages about pin states.

NapoleonStallone, 2023-10-24:
> If you send $message/level=debug, you might get some helpful messages about pin states.

Thanks in advance for the help. I send $message/level=debug and got this response (Using CNCjs, same for WebUI).
`> $message/level=debug
[MSG:INFO: Valid options: Debug Error Info None Verbose Warning]
error:2 (Bad number format)`

> I have an xPro. I do not have it hooked up right now. I don't have time to test your file today.
> 
> We have a wiki page for it that might help.
> 
> http://wiki.fluidnc.com/en/hardware/3rd-party/xPro_V5

Thanks for the link. I made some changes, which I think are not the same as described in the link above, but now homing and axis movement are working. Anyhow, I did not manage to get the Huanyang VFD running via RS485 yet. Current config.yaml: 
[config24102023.txt](https://github.com/bdring/FluidNC/files/13113497/config24102023.txt)

**Start-up message:**
```
CNCjs 1.10.1 [Grbl]
Verbunden mit dem seriellen Port /dev/tty.usbserial-0001 mit einer Baud-Rate von 115200
�%-
[MSG:INFO: FluidNC v3.7.9 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine Bulkman 3d 1515]
[MSG:INFO: Board XPro V5]
[MSG:INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1245.000)]
[MSG:INFO:   Motor0]
[MSG:DBG: TMCStepper Library Ver. 0x703]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35]
[MSG:DBG:  X Neg Limit 0]
[MSG:INFO: Axis Y (0.000,1250.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26:low:pu CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Neg Limit gpio.34]
[MSG:DBG:  Y Neg Limit 0]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32:low:pu CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO: Axis Z (-85.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:  Z Pos Limit gpio.39]
[MSG:DBG:  Z Pos Limit 0]
[MSG:INFO: X Axis driver test passed]
[MSG:DBG: X Axis Coolstep]
[MSG:INFO: Y Axis driver test passed]
[MSG:DBG: Y Axis Coolstep]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:DBG: Y2 Axis Coolstep]
[MSG:INFO: Z Axis driver test passed]
[MSG:DBG: Z Axis Coolstep]
[MSG:INFO: safety_door_pin gpio.16:low]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: Connecting to STA SSID:Camper_Imperium-2.4G]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[... 41 lines cut ...]
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  off_on_alarm: false
```

Is that plausible or are there any suggestions for changes?

What I found on wiring: CNC xPro // VFD Huanyang
- Relay N/O // DCM (on the upper row, but I think it makes no difference since both DCM or ACM are connected)
- Relay COM // FOR
- Toolhead GND // ACM (on the upper row)
- Toolhead 0-10 V // 10 V (with multimeter I can measure around 10 V between ACM/10V)
- internal jumper in the connector set to: Mist+rly
- jumper VI/VR on the VFD set to: VI
- Programming on the VFD of Pd001=1, Pd002=1, Pd003-5=400, Pd007=0.5, Pd008=220, Pd009=15, Pd010= 10, Pd014=10, Pd015=10, Pd0023=0,Pd025=1, Pd027=0.8, Pd041=220, Pd044=1, Pd072=400, Pd073=133.33, Pd141=220, Pd142= 10, Pd143=2, Pd144= 3000, Pd163-165=factory settings.

Anyhow, I cannot make it run that way.

And sorry for the long post!

NapoleonStallone, 2023-10-24:
Well, 0-10 V, configured and wired as suggested above, is...not working ;)

Gpio Pins for the 0-10V analogue signal control are not explicitly mentioned here [https://github.com/Spark-Concepts/xPro-V5-Firmware](url) or is there another source for this information?

Thanks in advance for any tip.

`[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:ERR: ERR: gpio.13 - Pin is already used.]
[MSG:ERR: Configuration error at /machine/10V: ]
[MSG:ERR: ERR: gpio.17 - Pin is already used.]
[MSG:ERR: Configuration error at /machine/10V: ]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:ERR: Configuration is invalid]
[MSG:INFO: Using default configuration]`

MitchBradley, 2023-10-24:
Try testing with FluidTerm instead of CNCjs.  CNCjs supposedly works with FluidNC but the way it is formatting the messages is making it hard for us to determine exactly what is happening.

MitchBradley, 2023-10-24:
Notice this error message: `[MSG:ERR: ERR: gpio.13 - Pin is already used.] [MSG:ERR: Configuration error at /machine/10V: ] [MSG:ERR: ERR: gpio.17 - Pin is already used.]`

Your startup messages show:
```
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
...
[MSG:INFO: macro0_pin gpio.13:low]
```
You cannot assign gpio.17 and gpio.13 for spindle control if they are already used for other purposes (gpio.13 for macro0 and gpio.17 for a TMC CS).

The [XPro page on the FluidNC wiki](http://wiki.fluidnc.com/en/hardware/3rd-party/xPro_V5) has all of the information that we know about the XPro.  That page lists the GPIO usage and also says this:  `XProV5 is a commercial product so your first line of support should be the vendor`

When we write something like that, what we mean is that you paid for their product so they should help you.  FluidNC is free unless people donate - most people do not.  Ask for help from the people who have your money, not from the people who are giving away their time and effort.

NapoleonStallone, 2023-10-25:
Dear Mr. Bradley,

thank you for the feedback. I am sure that people are more willing to donate if they are successful using provided information. I will contact Spark Concepts about this issue and post the results, if positive. Anyhow, our experience with Spark Concepts in providing proper support or any support on their products was not positive so far.

Have a great day.

AizenJF, 2023-11-29:
im having this same issue did you ever end up getting it resolved ?

MitchBradley, 2023-11-29:
> im having this same issue did you ever end up getting it resolved ?

Since this issue has so many different aspects - config file problems and motors and spindles - it is hard to interpret "the same issue".  Please file a separate ticket with all the information for *your* system - config file and startup messages and everything that the issue template asks for.

NapoleonStallone, 2023-11-29:
> im having this same issue did you ever end up getting it resolved ?

Yes, I got the machine running... I did not have the time to properly compile all hardware & config settings yet...so what exactly is your issue? Maybe I can share what I did in my case.

Bigrigins, 2023-12-12:
I'm having the same issue,  got all axis working all moters turn except y0. Ya2 moves but not Y0, not sure what to do . Looking for advice . Here are my settings. 

board: tmc_5160_xpro v5
name: ULTIMATE BEE
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 0
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.00
    max_rate_mm_per_min: 300.00
    acceleration_mm_per_sec2: 100.00
    max_travel_mm: 1240.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.00
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.075
        run_amps: 2.5
        hold_amps: 0.50
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.12
        direction_pin: gpio.14
        tpfd: 4
       

  y:
    steps_per_mm: 400.00
    max_rate_mm_per_min: 300.00
    acceleration_mm_per_sec2: 100.00
    max_travel_mm: 1240.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 1500.00
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor1:
[... 200 lines cut ...]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.075]
[MSG:INFO:  X Neg Limit gpio.35]
[MSG:INFO: Axis Y (-1090.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:4 R:0.075]
[MSG:INFO: Axis Z (-10.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:3 R:0.075]
[MSG:INFO:  Z Neg Limit gpio.39]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y2 Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: safety_door_pin gpio.16:low]
[MSG:INFO: macro0_pin gpio.13:low]
[MSG:INFO: macro1_pin gpio.0:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: BT Started with FluidNC]

Grbl 3.7 [FluidNC 3.7.10 (bt) '$' for help]
[MSG:INFO: ALARM: Unhomed]
ALARM:14

MitchBradley, 2023-12-13:
a) Notice the ERR and WARN messages.  That indicates a problem with the config file, usually an indentation issue.  We cannot tell what your indentation actually is because of the way the data is pasted into the ticket.
b) Please file a separate ticket even though you think it is the same problem.  It is very difficult to keep track of things when people pile separate information onto one ticket.

Bigrigins, 2023-12-13:
Ok thank you, I will open a new ticket tomorrow.  Thanks again.

Bigrigins, 2023-12-31:
you can close this issue out, all axis are moving as they should and homing. problem now with spindle not responding through cncjs software, plan to open a new ticket. thanks for help this far.

jessemoc, 2024-01-16:
>VFD Huanyang issue:
Info on the RS485-connection between Huanyang VFD with 2.2 kW watercooled 220 V spindle to CNC xPro v5:

>twisted cable, length around 15 cm
RS- (VFD) to A (xPro), RS+ to B //// the green LED on the controller is on; when changing to RS+ to A, as shown in the Spark Concept wiki, the LED goes off and NoSpindle type feedback after restart.
EN/PWM-RS485 A/B switch is switched to RS485
programming of the VFD is done as described in the Spark Concept wiki
I worked through the FluidNC-wiki as well as the issue on that topic which was discussed here [[V5/wiki/RS485-VFD](https://github.com/Spark-Concepts/xPro-V5/wiki/RS485-VFD)](url), but no success so far.
just for info: spindle works in manual programming mode and can be tuned up to 24k RPM via poti on the VFD.
I would be glad if you have any suggestions or tips for trouble shooting or solutions.
Anyhow, plan B would be to go for analogue mode via toolhead-pin 0-10 V on the controller, but unfortunately I was not able to find consistent information on how to properly wire the VFD to the CNC xPro as well as am not sure about the config. 

NapoleonStallone I have an almost identical setup to you (with the exception that I have a 110V spindle).  I have the same configuration as you for the RS-485 and mine works fine (I also used the instructions on the sparks Wiki for wiring and wired it as shown).  Did you make sure that you fully powered up the VFD before turning on the controller and plugging it into a computer?