bdring/FluidNC #1051: Problem: Configuration problem will cause an Alarm state only after the `$verbose_errors=true` command is issued

breiler, 2023-10-25:
### Wiki Search Terms

N/A

### Controller Board

ESP32 CNC Dev Controller v4.1

### Machine Description

No peripherals connected

### Input Circuits

```yaml
No peripherals connected
```

### Configuration file

```yaml
board: XPro V5
name: xPro V5 Ondrej
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 159.6
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 100
    max_travel_mm: 722
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 2000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.5
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14:low
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.05
        run_amps: 1.8
        hold_amps: 0.25
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
  y:
    steps_per_mm: 160.1
    max_rate_mm_per_min: 2500
    acceleration_mm_per_sec2: 100
    max_travel_mm: 1252
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 2000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.58
      tmc_5160:
        step_pin: gpio.27
        direction_pin: gpio.26
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 2
        r_sense_ohms: 0.05
        run_amps: 1.8
        hold_amps: 0.25
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
    motor1:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.3
      tmc_5160:
        step_pin: gpio.33
        direction_pin: gpio.32
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.05
        run_amps: 1.8
        hold_amps: 0.25
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 1500
    acceleration_mm_per_sec2: 50
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150
      feed_mm_per_min: 100
      seek_mm_per_min: 500
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2.5
      tmc_5160:
        step_pin: gpio.15
        direction_pin: gpio.2:low
        use_enable: true
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.05
        run_amps: 1.8
        hold_amps: 0.25
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
        toff_coolstep: 3
        tpfd: 4
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  frequency_hz: 10000000
probe:
  pin: gpio.22:low
  check_mode_start: true
control:
  safety_door_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
macros:
  startup_line0: null
  startup_line1: null
  macro0: null
  macro1: null
  macro2: null
  macro3: null
start:
  must_home: false
coolant:
  mist_pin: gpio.21
  delay_ms: 0
uart1:
  txd_pin: gpio.4
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  baud: 9600
  mode: 8N1
huanyang:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false
```

### Startup Messages

```text
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC v3.7.9 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config-monberg.yaml]
[MSG:INFO: Machine xPro V5 Ondrej]
[MSG:INFO: Board XPro V5]
[MSG:INFO: UART1 Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:10000000]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-572.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14:low CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:INFO: Axis Y (-1102.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:  Y Neg Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO:  Y2 Neg Limit gpio.36:low]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2:low CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:  Z Neg Limit gpio.39:low]
[MSG:ERR: X Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:ERR: Y Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:ERR: Y2 Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:ERR: Z Axis TMC driver not detected - expected 0x30 got 0x0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle  Tx:gpio.4 Rx:gpio.25 RTS:NO_PIN Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low]
[MSG:INFO: Connecting to STA SSID:Murkelslottet]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: VFD RS485 Unresponsive]
[MSG:INFO: Connected - IP is 192.168.1.242]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.9 (wifi) '$' for help]
[MSG:INFO: Running macro startup_line0: null]
[MSG:INFO: Running macro startup_line1: null]

Grbl 3.7 [FluidNC v3.7.9 (wifi) '$' for help]
```

### User Interface Software

UGS and FluidNC Web Installer terminal

### What happened?

I got a support problem issued in UGS, where the user could connect with version 2.0.20 without any problems. In version 2.0.21 they got an alarm. 

In the later version we are sending `$verbose_errors=true`.

The root cause is a configuration error in the VFD section which should probably fixed by the user. But the controller should probably enter the alarm state even if the command `$verbose_errors=true` wasn't sent.

To reproduce, load the configuration and cold start the controller, then:
```
Grbl 3.7 [FluidNC v3.7.9 (wifi) '$' for help]
?
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:P|WCO:0.000,0.000,0.000>
$verbose_errors=true
[MSG:ERR: Validation error at /Huanyang: VFD: conflicting UART configuration]
ok
?
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:P|Ov:100,100,100>
```

This is low priority and a cosmetic problem.

### GCode File

N/A

### Other Information

N/A

MitchBradley, 2023-10-25:
Wow, that is weird.  But I was able to reproduce it...  Investigating.

MitchBradley, 2023-10-25:
The act of setting verbose_errors triggers a tree validation run that is somehow skipped in the initial config loading pass.

MitchBradley, 2023-10-25:
Okay I think I see the problem.  It is an artifact of the change that allowed you to specify the VFD uart either as a separate uart1 section or as a uart section below the VFD spindle node.  The initial validation pass succeeds - because the UART really is valid - but then the init() method changes the value of the _uart variable, causing a later validation pass to fail.  The fix is to change the logic in VFDSpindle::validate()

breiler, 2023-10-27:
Awesome, great work!

commit 73763acd (2023-11-05): Merge pull request #1052 from bdring/Fix1051

Fix #1051 - bad VFD UART validation and generation logic
commit f98ad130 (2023-11-05): Fix #1051 - bad VFD UART validation and generation logic
closed 2023-11-08 by bdring