bdring/FluidNC #1053: Problem: I2S does not work?

AlexMoiseevs, 2023-10-28:
### Wiki Search Terms

read everything

### Controller Board

6-pack

### Machine Description

Test

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack OLED
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  dir_delay_us: 10
  pulse_us: 5
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: i2so.2
        direction_pin: i2so.1:low
        disable_pin: i2so.0:low

  y:
    motor0:
      null_motor:

  z:
    motor0:
      null_motor:

i2c0:
   sda_pin: gpio.14
   scl_pin: gpio.13
        
oled:
   i2c_num: 0
   i2c_address: 60
   width: 128
   height: 64
   radio_delay_ms: 1000

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.18
```

### Startup Messages

```text
Ok
```

### What happened?

when moving , the date signal changes . actually it is logical. but because of the awkward synchronization signals, nothing is written to the register.

the chip 74hc595n is powered by 3.3 volts. manually closing the corresponding contacts, the register is recorded.
[image]

AlexMoiseevs, 2023-10-28:
the data line seems to be true. synchronization is about nothing at all.
[image]
[image]

AlexMoiseevs, 2023-10-28:
[image]

AlexMoiseevs, 2023-10-28:
I'm sorry. my analyzer was not configured that way.
[image]
BUT the register doesn't work anyway....

MitchBradley, 2023-10-28:
What exactly do you mean by "the register doesn't work"?  Do the outputs not change?  Basic things to check: Are the other signals like reset, oe, power, gnd, etc. at the right values?  Is the daisy-chain intact so that the data signal is propagated to subsequent chips in the chain?  Are you sure that you are looking at the right outputs?

It is possible that the 74AHCT595 chips are bad, but it would be somewhat surprising if they were all bad.  The only thing I can think of that would destroy them all at once would be applying a too-high voltage to their VCC pins.  If the first chip in the chain is bad and thus is not propagating the data signal through the daisy chain, that could block the function of subsequent chips.

AlexMoiseevs, 2023-10-28:
Thank you very much for the answer.

I think that the chips are really defective. the value does not change at the outputs. if the inputs are closed manually, then the data can be stuffed into the registers and the outputs change. there is a suspicion that the chips do not work at the required frequency. and they don't look very good, the logo is worn out....
[image]

MitchBradley, 2023-10-28:
How can you "close the inputs manually" while the device is in-circuit?

AlexMoiseevs, 2023-10-28:
disconnect from the signal source and poke the corresponding legs in plus and minutes according to the algorithm of the register. we look at the status of the outputs.

MitchBradley, 2023-10-28:
Please provide a photo of your setup

AlexMoiseevs, 2023-10-29:
[image]

bdring, 2023-10-29:
The developers have a policy on DIY hardware.

<http://wiki.fluidnc.com/en/support/controller_design_guidelines#developer-policy-on-user-created-hardware>

We are happy to leave this issue open if other users want to offer help.

AlexMoiseevs, 2023-10-29:
There are no questions about the firmware in this context. I'm sorry for my stupidity.
The question may be: How to test the repeatability of the end caps? I read the entire wiki and didn't find an answer.
The task is as follows: To drive the machine and check whether it has missed the steps by checking the home position.

AlexMoiseevs, 2023-11-03:
Everything works fine. The problem was in the chip. It is worth paying attention in the wiki to the choice of a chip operating at a high frequency of 16 MHz. Thanks again.
[image]

closed 2023-11-03 by MitchBradley