bdring/FluidNC #1055: Machine no longer responds to $HZ after upgrading to v3.7.9

AussieMakerGeek, 2023-11-02:
### Wiki Search Terms

Limit Switches
Homing

### Controller Board

MKS Tinybee

### Machine Description

Gantry Co2 Laser with Z axis
Gravograph LS800 converted to FluidNC

### Configuration file

```yaml
board: MKS-Tinybee
name: LS800
meta: 
stepping: 
  engine: I2S_static
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 2
  segments: 12

i2so: 
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi: 
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard: 
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  baud: 115200
  mode: 8N1

uart_channel1:
  uart_num: 1

kinematics: 
  Cartesian: 

axes: 
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x: 
    steps_per_mm: 59.500000
    max_rate_mm_per_min: 80000.000000
    acceleration_mm_per_sec2: 12000
    max_travel_mm: 610.000000
    soft_limits: true
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 1.000000
      feed_mm_per_min: 1500.000000
[... 315 lines cut ...]
[MSG:DBG: Homing limited X]
[MSG:DBG: Homing nextPhase Pulloff0]
[MSG:DBG: Starting from -43.765,304.000,0.000]
[MSG:DBG: Planned move to -39.765,304.000,0.000 @ 1500.000]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG: CycleStop Pulloff0]
[MSG:DBG: Homing nextPhase SlowApproach]
[MSG:DBG: Starting from -39.765,304.000,0.000]
[MSG:DBG: Planned move to -44.165,304.000,0.000 @ 1500.000]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Homing limited X]
[MSG:DBG: Homing nextPhase Pulloff1]
[MSG:DBG: Starting from -43.681,304.000,0.000]
[MSG:DBG: Planned move to -39.681,304.000,0.000 @ 1500.000]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG: CycleStop Pulloff1]
[MSG:DBG: Homing nextPhase Pulloff2]
[MSG:DBG: mpos was -39.681,304.000,0.000]
[MSG:DBG: mpos becomes 1.000,304.000,0.000]
[MSG:DBG: mpos transformed 1.008,304.000,0.000]
[MSG:DBG: Homing done]

Homing worked fine with the same config and wiring on the older release.

I really thought I was doing something wrong and thats still possible but all my validation is checking out.

Avataar120, 2023-11-02:
Z homing cycle = 0 means no homing for this axe.
You need to be strictly above 0

AussieMakerGeek, 2023-11-02:
> Z homing cycle = 0 means no homing for this axe. You need to be strictly above 0

I don't beleive you are correct unless this has changed - It means no homing when $H is issued but will still home with $HZ

From the Wiki:
A setting of 0 means it will not home with $H, but you can still home it with $H<axis>

And this previously worked.

bdring, 2023-11-02:
I'll try to look into this later today or tomorrow.

Ref: http://wiki.fluidnc.com/en/config/axes#homing

AussieMakerGeek, 2023-11-03:
Thanks @bdring , I just tried changing the Z homing sequence to 3 and it will now home all on $H and $HZ also works separately so I am pretty sure it's something in the code.  Once again, I tried poking around the code but got lost.

bdring, 2023-11-03:
I did some testing. I can repeat the problem.

Strangely...

If the z cycle is 0 in the config file it does not work right.

if I set `$axes/z/homing/cycle=0` temporarily via the console, it works correctly and I can home with $HZ.

AussieMakerGeek, 2023-11-03:
Thanks @bdring  - I was pretty sure I wasn't doing something wrong, and that at least gives me a workaround for the moment.

I did go up a few versions so i'm not certain in which release this would have stopped working.

PS - Is there something up with the wiki host?  It seems to be regularly unavailable recently.

bdring, 2023-11-03:
I think we found the bug. I will push it to the devt branch soon.

I have used the wiki all day, but I have seen other issue, like Github was 404ing for a while.

closed 2024-02-02 by bdring

21reed, 2026-10-20:
Hey @AussieMakerGeek ! Sorry to resurrect an old thread, but I'm working on converting my own Gravograph LS800 to FluidNC and found your configuration incredibly useful.

I have the original 40W Synrad V40 laser and 4-wire stepper motors, and I'm planning to use the same MKS TinyBee controller.

Would you happen to remember which stepper drivers you used (TMC2209, A4988, external drivers, etc.), along with their current settings, microstepping configuration, and supply voltage?

Also curious whether you encountered any challenges interfacing the Synrad laser with the TinyBee.

Any advice or photos from your conversion would be greatly appreciated. Thanks!

AussieMakerGeek, 2026-10-21:
Hi @21reed I'm genuinely excited to be able to help you out here!

I spent weeks researching and getting it all to work - Including with the original Synrad laser.  I later converted it to a glass tube because the Synrad laser was needing a recharge and that was not appropriate for the budget of the project. I should be able to give you all of the information you require. I even made up a PCB interface board that directly plugs into the factory loom connectors to make it plug and play.

I'm working from memory here, since this was a few years ago, but I did document pretty much everying, and I still  have that, just not at my fingertips. Don't take everything I write right now as fact...

There are 3 main connectors to deal with - an 24(?) pin 2x12 0.1" pitch connector, that deals with all the limit switches (maybe lid too). The main 'molex' connector is for all of the motors.  The 18(?) pin connector deals with the laser. Other than that, there's just the fan to deal with if you want to control that like the factory one (or just have it permanently on).

The laser is straight simple to drive and just works - I can't remember exactly how i did it, but i have the info. It's 5V logic level controlled, so you will need a TTL logic converter to drive it with the ESP32 though. I used a 74AHCT125 (on the interface board). There was a specific reason to use this one, and I can't remember why right now.

I later made up a control panel to replace the stock one too, based on a 3.5" capacative touch screen takling via serial to control the basic functions like homing and jogging, but since I built that, FluidNC now supports newer options for controls that might work better.

48V from 2 X IBM server power supplies in series (mods needed) **CAUTION**
4 X TMC2160_OC boards (MKS) 1/16 Microstep, 2.3A, Mhold On (1000001 on DIP Switches)

Layout

[image]

Interface board (spot the mistake I made?)

[image]

Feel free to hit me up directly.

21reed, 2026-10-23:
Hi @AussieMakerGeek , this is excellent info! How can I contact you directly? I don't see any contact information in your profile. I just ordered some parts!

AussieMakerGeek, 2026-10-26:
It's right there in my profile...