bdring/FluidNC #1056: Problem: New Config file for Elegoo Phecda 10W

fatihsevinc, 2023-11-02:
### Wiki Search Terms

Laser pwm and stepper motor step/dir configuration

### Controller Board

mks-like custom board, esp32 controller

### Machine Description

Elegoo Phecda 10W
2 axis Nema 17 Stepper Motor (42HD4027-02)
10W(Optic Output) laser
ESP32-32U  Controller 
X and Yaxis driver TMC2209

ESP Controller
[image]

Motor Driver
[image]

Laser PWM Driver
[image]

### Configuration file

```yaml
board: Elegoo Phecda 10W
name: Elego Phecda Mod by Fatih Sevinc
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.33
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 25000.000
    acceleration_mm_per_sec2: 1000.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:       
        step_pin: gpio.25
        direction_pin: gpio.26

  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 18000.000
    acceleration_mm_per_sec2: 300.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:       
        step_pin: gpio.27
        direction_pin: gpio.5

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 6000.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 50.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_2209:       
        step_pin: gpio.18
        direction_pin: gpio.19

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.35

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

probe:
  pin: gpio.36
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

laser:
  pwm_hz: 2000 
  output_pin: gpio.32:high
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true

  coolant:
  # Heated Bed Terminal Block
  flood_pin: i2so.16
  # HE0 Terminal Block
  mist_pin: i2so.17
  delay_ms: 0
```

### Startup Messages

```text
I want to install Fluidnc on my device. I prepared such a config file. Can you check? Is PWM correct? Are the step settings correct?
```

### User Interface Software

FluidNC

### What happened?

I want to install fluidnc on my device. I prepared such a config file. Can you check? Is PWM correct? Are the step settings correct?

bdring, 2023-11-02:
Please edit your post to show the startup messages.

See the wiki

Wiki.fluidnc.com

bdring, 2024-02-02:
Closing stale issue

closed 2024-02-02 by bdring

joshfedo, 2025-07-23:
For anyone looking for this in the future i was able to get this working.  Elegoo uses the z step pin (gpio 18) as a enable pin for the laser.  This is the config with homing in the bottom right corner (by the screen) with end stops:
```
name: Phecda_FluidNC
board: TMC2209_XY
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: gpio.33
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 250
    max_travel_mm: 400
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 1500
      seek_mm_per_min: 4000
      settle_ms: 250
    motor0:
      pulloff_mm: 3
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.36:low
      hard_limits: false
      stepstick:
        step_pin: gpio.25
        direction_pin: gpio.26
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 250
    max_travel_mm: 400
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 1500
      seek_mm_per_min: 4000
      settle_ms: 250
    motor0:
      pulloff_mm: 4
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.39:low
      hard_limits: false
      stepstick:
        step_pin: gpio.27
        direction_pin: gpio.5:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_all_pin: NO_PIN
spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
sdcard:
  cs_pin: gpio.15
  card_detect_pin: gpio.35:low
uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  baud: 115200
  mode: 8N1
  rts_pin: NO_PIN
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
probe:
  check_mode_start: true
  pin: NO_PIN
  toolsetter_pin: NO_PIN
start:
  must_home: true
Laser:
  output_pin: gpio.32
  s0_with_disable: false
  pwm_hz: 2500
  enable_pin: gpio.18 
  off_on_alarm: true
  speed_map: 0=0.000% 1000=100.000%
  disable_with_s0: true
```
This still needs a fix for the built in screen on the laser to be able to send/receive data and the PWM/power settings need to be tuned but this config gets everything moving, the SD card works, and the laser will fire.