bdring/FluidNC #1060: Problem: Unknown pin prefix when firmware > 3.7.0

srgkasuro, 2023-11-07:
### Wiki Search Terms

my config worked fine with firmware 3.7.0
I did an update to 3.7.9. because I fond a bug with G-Code parsing an wanted to see if it is already fixed.
but my old Config-file stopped working with multiple errors like:

[MSG:ERR: Unknown prefix:gpio:13:low]
[MSG:ERR: Setting up pin: gpio:13:low failed:Unknown pin prefix]
[MSG:ERR: Unknown prefix:gpio:25]
[MSG:ERR: Setting up pin: gpio:25 failed:Unknown pin prefix]

I tried different Firmwares down to 3.7.4 with no luck.

back to 3.7.0 it is working fine.

### Controller Board

6-pack

### Machine Description

Router with external Stepper drivers

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYZ Haunyang
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 4500.000
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 480.000
    soft_limits: false

    homing:
      cycle: 2
      positive_direction: true
      allow_single_axis: true
      mpos_mm: 114.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio:13:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 4500.000
    acceleration_mm_per_sec2: 200.000
[... 280 lines cut ...]
[MSG:DBG: Stabwechsel.nc hash "68A88D6CDBA4FCA63D329D3AFEC88C6EB61983532B087E7F50B944145BB6E59F"]
[MSG:DBG: Test1.nc hash "5418B03ED833D36CA71176A9FD7F8C55607D68CAED48F12AE347403AAF1276EB"]
[MSG:DBG: Test2.nc hash "8D9F5A68C676CD02C0BF98AB785FFD8D09FCFFE88224132E5E70E7040755F60A"]
[MSG:DBG: favicon.ico hash "B8B2871A343CA0F9A7A130213226962AD2BE4DFD28D7D8B756C3557569CB876D"]
[MSG:DBG: index.html.gz hash "3B628023FBEBA2C44FCC0CD494849DE07F5D327C98AD70F5DF5B2CD374B0DD81"]
[MSG:DBG: labyrinth_v6_nF_mitSchleifen.nc hash "B57C4D26A7815FAD2E533C6D7A9FFA9AE624093BFC03518A2E6B20B615291CB1"]
[MSG:DBG: labyrinth_v6_neueFraese.nc hash "F496D7683F1C3A9FD34EE046530E95EED8A4DCB54DC316C55412DC9F177F1C87"]
[MSG:DBG: macrocfg.json hash "B2286D84128AA530EC8505E427D3B1B370332CE93AA8B240C428EB23B609E2FB"]
[MSG:DBG: preferences.json hash "27C4E23A5A808CBFCC12CBD3B06F7701C486CFDDA7A0A59B823E1A8166A122C9"]
[MSG:DBG: schleifen.nc hash "EE2141634A559B30A7569EA5442D21CFAB143CBA187E235EE70F743B1ACFD831"]
[MSG:INFO: Telnet started on port 23]
ok
```

### User Interface Software

WebUI

### What happened?

I did a firmwareupdate OTA:
uploaded firmware.bin over Web-Interface
uploaded index.html.gz

restart and the Pins did not work anymore

breiler, 2023-11-07:
The GPIO pins should not be defined with a colon (`:`).

Change every pin that is defined like this `gpio:36` to `gpio.36`

bdring, 2023-11-07:
Try sending `$localfs/show=6_pack_drehfraese_4.yaml` to show the actual file stored on the chip.

It seems really strange that the old revision was able to load a file with `gpio:13`

srgkasuro, 2023-11-07:
yes, it is strange, but it is.
3.7.0 did load the File with gpio:13 (and other pins) and it worked correct.

But now its fine and working with dots.
thank you very much!

closed 2023-11-07 by srgkasuro