bdring/FluidNC #1061: Since version 3.6.5 FluidNC does not wait for the VFD spindle to spin up

gjkrediet, 2023-11-07:
### Wiki Search Terms

VFD spin up 

### Controller Board

External drivers - standard stepper

### Machine Description

Gantry router, external steppers, dual Y-axis VFD  Huanyang  HY01D523B connected through RS485.

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"
start:
    must_home: true
    
stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 3
  pulse_us: 3
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: gpio.13:high
  

  x:
    steps_per_mm: 640
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 150
    max_travel_mm: 837
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 1500
      feed_mm_per_min: 200
      settle_ms: 100
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:high
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        direction_pin: gpio.16
        step_pin: gpio.17

  y:
    steps_per_mm: 640
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 130
    max_travel_mm: 690
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 1500
      feed_mm_per_min: 200
      settle_ms: 100
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:high
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        direction_pin: gpio.27
        step_pin: gpio.26

    motor1:
      limit_neg_pin: gpio.36:high
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        direction_pin: gpio.12
        step_pin: gpio.14

  z:
    steps_per_mm: 2500
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 120
    soft_limits: true
    homing:
      cycle: 1
      mpos_mm: 0
      positive_direction: true
      seek_mm_per_min: 1000
      feed_mm_per_min: 200
      settle_ms: 100
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.35:high
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        direction_pin: gpio.5:low
        step_pin: gpio.22

Huanyang:
  uart:
    txd_pin: gpio.15
    rxd_pin: gpio.4
    rts_pin: gpio.2
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=12.5% 3000=12.5% 24000=100%  
  
sdcard:
    cs_pin: NO_PIN
    card_detect_pin: NO_PIN
 
spi:
    miso_pin: NO_PIN
    mosi_pin: NO_PIN
    sck_pin: NO_PIN
    
control:
  feed_hold_pin: gpio.33:low:pu
  cycle_start_pin: gpio.32:low:pu
  
probe:
  pin: gpio.25:low:pu
  check_mode_start: true
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.6.4]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI not defined]
[MSG:INFO: No SD Card CS Pin]
[MSG:INFO: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:3us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,837.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.17 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.34]
[MSG:INFO: Axis Y (0.000,690.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.36]
[MSG:INFO: Axis Z (-120.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.22 Dir:gpio.5:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.35]
[MSG:INFO: feed_hold_pin gpio.33:low:pu]
[MSG:INFO: cycle_start_pin gpio.32:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Huanyang Spindle Uart Tx:gpio.15 Rx:gpio.4 RTS:gpio.2 Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Probe Pin: gpio.25:low:pu]
[MSG:INFO: BT Started with FluidNC]
ok

[MSG:INFO: FluidNC v3.7.9 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:3us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,837.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.17 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.34]
[MSG:INFO: Axis Y (0.000,690.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.36]
[MSG:INFO: Axis Z (-120.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.22 Dir:gpio.5:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.35]
[MSG:INFO: feed_hold_pin gpio.33:low:pu]
[MSG:INFO: cycle_start_pin gpio.32:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: UART1 Tx:gpio.15 Rx:gpio.4 RTS:gpio.2 Baud:9600]
[MSG:INFO: Huanyang Spindle  Tx:gpio.15 Rx:gpio.4 RTS:gpio.2 Baud:9600]
[MSG:INFO: Using spindle Huanyang]
[MSG:INFO: Probe Pin: gpio.25:low:pu]
[MSG:INFO: Huanyang PD0011, PD005 Freq range (50,400) Hz (3000,24000) RPM]
[MSG:INFO: BT Started with FluidNC]
```

### User Interface Software

Candle 1.2.15

### What happened?

I recently  updated FluidNC to the latest version and since then the router is not waiting for the spindle to spin up. So it dives into the material while spinning up.
In my attempts to detect/solve the problem I reset the VFD to factory settings and set the registers as prescripted in the wiki (among which PD0014 and PD0015 to 6.00). This did not make any difference. I then backed up to previous versions and found version 3.6.4 to be the last one not showing this problem. In the release-notes of version 3.6.5 is mentioned that spin-up times were added. Am I doing something wrong or is this an error in FluidNC?
Btw the startup-messages I provided are from the latest working version followed by the the output from the most recent version.

bdring, 2023-11-07:
Add spinup_ms/spindown_ms delays to your config as shown below. Use delays similar to the ones on the VFD.

```yaml
Huanyang:
  uart:
    txd_pin: gpio.15
    rxd_pin: gpio.4
    rts_pin: gpio.2
    baud: 9600
    mode: 8N1
  modbus_id: 1
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0% 0=12.5% 3000=12.5% 24000=100%
  ```

gjkrediet, 2023-11-07:
OK. Thanks a lot. Is this documented anywhere? I noticed in the Wiki that the startup-message contains values P0014 and P0015 while mine does not.

bdring, 2023-11-07:
The spin delays are documented on the spindle page. Not all examples show all parameters. You can send `$Huanyang` to see any default parameters you did not set in the config.

http://wiki.fluidnc.com/en/config/config_spindles

I don't know why those values do not show in your startup. It has been a while since I used a Huanyang. It takes a long time for me to setup that test rig.

gjkrediet, 2023-11-07:
I assumed fluidNC waits for a 'clear' from the VFD when it has reached its speed. The hardware provides such a signal but aparently not so via RS485. Howerever, this solves the problem. Thanks again.

closed 2023-11-07 by gjkrediet