bdring/FluidNC #1062: Problem: Jog moves wrong way after pressing stop button in UGS

banana-legs, 2023-11-07:
### Wiki Search Terms

Jogging, UGS

### Controller Board

External drivers with standard steppers

### Machine Description

XYYZ gantry router with 2.2kW spindle on 10v drive.

### Configuration file

```yaml
board: "ESP32 Dev Controller V1 EJH"
name: "ESP32 Dev Controller V1 EJH 20221226"

stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4.0
  dir_delay_us: 6
  disable_delay_us: 10

axes:
  shared_stepper_disable_pin: gpio.13
  x:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 7000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 802.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.36
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: gpio.33
        direction_pin: gpio.32
        disable_pin: NO_PIN

  y:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 560.000
    soft_limits: true
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.39
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.25
        disable_pin: NO_PIN

    motor1:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: gpio.14
        direction_pin: gpio.27
        disable_pin: NO_PIN

  z:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 80.000
    soft_limits: true
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: gpio.19
        direction_pin: gpio.18
        disable_pin: NO_PIN

  a:
    steps_per_mm: 106.667
    max_rate_mm_per_min: 720.000
    acceleration_mm_per_sec2: 20.000
    max_travel_mm: 360.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 200.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.2
        direction_pin: gpio.15
        disable_pin: NO_PIN

control:
  safety_door_pin: gpio.21:low
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: gpio.16:low
  check_mode_start: true

macros:
  startup_line0:
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  deactivate_parking: false
  check_limits: true

parking:
  enable: true
  axis: Z
  target_mpos_mm: 0.0
  rate_mm_per_min: 3000.000
  pullout_distance_mm: 10.000
  pullout_rate_mm_per_min: 400.000

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: gpio.23
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: true

10V:
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  pwm_hz: 1000
  direction_pin: NO_PIN
  output_pin: gpio.17
  enable_pin: gpio.5
  disable_with_s0: true
  s0_with_disable: true
  spinup_ms: 10000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0.000% 0=20.200% 6000=20.200% 12000=45.400% 18000=70.400% 24000=100.000%
  off_on_alarm: true

Laser:
  pwm_hz: 1000
  output_pin: gpio.4:pd
  enable_pin: gpio.22
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 100
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: true
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.9 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine ESP32 Dev Controller V1 EJH 20221226]
[MSG:INFO: Board ESP32 Dev Controller V1 EJH]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:10us Dir Delay:6us Idle Delay:255ms]
[MSG:INFO: User Digital Output:1 on Pin:gpio.23]
[MSG:INFO: Axis count 4]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (0.000,802.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.33 Dir:gpio.32 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.36]
[MSG:INFO: Axis Y (0.000,560.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.39]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y2 Neg Limit gpio.34]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.19 Dir:gpio.18 Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.35]
[MSG:INFO: Axis A (-360.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.2 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO: safety_door_pin gpio.21:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:gpio.5 Out:gpio.17 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:1000Hz Period:65535]
[MSG:INFO: Laser Ena:gpio.22 Out:gpio.4:pd Freq:1000Hz Period:65535]
[MSG:INFO: Using spindle 10V]
[MSG:INFO: Probe Pin: gpio.16:low]
[MSG:INFO: Connecting to STA SSID:workshop]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.2.50]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC v3.7.9 (wifi) '$' for help]
[MSG:INFO: ALARM: Unhomed]
ALARM:14
```

### User Interface Software

UGS

### What happened?

When running a programme in UGS, if the 'stop' button is pressed to abort the milling operation, the machine comes to a halt and the spindle stops as expected (usually with the bit still in the channel being cut).    The very next jog operation, which is usually one to move Z up to clear the bit from the work piece, does not usually move in the direction commanded; the jog almost always moves sideways instead of up and breaks the milling cutter.   After that erroneous jog operation has completed (it usually moves really slowly too), all subsequent jog operations work fine as if nothing was ever wrong.    I have found the problem does not matter which gcode file I am running as it has happend on many occasions (yes, I am that bad at starting a job correctly first time!).

I have checked back by installing older firmware versions and version 3.7.5 and earlier all work fine and the jogging does not have the fault, but the problem exists in all versions from 3.7.6 onwards.

bdring, 2023-11-07:
I am not sure exactly what the "stop" button does in UGS.

Ideally you would do a reset and then home ($H or $HZ) to recover from that problem.

If you are doing a feed hold, you would do a reset and then could either home or jog up.

banana-legs, 2023-11-07:
luckily $HZ on its own works fine, so I have added a macro button to do that as a work-around.   Force of habit still has me sometimes hitting the Z-up on the jog controller berfore remembering though!

bdring, 2023-11-07:
Many senders do different things with "stop".

There is no stop in the Grbl protocol.

Can we close this issue?

banana-legs, 2023-11-07:
I had a look through the code changes that occured between 3.7.5 and 3.7.6.   There was nothing obvious that jumped out, but there were a number of adjustments to what happens on abort; in 3.7.5 it looks like an abort threw back into the main loop so effectively did a reset.   Now the code keeps going instead of directly resetting and I wonder if the planner buffer is not being flushed properly prior to the jog?     

Fundamentally, a bug crept in somewhere from 3.7.5 to 3.7.6, but it looks like it could be a subtle one that is not easy to trace.    It may need someone who knows about what the 'stop' sequence in UGS is to ascertain what sequence of events are being triggered in the abort process in order to work out what is happening.     For now I can live with the work around, but I suspect it is worth keeping an eye on rather than closing as it can directly lead to a break in an expensive milling cutter, which is not great PR for such a fantastic CNC controller.

breiler, 2023-11-07:
Just to make sure, are you connecting with the FluidNC firmware profile in UGS?
What version of UGS are you using?

The code for a "stop" in UGS is done here: https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/firmware/fluidnc/FluidNCController.java#L492

1. Clears internal send buffer
2. Stops any live stream
3. Issues a feed hold command
4. Waiting for HOLD:0 (a complete stop of the machine)
5. Sends a a reset command (0x18)

Can you post the console output from UGS when this happens?

banana-legs, 2023-11-07:
I am using version 20.21, but it also happens with 20.20.   Yes it is using the FluidNC firmware profile.   All I get in the command window when I cancel is  
ok
*** Canceling command stream   

When all has stopped and I then do a jog:
$J=G21G91Z0.099F593.829

The CNC moves slowly for a few seconds in x and y and not z.

bdring, 2023-11-07:
What does the '?' status command reply?

bdring, 2023-11-08:
I think we found the issue. It is something we fixed a little while ago, but then we had a github mishap where we lost few commits. The problem was fixed again and has been waiting for a couple weeks in the devt branch for the next release.

It has to do with some residual arc segments left in the planner.

I hope to do a release in the next few days.

https://github.com/bdring/FluidNC/commit/2da13de

banana-legs, 2023-11-08:
Awesome, very many thanks for all your help :)

closed 2024-02-02 by bdring