bdring/FluidNC #1064: Problem: Uart 1 report_interval_ms: 400 / [MSG:WARN: Ignored key report_interval_ms]

tower-guy, 2023-11-09:
### Wiki Search Terms

[PendantsForFluidNC](https://github.com/bdring/PendantsForFluidNC/tree/main)

/DisplayToTerminal

### Controller Board

DIY

### Machine Description

Gantry router with DM542 on X axis and MKS TMC2160-OC on Y & Z

### Configuration file

```yaml
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

start:
  must_home: false
  deactivate_parking: false
  check_limits: false

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 10
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: i2so.3:high
  
  x:
    steps_per_mm: 320
    max_rate_mm_per_min: 4500
    acceleration_mm_per_sec2: 500
    max_travel_mm: 540
    homing:
      cycle: 2
      mpos_mm: 10
      seek_mm_per_min: 1500
      positive_direction: false
  
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:    
        direction_pin: gpio.12
        step_pin: gpio.26
    motor1:
      null_motor:

  y:
    steps_per_mm: 400
    max_rate_mm_per_min: 4500
    acceleration_mm_per_sec2: 500
    max_travel_mm: 320
    homing:
      cycle: 2
      mpos_mm: 10
      seek_mm_per_min: 1500
      positive_direction: false

    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper: 
        direction_pin: gpio.14
        step_pin: gpio.25
    motor1:
      null_motor:

  z:
    steps_per_mm: 640
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 100
    homing:
      cycle: 1
      mpos_mm: 10
      seek_mm_per_min: 500
      positive_direction: true

    motor0:
      limit_all_pin: gpio.39:low
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper: 
        direction_pin: gpio.27
        step_pin: gpio.33
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.0
  
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
   
probe:
  pin: gpio.36:low

PWM:
  pwm_hz: 5000
  output_pin: gpio.32
  enable_pin: i2so.1
  direction_pin: i2so.2
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 8600=50% 12500=100%

status_outputs:
  idle_pin: i2so.4
  run_pin: i2so.5
  hold_pin: i2so.6
  alarm_pin: i2so.7
  
uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  baud: 115200
  mode: 8N1
  
uart_channel1:
  uart_num: 1
  report_interval_ms: 400
```

### Startup Messages

```text
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4

[MSG:INFO: FluidNC 3.7.10 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key report_interval_ms]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:115200]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.0 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Status outputs Interval:500 Idle:I2SO.4 Cycle:I2SO.5 Hold:I2SO.6 Alar       m:I2SO.7]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:10us Idle Delay:250ms       ]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.3]
[MSG:INFO: Axis X (10.000,550.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,330.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.25 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-90.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.33 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:I2SO.1 Out:gpio.32 Dir:I2SO.2 Freq:5000Hz Period:8191       ]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.36:low]
[MSG:INFO: Connecting to STA SSID:Engine House]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.48]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.7 [FluidNC 3.7.10 (wifi) '$' for help]
```

### User Interface Software

Putty

### What happened?

No status report as expected

bdring, 2023-11-09:
I am sorry for the confusion. auto reporting is not part of ***uart_channels*** yet. It is in the ***channel_pin_extender***  branch.

https://github.com/bdring/FluidNC/blob/channel_pin_extender/FluidNC/src/UartChannel.h#L44

That branch is currently under active development. It has a lot of open issues and will not run normal configs yet.

bdring, 2023-11-09:
Here is the commit to that branch.

https://github.com/bdring/FluidNC/commit/05641dc

That wiki page is being updated to track the proposed changes. Sorry about the confusion.

bdring, 2023-11-11:
I created a PR with the changes needed.

https://github.com/bdring/FluidNC/pull/1068

You will have to compile yourself until it is released.

tower-guy, 2023-11-11:
Thanks for looking at it.

bdring, 2024-12-12:
Cleaning up old issues

closed 2024-12-12 by bdring