bdring/FluidNC #1065: Problem: While running a file, Status changes from "run" to "Idle" while waiting for Spindle spinup/down or G4

srgkasuro, 2023-11-10:
### Wiki Search Terms

Status, run, idle 

### Controller Board

6-pack

### Machine Description

Not relevant

### Input Circuits

```yaml
Not relevant
```

### Configuration file

```yaml
Not relevant
```

### Startup Messages

```text
Not relevant
```

### What happened?

While running a File with: 
$LocalFS/Run=<filename>
the Status changes from "run" to "Idle" while waiting for Spindle spinup/down or G4.
I think it should stay in "run".
There are multiple M3/M5 in my File due to tool changes.

I need to know, when the job is really completed, because I want it to start automatically again. (Pendant reads the messages and starts $LocalFS/Run=<filename> again)

I tried waiting for MSG "...file job succeeded]"
but there is the Problem, that this Message is sent when the last line is sent to buffer, not when the job is done.

thank you very much,
Suheel

bdring, 2023-11-10:
Unfortunately, that is normal behavior. The machine goes into idle when the motion planner is empty. We are stuck with the Grbl protocol to keep compatibility with gcode senders.

We are working on a job context system, where we could signal when the job is done in most cases. We don't want to work on "simple" workarounds, because edge cases would be a support nightmare.

bdring, 2023-11-10:
Here is a wiki page discussing this issue.

http://wiki.fluidnc.com/en/development/job-context-design

srgkasuro, 2023-11-12:
Hi, thank you very much.
sorry, I did not found this Wikipage before.
I found a solution for my needs, that seems to work:
at the end of my NC-program I put a comment with (MSG "Finished-flag")
and wait for this Message in the Serial-Stream.

bdring, 2023-11-12:
That sounds like a good method.

There are many pages that are not on the nav panel on the side. It pays to search a lot of terms are look at all the results.

closed 2023-11-12 by bdring